lvgl-micropython/lvgl_micropython #222: Build failes for esp32s3

VsevolodVasin, 2024-12-14:
**Describe the bug**
The build fails when doing
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=rgb_display INDEV=FT6x36

If you are having a problem compiling paste the build output below. (NO SCREENSHOTS)

attached
[issue.txt](https://github.com/user-attachments/files/18136811/issue.txt).

h-milz, 2024-12-14:
I compiled a couple of build today with the same config except the touch. You may want to  make a fresh cloned and start over to make sure everything is up to date. Also I believe the ft part should be written in small letters, not FT.

ANARHIST1984, 2024-12-14:
Hello
I got same error.
Fresh system
[image]

try with capital FT and small

kdschlosser, 2024-12-14:
You have to read the readme. You need to install doxygen using brew.

closed 2024-12-15 by VsevolodVasin