lvgl-micropython/lvgl_micropython #223: Color error after ILI9488 driver

center789, 2024-12-14:
```
import machine  # For hardware interface
import lcd_bus  # For LCD screen bus management
import ili9488  # For driving the ST7789 display
import lvgl as lv  # GUI development library
import random  # For random number generation
import time  # For time management
import task_handler  # For task handling module

# Initialize SPI bus
spi_bus = machine.SPI.Bus(
    host=2,  # Use SPI2 host
    mosi=11,  # MOSI pin GPIO 11
    miso=-1,  # MISO not used, pin set to -1
    sck=12  # SCK pin GPIO 12
)

# Initialize the display's SPI bus interface
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=13,  # Data/Command control pin GPIO 13
    cs=10,  # Chip select pin GPIO 10
    freq=60000000  # SPI communication frequency 60 MHz
)

# Allocate double buffer
buf1 = display_bus.allocate_framebuffer(46080, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
buf2 = display_bus.allocate_framebuffer(46080, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

# Initialize ST7789 display
display = ili9488.ILI9488(
    data_bus=display_bus,
    display_width=320,
    display_height=480,
    frame_buffer1=buf1,
    frame_buffer2=buf2,
    reset_pin=9,
    reset_state=ili9488.STATE_LOW,
    backlight_pin=15,
    backlight_on_state=ili9488.STATE_PWM,
    color_space=lv.COLOR_FORMAT.RGB888,
    color_byte_order=ili9488.BYTE_ORDER_BGR,
    rgb565_byte_swap=False, 
)
# display.set_color_inversion(False) 
display.set_power(True)
display.init()
display.set_backlight(90)
# Set screen orientation: 0, 1, 2, 3
display.set_rotation(3)
# Set horizontal and vertical offsets
display._offset_y = (0)  # rotation(0 or 2) Y direction offset by 20 pixels down
display._offset_x = (0)  # rotation(1 or 3) X direction offset by 20 pixels down

# Initialize task handler
task_handler.TaskHandler()

# Get the current active screen
scrn = lv.screen_active()

# colors = [
#     lv.color_hex(0xFF0000),  # BGR red turns into yellow, RGB red turns into cyan
#     lv.color_hex(0x00FF00),  # BGR green turns into purple, RGB green turns into purple
#     lv.color_hex(0x0000FF)   # BGR blue turns into cyan, RGB blue turns into yellow
# ]
colors = [
    lv.color_hex(0x0000FF),  # BGR turns blue into cyan, RGB turns blue into yellow
    lv.color_hex(0xFF00FF),  # BGR turns purple into yellow, RGB turns purple into green
    lv.color_hex(0xFFFF00),  # BGR turns purple into red, RGB turns yellow into blue
]

start_time = time.ticks_ms()  # Get start time
count = 0  # Used to track loop count
i = 0  # Current color index

while count != 10:  # Loop twice
    end_time = time.ticks_ms()  # Get current time
    if time.ticks_diff(end_time, start_time) >= 5000:  # Check if 5 seconds passed
        scrn.set_style_bg_color(colors[i], 0)  # Set background color
        print(f"Current color: {colors[i]}") 
        i = (i + 1) % len(colors)  # Cycle through color indices
        count += 1  # Increment count
        start_time = end_time  # Reset start time
``` 
Do I need to modify the configuration file of _ili9488_init.py? My screen driver is ili9488, and the touch chip is FT6336. The issue I'm facing right now is incorrect color display, and I need your help and guidance.

kdschlosser, 2024-12-14:
There are a few things that can effect the color.

Now you said with the byte order set to RGB and setting the color to `0xFF0000` you get yellow. So what is Yellow? Yellow is 0xFFFF00. which is 255, 255, 0. and the color you are setting is 255, 0, 0. Now what would cause that? Well if the display is writing the data as BGR then the color you are setting is 0x0000FF or 0, 0, 255. and if the display is looking at those colors as being inverted you get 255, 255, 0.

If we leave inversion set the way it is and change the byte order to BGR you end up with the color being set as Cyan or 0, 255, 255. So you are 1/2 way to having the correct color show up. Now you just need to turn inversion either on or off. so use `set_color_inversion` to turn it on or off. It could be defaulted to being on so you may have to turn it off or it could be defaulted to being turned off and you need to turn it on.

color inversion is literally the opposite bits being set. so 0xFF00FF would become 0x00FF00 

So setting the `color_byte_order` to `BYTE_ORDER_BGR` and turning color inversion either on or off will correct your color issue.

center789, 2024-12-15:
You explained it very thoroughly, and I understand it now. I tried using ```display.set_color_inversion(False)``` and ```display.set_color_inversion(True)```, but the screen's colors didn't change. Am I using the wrong method? But this is the only method I found through code completion.

kdschlosser, 2024-12-15:
I am going to look at the data sheet for the display IC. I could have something mucked up with it.

center789, 2024-12-15:
[Uploading ILI9488_Preliminary_DS_V090.pdf…]()

kdschlosser, 2024-12-15:
??? that links right back to this issue

center789, 2024-12-15:
The vendor gave me this ILI9488_Preliminary_DS_V090.pdf document, but I'm not sure if it will help you identify the issue. I just wanted to share whatever resources I have with you.

center789, 2024-12-15:
```
def init(self):
   #.............

    self.set_params(0x21)  

    self.set_params(_DISPON)
    time.sleep_ms(100)
``` 
"I've found a solution. I modified the lvgl_micropython/api_drivers/common_api_drivers/display/ili9488/_ili9488_init.py file and added self.set_params(0x21) at the end to forcibly enable or disable color inversion to achieve normal color display.

closed 2024-12-16 by center789