lvgl-micropython/lvgl_micropython #226: FT6X36 touch coordinate mirroring issue

center789, 2024-12-16:
```
import machine 
import lcd_bus  
import ili9488  
import lvgl as lv  

spi_bus = machine.SPI.Bus(
    host=2,  
    mosi=11,  
    miso=-1,  
    sck=12  
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=13,  
    cs=10,  
    freq=60000000  
)

buf1 = display_bus.allocate_framebuffer(30720, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
buf2 = display_bus.allocate_framebuffer(30720, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

lv.init()

display = ili9488.ILI9488(
    data_bus=display_bus,
    display_width=320,
    display_height=480,
    frame_buffer1=buf1,
    frame_buffer2=buf2,
    reset_pin=9,
    reset_state=ili9488.STATE_LOW,
    backlight_pin=15,
    backlight_on_state=ili9488.STATE_PWM,
    color_space=lv.COLOR_FORMAT.RGB888,
    color_byte_order=ili9488.BYTE_ORDER_RGB,
    rgb565_byte_swap=True, 
)

import i2c  
import task_handler
import ft6x36  
_SCL = const(17)
_SDA = const(16)

display.init()

i2c_bus = i2c.I2C.Bus(
    host=1,
    scl=_SCL,
    sda=_SDA,
    use_locks=False  
)

touch_dev = i2c.I2C.Device(bus=i2c_bus, dev_id=ft6x36.I2C_ADDR, reg_bits=ft6x36.BITS)
indev = ft6x36.FT6x36(touch_dev,startup_rotation=lv.DISPLAY_ROTATION._90,debug=True)

if not indev.is_calibrated:
    display.set_backlight(100)
    indev.calibrate()

display.set_rotation(lv.DISPLAY_ROTATION._90)
display.set_power(True)
display.set_backlight(60)

th = task_handler.TaskHandler(1)

scrn = lv.screen_active()  
ui_Screen1 = lv.obj()
[... 547 lines cut ...]
FT6x36(raw_x=107, raw_y=265, x=116, y=286, state=PRESSED)
FT6x36(raw_x=103, raw_y=261, x=112, y=281, state=PRESSED)
FT6x36(raw_x=97, raw_y=254, x=107, y=274, state=PRESSED)
FT6x36(raw_x=95, raw_y=251, x=105, y=270, state=PRESSED)
FT6x36(raw_x=92, raw_y=247, x=102, y=266, state=PRESSED)
FT6x36(raw_x=91, raw_y=243, x=101, y=262, state=PRESSED)
FT6x36(raw_x=92, raw_y=235, x=102, y=253, state=PRESSED)
FT6x36(raw_x=93, raw_y=228, x=103, y=246, state=PRESSED)
FT6x36(raw_x=95, raw_y=219, x=105, y=237, state=PRESSED)
FT6x36(raw_x=97, raw_y=213, x=107, y=231, state=PRESSED)
FT6x36(raw_x=104, raw_y=203, x=114, y=221, state=PRESSED)
FT6x36(raw_x=109, raw_y=198, x=118, y=216, state=PRESSED)
FT6x36(raw_x=117, raw_y=192, x=126, y=211, state=PRESSED)
FT6x36(raw_x=121, raw_y=188, x=130, y=207, state=PRESSED)
FT6x36(raw_x=126, raw_y=184, x=134, y=203, state=PRESSED)
FT6x36(raw_x=128, raw_y=182, x=136, y=201, state=PRESSED)
FT6x36(raw_x=130, raw_y=180, x=138, y=199, state=PRESSED)
FT6x36(raw_x=132, raw_y=179, x=140, y=198, state=PRESSED)
FT6x36(raw_x=137, raw_y=176, x=145, y=196, state=PRESSED)
FT6x36(raw_x=137, raw_y=176, x=145, y=196, state=RELEASED)
```

The touch functionality of the FT6X36 is experiencing a mirroring issue, and calibrating the screen doesn’t seem to have any effect. I’ve also seen that others have encountered this issue before. Is there currently no solution to this bug?

https://github.com/user-attachments/assets/5db432b5-1850-4202-bc6a-41536eebb7da

tetopik, 2024-12-16:
Sorry to ask you outside the topic, but how to add those FPS at the bottom right corner of the screen? Thankyou

center789, 2024-12-16:
> FPS

edit the lvgl_micropython/lib/lv_conf.h file and change the following settings
#define LV_USE_OBSERVER 1
#define LV_USE_SYSMON   1
#define LV_USE_PERF_MONITOR 1

center789, 2024-12-16:
> 抱歉问你主题之外的问题，但是如何在屏幕右下角添加那些 FPS？谢谢

But you must compile the latest 9.2.2 firmware because editing it in 9.1.1 will not take effect.

tetopik, 2024-12-16:
> But you must compile the latest 9.2.2 firmware because editing it in 9.1.1 will not take effect.

Thanks, going to test it very soon!

kdschlosser, 2024-12-16:
I will look at the mirroring issue to see what the best way of handling it is. I just thought of the way to to handle it (I think)

kdschlosser, 2024-12-16:
I was looking over the log you posted. The calibration is being done properly you can see this in the log files.

I am thinking the rotation values are incorrect for your display.

LVGL rotates the coordinates in a clockwise fashion.

rotation 0: 
x = x
y = y

rotation 90:
tmp = x
x = y
y = tmp

rotation 180:
x = display_width - x - 1
y = display_height - y - 1

rotation 270:
tmp = x
x = y
y = display_width - tmp - 1

so if the rotation is actually happening counter clockwise the calculations would actually need to be

rotation 0: 
x = x
y = y

rotation 90:
tmp = x
x = y
y = display_width - tmp - 1

rotation 180:
x = display_width - x - 1
y = display_height - y - 1

rotation 270:
tmp = x
x = y
y = tmp

So an adjustment needs to be made in the display driver and not in the touch driver.

Easy way to check this is to set the rotation to zero and hold the display the way the displayed UI is orientated and then change the rotation to 90 and see which way you need to physically rotate the display to see the UI correctly. If it is counter clockwise then the rotation parameters for the display need to be adjusted.

kdschlosser, 2024-12-16:
Try adding this right after your imports at the top...

```python
ili9488.ILI9488._ORIENTATION_TABLE = (0x00, 0x20, 0xC0, 0xE0)
```

kdschlosser, 2024-12-16:
I forgot to mention using the `enable_input_priority` function in the touch driver. If you want you UI to respond to user input as the top most priority then call that function. Once it is turned on it cannot be turned off. ONLY call that function at the very end of your startup code before the task handler is started. The improvement will be realized when performing dragging input like scrolling or moving a slider. Click type operations are not going to be impacted a huge amount so you may not notice any kind of an improvement in responsiveness for those kinds of input.

kdschlosser, 2024-12-16:
The other thing is the calibration really only needs to be performed when using resistive type touch panels. The FT line of touch panels are capacitive and typically do not need to be calibrated. The display manufacturers will typically set the firmware in the touch IC so the resolution of the touch panel will match the display resolution. This might not always be the case but they are pretty good about doing it. How you can tell if it has been done is by looking at the debugging output for the touch driver. 

```
FT6x36(raw_x=17, raw_y=431, x=30, y=451, state=PRESSED)
```

the `raw_x` and `raw_y` are the coordinates that are received from the touch IC and the `x` and `y` are what is being passed to LVGL after the calibration has been done. if you notice there is not much of a difference between the numbers and that is because the firmware is set properly in your touch IC so the touch resolution is the same as the display resolution. The variation you are seeing is being caused by your finger due to the size of it.

center789, 2024-12-17:
> 尝试一下此代码，
> 
```
Touch Device ID: 0x11
Touch Chip ID: 0x64
Touch Device mode: 0x00
Touch Firmware ID: 0x03
Touch Release code: 0x01
point 1 of 3
   1 of 8: (294, 19)
   2 of 8: (294, 19)
   3 of 8: (294, 19)
   4 of 8: (294, 19)
   5 of 8: (294, 19)
   6 of 8: (294, 19)
   7 of 8: (294, 19)
   8 of 8: (294, 19)
point 2 of 3
   1 of 8: (9, 26)
   2 of 8: (9, 26)
   3 of 8: (9, 26)
   4 of 8: (9, 26)
   5 of 8: (9, 26)
   6 of 8: (9, 26)
   7 of 8: (9, 26)
   8 of 8: (9, 26)
point 3 of 3
   1 of 8: (296, 436)
   2 of 8: (296, 436)
   3 of 8: (296, 436)
   4 of 8: (296, 436)
   5 of 8: (296, 436)
   6 of 8: (296, 436)
   7 of 8: (296, 436)
   8 of 8: (296, 436)

averaged trimmed points
  point 1 of 3: (294, 19)
  point 2 of 3: (9, 26)
  point 3 of 3: (296, 436)

calibration values
  alphaX: -0.9121733
  betaX: 0.004374932
  deltaX: 298.0958
  alphaY: 0.02473519
  betaY: 1.007076
  deltaY: 3.593417
>>> FT6x36(raw_x=296, raw_y=431, x=289, y=438, state=PRESSED)
FT6x36(raw_x=296, raw_y=431, x=289, y=438, state=RELEASED)
FT6x36(raw_x=50, raw_y=425, x=65, y=438, state=PRESSED)
FT6x36(raw_x=50, raw_y=425, x=65, y=438, state=RELEASED)
FT6x36(raw_x=50, raw_y=430, x=65, y=443, state=PRESSED)
FT6x36(raw_x=51, raw_y=430, x=66, y=443, state=PRESSED)
FT6x36(raw_x=51, raw_y=430, x=66, y=443, state=RELEASED)
FT6x36(raw_x=62, raw_y=43, x=77, y=53, state=PRESSED)
FT6x36(raw_x=62, raw_y=43, x=77, y=53, state=RELEASED)
FT6x36(raw_x=58, raw_y=41, x=74, y=51, state=PRESSED)
FT6x36(raw_x=58, raw_y=41, x=74, y=51, state=RELEASED)
FT6x36(raw_x=61, raw_y=41, x=76, y=51, state=PRESSED)
FT6x36(raw_x=61, raw_y=41, x=76, y=51, state=RELEASED)
FT6x36(raw_x=62, raw_y=37, x=77, y=47, state=PRESSED)
FT6x36(raw_x=62, raw_y=37, x=77, y=47, state=RELEASED)
FT6x36(raw_x=72, raw_y=54, x=86, y=64, state=PRESSED)
FT6x36(raw_x=72, raw_y=54, x=86, y=64, state=RELEASED)
FT6x36(raw_x=62, raw_y=57, x=77, y=67, state=PRESSED)
FT6x36(raw_x=62, raw_y=57, x=77, y=67, state=RELEASED)
FT6x36(raw_x=47, raw_y=55, x=64, y=65, state=PRESSED)
FT6x36(raw_x=47, raw_y=55, x=64, y=65, state=RELEASED)
FT6x36(raw_x=84, raw_y=54, x=97, y=63, state=PRESSED)
[... 97 lines cut ...]
FT6x36(raw_x=313, raw_y=41, x=306, y=45, state=PRESSED)
FT6x36(raw_x=310, raw_y=41, x=303, y=45, state=PRESSED)
FT6x36(raw_x=308, raw_y=41, x=302, y=45, state=PRESSED)
FT6x36(raw_x=306, raw_y=41, x=300, y=45, state=PRESSED)
FT6x36(raw_x=304, raw_y=41, x=298, y=45, state=PRESSED)
FT6x36(raw_x=303, raw_y=41, x=297, y=45, state=PRESSED)
FT6x36(raw_x=303, raw_y=41, x=297, y=45, state=RELEASED)
FT6x36(raw_x=301, raw_y=42, x=295, y=46, state=PRESSED)
FT6x36(raw_x=301, raw_y=42, x=295, y=46, state=RELEASED)
FT6x36(raw_x=302, raw_y=50, x=296, y=55, state=PRESSED)
FT6x36(raw_x=302, raw_y=50, x=296, y=55, state=RELEASED)
FT6x36(raw_x=296, raw_y=37, x=291, y=42, state=PRESSED)
FT6x36(raw_x=296, raw_y=37, x=291, y=42, state=RELEASED)
FT6x36(raw_x=271, raw_y=31, x=268, y=36, state=PRESSED)
FT6x36(raw_x=271, raw_y=31, x=268, y=36, state=RELEASED)
FT6x36(raw_x=291, raw_y=430, x=284, y=438, state=PRESSED)
FT6x36(raw_x=290, raw_y=430, x=284, y=438, state=PRESSED)
FT6x36(raw_x=289, raw_y=430, x=283, y=438, state=PRESSED)
FT6x36(raw_x=289, raw_y=430, x=283, y=438, state=RELEASED)
FT6x36(raw_x=294, raw_y=434, x=287, y=441, state=PRESSED)
FT6x36(raw_x=294, raw_y=434, x=287, y=441, state=RELEASED)
FT6x36(raw_x=274, raw_y=431, x=269, y=439, state=PRESSED)
FT6x36(raw_x=274, raw_y=431, x=269, y=439, state=RELEASED)
```
I tried this code, but there was no change. I want to try modifying the display driver file to swap the 90-degree and 270-degree rotations to see if it achieves the result I want. However, I’m not sure which file path to modify: one is ```lvgl_micropython/api_drivers/common_api_drivers/displaydriver_use_exmple.py```, and the other is ```lvgl_micropython/api_drivers/lvgl_api_drivers/frozen/display/display_driver_framework.py```.

kdschlosser, 2024-12-17:
use the script I gave you and add the following line to it.

```
ili9488.ILI9488._ORIENTATION_TABLE = (0x40, 0xE0, 0x80, 0x20)
```

that flip flops the 90 and 270 degree rotations from what the original ones are set to.

kdschlosser, 2024-12-17:
This is what is set as the default

```python
_MADCTL_ML = const(0x10)  # Refresh 0=Top to Bottom, 1=Bottom to Top
_MADCTL_MV = const(0x20)  # 0=Normal, 1=Row/column exchange
_MADCTL_MX = const(0x40)  # 0=Left to Right, 1=Right to Left
_MADCTL_MY = const(0x80)  # 0=Top to Bottom, 1=Bottom to Top

    _ORIENTATION_TABLE = (
        _MADCTL_MX,  # 0° rotation
        _MADCTL_MV,  # 90° rotation
        _MADCTL_MY,  # 180° rotation
        _MADCTL_MX | _MADCTL_MY | _MADCTL_MV  # 270° rotation
    )

```

kdschlosser, 2024-12-17:
what needs to be known is if rotation zero gets displayed properly and if the touch work correctly with it. If it does then we know where the starting point is.

kdschlosser, 2024-12-17:
I am thinking it is going to need to be something along these lines...

```python
    _ORIENTATION_TABLE = (
        _MADCTL_MX,
        _MADCTL_MV | _MADCTL_MY | _MADCTL_MX,
        _MADCTL_MY,
        _MADCTL_MV
    )
```

kdschlosser, 2024-12-17:
I try to not just hand the code out to people without explaining what is happening. This is how we learn. The graphics driver portion is written in a manner that makes it easily extended and also easily altered. This is done so a user can add a new driver or alter an existing driver to make it work correctly for their use case. Not all displays are made the same even tho the display might be using the same IC as another display. There are differences in how the circuits are connected that cause issues that need to be adjusted out.

This specific issue I believe is an actual driver thing on my end as I have not tested all of these drivers to see if they actually work or not. as people come along that have IC's that have not been tested we work through the glitches and get it running properly. As is with this case. Rotation is one of those areas that is always going to require alterations and there is nothing I can do about that. There is no easy way to know what is going to need to be done either. Good thing is the touch driver is working as it should be and you really don't need to run the calibration with your setup either. It will function like it is supposed to if you eliminate the calibration code.

There is still an issue I need to fix with the calibration which is the saving of the calibration data. I am not sure why it's not saving. I have to look at the code to see what is going on with that end of things.

kdschlosser, 2024-12-17:
OK I just updated the repo. I changed the orientation table for the ili9488 driver and I also updated the touch calibration so it handles rotation a little bit better. I still have to figure out why it's not saving the calibration data...

kdschlosser, 2024-12-17:
I just made some changes to the touch calibration code. If you want to give it a try that would be really helpful. If you could paste the output to me that would be really helpful. Run it once and paste the output and then reset your display so it will run again and paste the output. I am curious to know if it runs the calibration a second time or not.

center789, 2024-12-17:
```
Touch Device ID: 0x11
Touch Chip ID: 0x64
Touch Device mode: 0x00
Touch Firmware ID: 0x03
Touch Release code: 0x01
Traceback (most recent call last):
  File "touch_cal_data.py", line 102, in __init__
OSError: (-4354, 'ESP_ERR_NVS_NOT_FOUND')
point 1 of 3
   1 of 8: (19, 27)
   2 of 8: (19, 27)
   3 of 8: (19, 27)
   4 of 8: (19, 27)
   5 of 8: (19, 27)
   6 of 8: (19, 27)
   7 of 8: (19, 27)
   8 of 8: (19, 27)
point 2 of 3
   1 of 8: (302, 26)
   2 of 8: (302, 26)
   3 of 8: (302, 26)
   4 of 8: (302, 26)
   5 of 8: (302, 26)
   6 of 8: (302, 26)
   7 of 8: (302, 26)
   8 of 8: (302, 26)
point 3 of 3
   1 of 8: (36, 447)
   2 of 8: (36, 447)
   3 of 8: (36, 447)
   4 of 8: (36, 447)
   5 of 8: (36, 447)
   6 of 8: (36, 447)
   7 of 8: (36, 447)
   8 of 8: (36, 447)

averaged trimmed points
  point 1 of 3: (19, 27)
  point 2 of 3: (302, 26)
  point 3 of 3: (36, 447)
mirroring
  mirrored x: False
  mirrored y: False

calibration values
  alphaX: 0.9185965
  betaX: -0.03718129
  deltaX: 13.55056
  alphaY: 0.003533064
  betaY: 0.999857
  deltaY: 2.936733
>>> FT6x36(raw_x=169, raw_y=143, x=163, y=146, state=PRESSED)
FT6x36(raw_x=167, raw_y=143, x=162, y=146, state=PRESSED)
FT6x36(raw_x=160, raw_y=145, x=155, y=148, state=PRESSED)
FT6x36(raw_x=151, raw_y=147, x=147, y=150, state=PRESSED)
FT6x36(raw_x=145, raw_y=150, x=141, y=153, state=PRESSED)
FT6x36(raw_x=138, raw_y=153, x=135, y=156, state=PRESSED)
FT6x36(raw_x=132, raw_y=155, x=129, y=158, state=PRESSED)
FT6x36(raw_x=122, raw_y=158, x=120, y=161, state=PRESSED)
FT6x36(raw_x=116, raw_y=161, x=114, y=164, state=PRESSED)
FT6x36(raw_x=107, raw_y=166, x=106, y=169, state=PRESSED)
FT6x36(raw_x=101, raw_y=171, x=100, y=174, state=PRESSED)
FT6x36(raw_x=99, raw_y=173, x=98, y=176, state=PRESSED)
FT6x36(raw_x=95, raw_y=183, x=94, y=186, state=PRESSED)
FT6x36(raw_x=97, raw_y=199, x=95, y=202, state=PRESSED)
FT6x36(raw_x=104, raw_y=212, x=101, y=215, state=PRESSED)
FT6x36(raw_x=113, raw_y=223, x=109, y=226, state=PRESSED)
FT6x36(raw_x=126, raw_y=238, x=120, y=241, state=PRESSED)
FT6x36(raw_x=132, raw_y=246, x=126, y=249, state=PRESSED)
[... 70 lines cut ...]
FT6x36(raw_x=187, raw_y=161, x=179, y=165, state=PRESSED)
FT6x36(raw_x=194, raw_y=164, x=186, y=168, state=PRESSED)
FT6x36(raw_x=209, raw_y=176, x=199, y=180, state=PRESSED)
FT6x36(raw_x=220, raw_y=195, x=208, y=199, state=PRESSED)
FT6x36(raw_x=224, raw_y=206, x=212, y=210, state=PRESSED)
FT6x36(raw_x=226, raw_y=215, x=213, y=219, state=PRESSED)
FT6x36(raw_x=228, raw_y=222, x=215, y=226, state=PRESSED)
FT6x36(raw_x=228, raw_y=222, x=215, y=226, state=RELEASED)

```

```
Touch Device ID: 0x11
Touch Chip ID: 0x64
Touch Device mode: 0x00
Touch Firmware ID: 0x03
Touch Release code: 0x01
Traceback (most recent call last):
  File "<stdin>", line 51, in <module>
  File "ft6x36.py", line 26, in __init__
  File "focaltech_touch.py", line 87, in __init__
  File "pointer_framework.py", line 22, in __init__
  File "touch_cal_data.py", line 120, in __init__
OverflowError: overflow converting long int to machine word
```

center789, 2024-12-17:
After calibration, it will throw an error when running, and it will no longer be usable.

center789, 2024-12-17:
```
Touch Device ID: 0x11
Touch Chip ID: 0x64
Touch Device mode: 0x00
Touch Firmware ID: 0x03
Touch Release code: 0x01
Traceback (most recent call last):
  File "touch_cal_data.py", line 102, in __init__
OSError: (-4354, 'ESP_ERR_NVS_NOT_FOUND')
>>> FT6x36(raw_x=16, raw_y=33, x=16, y=33, state=PRESSED)
FT6x36(raw_x=16, raw_y=33, x=16, y=33, state=RELEASED)
FT6x36(raw_x=19, raw_y=30, x=19, y=30, state=PRESSED)
FT6x36(raw_x=19, raw_y=30, x=19, y=30, state=RELEASED)
FT6x36(raw_x=38, raw_y=33, x=38, y=33, state=PRESSED)
FT6x36(raw_x=38, raw_y=33, x=38, y=33, state=RELEASED)
FT6x36(raw_x=283, raw_y=36, x=283, y=36, state=PRESSED)
FT6x36(raw_x=283, raw_y=36, x=283, y=36, state=RELEASED)
FT6x36(raw_x=265, raw_y=428, x=265, y=428, state=PRESSED)
FT6x36(raw_x=265, raw_y=428, x=265, y=428, state=RELEASED)
FT6x36(raw_x=77, raw_y=238, x=77, y=238, state=PRESSED)
FT6x36(raw_x=77, raw_y=237, x=77, y=237, state=PRESSED)
FT6x36(raw_x=77, raw_y=236, x=77, y=236, state=PRESSED)
FT6x36(raw_x=78, raw_y=0, x=78, y=0, state=PRESSED)
FT6x36(raw_x=80, raw_y=230, x=80, y=230, state=PRESSED)
FT6x36(raw_x=83, raw_y=227, x=83, y=227, state=PRESSED)
FT6x36(raw_x=85, raw_y=224, x=85, y=224, state=PRESSED)
FT6x36(raw_x=88, raw_y=220, x=88, y=220, state=PRESSED)
FT6x36(raw_x=91, raw_y=217, x=91, y=217, state=PRESSED)
FT6x36(raw_x=94, raw_y=213, x=94, y=213, state=PRESSED)
FT6x36(raw_x=98, raw_y=209, x=98, y=209, state=PRESSED)
FT6x36(raw_x=102, raw_y=206, x=102, y=206, state=PRESSED)
FT6x36(raw_x=107, raw_y=202, x=107, y=202, state=PRESSED)
FT6x36(raw_x=111, raw_y=198, x=111, y=198, state=PRESSED)
FT6x36(raw_x=116, raw_y=195, x=116, y=195, state=PRESSED)
FT6x36(raw_x=120, raw_y=191, x=120, y=191, state=PRESSED)
FT6x36(raw_x=124, raw_y=189, x=124, y=189, state=PRESSED)
FT6x36(raw_x=128, raw_y=186, x=128, y=186, state=PRESSED)
FT6x36(raw_x=131, raw_y=184, x=131, y=184, state=PRESSED)
FT6x36(raw_x=133, raw_y=183, x=133, y=183, state=PRESSED)
FT6x36(raw_x=136, raw_y=181, x=136, y=181, state=PRESSED)
FT6x36(raw_x=138, raw_y=180, x=138, y=180, state=PRESSED)
FT6x36(raw_x=140, raw_y=179, x=140, y=179, state=PRESSED)
FT6x36(raw_x=142, raw_y=178, x=142, y=178, state=PRESSED)
FT6x36(raw_x=143, raw_y=178, x=143, y=178, state=PRESSED)
FT6x36(raw_x=144, raw_y=178, x=144, y=178, state=PRESSED)
FT6x36(raw_x=146, raw_y=177, x=146, y=177, state=PRESSED)
FT6x36(raw_x=147, raw_y=177, x=147, y=177, state=PRESSED)
FT6x36(raw_x=148, raw_y=177, x=148, y=177, state=PRESSED)
FT6x36(raw_x=149, raw_y=177, x=149, y=177, state=PRESSED)
FT6x36(raw_x=149, raw_y=178, x=149, y=178, state=PRESSED)
FT6x36(raw_x=148, raw_y=178, x=148, y=178, state=PRESSED)
FT6x36(raw_x=147, raw_y=179, x=147, y=179, state=PRESSED)
FT6x36(raw_x=145, raw_y=179, x=145, y=179, state=PRESSED)
FT6x36(raw_x=143, raw_y=180, x=143, y=180, state=PRESSED)
FT6x36(raw_x=141, raw_y=181, x=141, y=181, state=PRESSED)
FT6x36(raw_x=139, raw_y=182, x=139, y=182, state=PRESSED)
FT6x36(raw_x=136, raw_y=183, x=136, y=183, state=PRESSED)
FT6x36(raw_x=133, raw_y=184, x=133, y=184, state=PRESSED)
FT6x36(raw_x=129, raw_y=186, x=129, y=186, state=PRESSED)
FT6x36(raw_x=125, raw_y=188, x=125, y=188, state=PRESSED)
FT6x36(raw_x=121, raw_y=191, x=121, y=191, state=PRESSED)
FT6x36(raw_x=117, raw_y=194, x=117, y=194, state=PRESSED)
FT6x36(raw_x=113, raw_y=197, x=113, y=197, state=PRESSED)
FT6x36(raw_x=109, raw_y=201, x=109, y=201, state=PRESSED)
FT6x36(raw_x=106, raw_y=205, x=106, y=205, state=PRESSED)
FT6x36(raw_x=103, raw_y=208, x=103, y=208, state=PRESSED)
FT6x36(raw_x=101, raw_y=211, x=101, y=211, state=PRESSED)
FT6x36(raw_x=100, raw_y=214, x=100, y=214, state=PRESSED)
FT6x36(raw_x=99, raw_y=217, x=99, y=217, state=PRESSED)
FT6x36(raw_x=99, raw_y=221, x=99, y=221, state=PRESSED)
FT6x36(raw_x=99, raw_y=225, x=99, y=225, state=PRESSED)
FT6x36(raw_x=99, raw_y=229, x=99, y=229, state=PRESSED)
FT6x36(raw_x=100, raw_y=234, x=100, y=234, state=PRESSED)
FT6x36(raw_x=101, raw_y=239, x=101, y=239, state=PRESSED)
FT6x36(raw_x=102, raw_y=244, x=102, y=244, state=PRESSED)
FT6x36(raw_x=104, raw_y=249, x=104, y=249, state=PRESSED)
FT6x36(raw_x=107, raw_y=254, x=107, y=254, state=PRESSED)
FT6x36(raw_x=110, raw_y=259, x=110, y=259, state=PRESSED)
FT6x36(raw_x=114, raw_y=264, x=114, y=264, state=PRESSED)
FT6x36(raw_x=118, raw_y=269, x=118, y=269, state=PRESSED)
FT6x36(raw_x=122, raw_y=273, x=122, y=273, state=PRESSED)
FT6x36(raw_x=126, raw_y=277, x=126, y=277, state=PRESSED)
FT6x36(raw_x=129, raw_y=280, x=129, y=280, state=PRESSED)
FT6x36(raw_x=132, raw_y=282, x=132, y=282, state=PRESSED)
FT6x36(raw_x=134, raw_y=284, x=134, y=284, state=PRESSED)
FT6x36(raw_x=137, raw_y=286, x=137, y=286, state=PRESSED)
FT6x36(raw_x=137, raw_y=286, x=137, y=286, state=RELEASED)

```
The one in the front has been calibrated, while this one has not. The one that hasn't been calibrated works fine. The calibrated one throws the above error in all other simple tests.

kdschlosser, 2024-12-17:
I just updated it again, do you mind giving it another go?

center789, 2024-12-17:
```
ll -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -O2 -fmacro-prefix-map=/home/suadmin/lvgl_micropython/lib/micropython/ports/esp32=. -fmacro-prefix-map=/home/suadmin/lvgl_micropython/lib/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -std=gnu17 -DLCD_RGB_OPTIMUM_FB_SIZE=0 -Wno-old-style-declaration -Wno-clobbered -Wno-deprecated-declarations -Wno-missing-field-initializers -DCFG_TUSB_MCU=OPT_MCU_ESP32S3 -DLV_KCONFIG_IGNORE=1 -Wno-unused-function -DMICROPY_FLOAT=1 -MD -MT esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/home/suadmin/lvgl_micropython/ext_mod/lcd_utils/src/binary_float.c.obj -MF esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/home/suadmin/lvgl_micropython/ext_mod/lcd_utils/src/binary_float.c.obj.d -o esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/home/suadmin/lvgl_micropython/ext_mod/lcd_utils/src/binary_float.c.obj -c /home/suadmin/lvgl_micropython/ext_mod/lcd_utils/src/binary_float.c
/home/suadmin/lvgl_micropython/ext_mod/lcd_utils/src/binary_float.c:26:7: error: expected '=', ',', ';', 'asm' or '__attribute__' before ':' token
26 | alphaX: 0.9185965     7
|       ^
[1699/2028] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/__/lv_mp.c.obj                                                            
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/suadmin/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_48622 and /home/suadmin/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_48622
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
```
Of course, I'd be happy to help you with the testing. However, I'm currently facing a compilation error, so I need some time to re-git clone the files.

center789, 2024-12-17:
```
Touch Device ID: 0x11
Touch Chip ID: 0x64
Touch Device mode: 0x00
Touch Firmware ID: 0x03
Touch Release code: 0x01
Traceback (most recent call last):
  File "touch_cal_data.py", line 101, in __init__
OSError: (-4354, 'ESP_ERR_NVS_NOT_FOUND')
>>> FT6x36(raw_x=289, raw_y=431, x=289, y=431, state=PRESSED)
FT6x36(raw_x=289, raw_y=431, x=289, y=431, state=RELEASED)
FT6x36(raw_x=291, raw_y=431, x=291, y=431, state=PRESSED)
FT6x36(raw_x=292, raw_y=431, x=292, y=431, state=PRESSED)
FT6x36(raw_x=292, raw_y=431, x=292, y=431, state=RELEASED)
FT6x36(raw_x=284, raw_y=431, x=284, y=431, state=PRESSED)
FT6x36(raw_x=284, raw_y=431, x=284, y=431, state=RELEASED)
FT6x36(raw_x=56, raw_y=428, x=56, y=428, state=PRESSED)
FT6x36(raw_x=56, raw_y=428, x=56, y=428, state=RELEASED)
FT6x36(raw_x=72, raw_y=434, x=72, y=434, state=PRESSED)
FT6x36(raw_x=71, raw_y=434, x=71, y=434, state=PRESSED)
FT6x36(raw_x=70, raw_y=434, x=70, y=434, state=PRESSED)
FT6x36(raw_x=70, raw_y=434, x=70, y=434, state=RELEASED)
FT6x36(raw_x=65, raw_y=433, x=65, y=433, state=PRESSED)
FT6x36(raw_x=65, raw_y=433, x=65, y=433, state=RELEASED)
FT6x36(raw_x=53, raw_y=431, x=53, y=431, state=PRESSED)
FT6x36(raw_x=53, raw_y=431, x=53, y=431, state=RELEASED)
FT6x36(raw_x=49, raw_y=42, x=49, y=42, state=PRESSED)
FT6x36(raw_x=49, raw_y=42, x=49, y=42, state=RELEASED)
FT6x36(raw_x=61, raw_y=34, x=61, y=34, state=PRESSED)
FT6x36(raw_x=61, raw_y=34, x=61, y=34, state=RELEASED)
FT6x36(raw_x=49, raw_y=36, x=49, y=36, state=PRESSED)
FT6x36(raw_x=49, raw_y=36, x=49, y=36, state=RELEASED)
FT6x36(raw_x=46, raw_y=43, x=46, y=43, state=PRESSED)
FT6x36(raw_x=46, raw_y=43, x=46, y=43, state=RELEASED)
FT6x36(raw_x=227, raw_y=255, x=227, y=255, state=PRESSED)
FT6x36(raw_x=227, raw_y=254, x=227, y=254, state=PRESSED)
FT6x36(raw_x=227, raw_y=253, x=227, y=253, state=PRESSED)
FT6x36(raw_x=227, raw_y=249, x=227, y=249, state=PRESSED)
FT6x36(raw_x=227, raw_y=245, x=227, y=245, state=PRESSED)
FT6x36(raw_x=226, raw_y=233, x=226, y=233, state=PRESSED)
FT6x36(raw_x=224, raw_y=226, x=224, y=226, state=PRESSED)
FT6x36(raw_x=218, raw_y=211, x=218, y=211, state=PRESSED)
FT6x36(raw_x=215, raw_y=203, x=215, y=203, state=PRESSED)
FT6x36(raw_x=205, raw_y=186, x=205, y=186, state=PRESSED)
```

```
Touch Device ID: 0x11
Touch Chip ID: 0x64
Touch Device mode: 0x00
Touch Firmware ID: 0x03
Touch Release code: 0x01
Traceback (most recent call last):
  File "touch_cal_data.py", line 101, in __init__
OSError: (-4354, 'ESP_ERR_NVS_NOT_FOUND')
point 1 of 3
   1 of 8: (23, 32)
   2 of 8: (23, 32)
   3 of 8: (23, 32)
   4 of 8: (23, 32)
   5 of 8: (23, 32)
   6 of 8: (23, 32)
   7 of 8: (23, 32)
   8 of 8: (23, 32)
point 2 of 3
   1 of 8: (296, 16)
   2 of 8: (296, 16)
   3 of 8: (296, 16)
   4 of 8: (296, 16)
   5 of 8: (296, 16)
[... 129 lines cut ...]
```
Touch Device ID: 0x11
Touch Chip ID: 0x64
Touch Device mode: 0x00
Touch Firmware ID: 0x03
Touch Release code: 0x01
>>> FT6x36(raw_x=30, raw_y=36, x=37, y=35, state=PRESSED)
FT6x36(raw_x=30, raw_y=36, x=37, y=35, state=RELEASED)
FT6x36(raw_x=267, raw_y=27, x=262, y=39, state=PRESSED)
FT6x36(raw_x=267, raw_y=27, x=262, y=39, state=RELEASED)
FT6x36(raw_x=288, raw_y=432, x=285, y=463, state=PRESSED)
FT6x36(raw_x=286, raw_y=432, x=283, y=463, state=PRESSED)
FT6x36(raw_x=286, raw_y=432, x=283, y=463, state=RELEASED)
FT6x36(raw_x=71, raw_y=235, x=77, y=245, state=PRESSED)
FT6x36(raw_x=71, raw_y=233, x=77, y=243, state=PRESSED)
FT6x36(raw_x=71, raw_y=229, x=77, y=239, state=PRESSED)
FT6x36(raw_x=71, raw_y=226, x=77, y=236, state=PRESSED)
FT6x36(raw_x=71, raw_y=222, x=77, y=231, state=PRESSED)
FT6x36(raw_x=72, raw_y=218, x=78, y=227, state=PRESSED)
FT6x36(raw_x=73, raw_y=216, x=79, y=225, state=PRESSED)
FT6x36(raw_x=73, raw_y=214, x=79, y=223, state=PRESSED)
FT6x36(raw_x=73, raw_y=215, x=79, y=224, state=PRESSED)
FT6x36(raw_x=73, raw_y=215, x=79, y=224, state=RELEASED)
```
OK.It looks like you've successfully solved the problem. The first one is before calibration, the second is during calibration, and the third is after calibration. Everything is running perfectly fine now.

kdschlosser, 2024-12-17:
ok sweet. so the saving of the calibration data is working properly now. All I have to do is remove the exception printout.

Thanks for the assistance with this. It has been a HUGE help.

closed 2024-12-17 by kdschlosser