lvgl-micropython/lvgl_micropython #228: Build fails since two days after fresh clone

hemonu, 2024-12-17:
Since at least two days I'm no more able to build last version. I did a fresh clone of the github and tried to build for my various boards, but I always end with errror  message following:

```shell
Including User C Module(s) from ../../../../../ext_mod/micropython.cmake
-- Configuring incomplete, errors occurred!
See also "/home/hermoers/projects/esp32/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/CMakeFiles/CMakeOutput.log".
Traceback (most recent call last):
File "/home/hermoers/projects/esp32/lvgl_micropython/gen/python_api_gen_mpy.py", line 137, in <module>
json_ast = fixed_gen_json.run(lv_config_path, os.path.join(project_path, 'build', 'lvgl_header.h'), False)
File "/home/hermoers/projects/esp32/lvgl_micropython/gen/fixed_gen_json.py", line 115, in run
ast.setup_docs(temp_directory)
File "/home/hermoers/projects/esp32/lvgl_micropython/lib/lvgl/scripts/gen_json/pycparser_monkeypatch.py", line 589, in setup_docs
docs = doc_builder.XMLSearch(temp_directory)
File "/home/hermoers/projects/esp32/lvgl_micropython/lib/lvgl/docs/doc_builder.py", line 1230, in __init__
sys.stderr.write(err)
TypeError: write() argument must be str, not bytes
OUTPUT:
RESULT: 1
CMake Error at /home/hermoers/projects/esp32/lvgl_micropython/ext_mod/lvgl/micropython.cmake:43 (message):
Failed to generate
/home/hermoers/projects/esp32/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/lv_mp.c
Call Stack (most recent call first):
/home/hermoers/projects/esp32/lvgl_micropython/ext_mod/micropython.cmake:6 (include)
/home/hermoers/projects/esp32/lvgl_micropython/lib/micropython/py/usermod.cmake:42 (include)
esp32_common.cmake:22 (include)
main_esp32s3/CMakeLists.txt:11 (include)
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: Leaving directory '/home/hermoers/projects/esp32/lvgl_micropython/lib/micropython/ports/esp32'
cmake failed with exit code 1, output of the command is in the /home/hermoers/projects/esp32/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_5608 and /home/hermoers/projects/esp32/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_5608
make: *** [Makefile:66: all] Error 1
```

I'm compiling for three different ESP32S3 boards. Same problem for all of them.

I'm building in WSL Ubuntu on a Windoes 11 machine

Last Build Command for a Waveshare ESP32-S§-Touch-LCD-7:

```shell
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=8 --usb-otg DISPLAY=rgb_display INDEV=gt911
```

0ut4t1m3, 2024-12-17:
doxygen has been added as a build dependency, `sudo apt install doxygen` should get you working again.

kdschlosser, 2024-12-17:
That is correct. I need to make a change upstream in LVGL to remove that dependency. I am going to submit a PR for the change shortly. It will also speed up the amount of time it takes to compile

kdschlosser, 2024-12-17:
Here is the PR in LVGL to get rid of the doxygen dependency.

https://github.com/lvgl/lvgl/pull/7471

closed 2025-01-05 by kdschlosser