lvgl-micropython/lvgl_micropython #232: Reclone current repo after delete the old one, rebuild, still got LVGL MicroPython 1.23.0 on 2024-12-19 after flashing

tetopik, 2024-12-20:
Why is rebuild latest repo still got me mpy version 1.23 instead of 1.24.1?
What do am i missing here?

LVGL is updated to 9.2.2, but not the mpy version.
The build console does outputting `collecting MicroPython 1.24.1`.

h-milz, 2024-12-20:
I believe the LVGL Micropython version string is fixed...

tetopik, 2024-12-20:
Right, my bad, `os.uname()` returns 1.24.1

Thanks!

closed 2024-12-20 by tetopik

kdschlosser, 2024-12-20:
I need to update the version string as I am overriding it in the build. I forgot about updating that.

tetopik, 2024-12-20:
That would be great Kevin, keep up the good work

h-milz, 2024-12-20:
I was about to suggest to create the version string automatically by parsing the mpy and lvgl version files ...

tetopik, 2024-12-21:
> I need to update the version string as I am overriding it in the build.

Just want to let you know @kdschlosser that current `Fixes current strings' patch failed to build.

```
[1276/1968] Building C object esp-idf/main_esp32/CMakeFiles/__idf_main_esp32.dir/__/panichandler.c.obj                                                                
FAILED: esp-idf/main_esp32/CMakeFiles/__idf_main_esp32.dir/__/panichandler.c.obj
/home/tetopik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc -DBERKELEY_DB_CONFIG_FILE=\"/home/tetopik/micropython/lvgl_micropython/lib/micropython/extmod/berkeley-db/berkeley_db_config_port.h\" -DESP_PLATFORM -DFFCONF_H=\"/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/oofatfs/ffconf.h\" -DIDF_VER=\"c9763f62\" -DLFS1_NO_ASSERT -DLFS1_NO_DEBUG -DLFS1_NO_ERROR -DLFS1_NO_MALLOC -DLFS1_NO_WARN -DLFS2_NO_ASSERT -DLFS2_NO_DEBUG -DLFS2_NO_ERROR -DLFS2_NO_MALLOC -DLFS2_NO_WARN -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMICROPY_ESP_IDF_4=1 -DMICROPY_MODULE_FROZEN_MPY="(1)" -DMICROPY_PY_BTREE=1 -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_VFS_FAT=1 -DMICROPY_VFS_LFS2=1 -DNDEBUG -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/config -I/home/tetopik/micropython/lvgl_micropython/lib/micropython -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/berkeley-db-1.xx/include -I/home/tetopik/micropython/lvgl_micropython/ext_mod/lcd_bus -I/home/tetopik/micropython/lvgl_micropython/ext_mod/lcd_bus/esp32_include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_lcd/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_lcd/interface -I/home/tetopik/micropython/lvgl_micropython/lib -I/home/tetopik/micropython/lvgl_micropython/lib/lvgl -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32 -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/boards/ESP32_GENERIC -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/newlib/platform_include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/config/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/config/include/freertos -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/config/xtensa/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/esp_additions/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include/soc -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/port/esp32/. -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/heap/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/log/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/soc/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/soc/esp32 -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/soc/esp32/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/hal/platform_port/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/hal/esp32/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/hal/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_rom/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_rom/include/esp32 -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32 -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_common/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_system/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_system/port/soc -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_system/port/include/private -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/xtensa/esp32/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/xtensa/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/xtensa/deprecated_include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/lwip/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/lwip/include/apps -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/lwip/include/apps/sntp -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/lwip/lwip/src/include -I/home/tet [... cut ...]
ninja failed with exit code 1, output of the command is in the /home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/log/idf_py_stderr_output_6460 and /home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/log/idf_py_stdout_output_6460
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: Leaving directory '/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32'
make: *** [Makefile:71: all] Error 1
```

kdschlosser, 2024-12-21:
give it a try now. I just made a change that should fix the problem.

tetopik, 2024-12-21:
> give it a try now. I just made a change that should fix the problem.

It works!