lvgl-micropython/lvgl_micropython #233: Build error

Bigmaxmafia, 2024-12-23:
Hi. [5fad2fc](https://github.com/lvgl-micropython/lvgl_micropython/commit/5fad2fc82e0636db5cd234099b6cf3e8466af7ae)

```shell
-- Configuring incomplete, errors occurred!
See also "/home/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC-SPIRAM/CMakeFiles/CMakeOutput.log".
Traceback (most recent call last):
File "/home/lvgl_micropython/gen/python_api_gen_mpy.py", line 137, in <module>
json_ast = fixed_gen_json.run(lv_config_path, os.path.join(project_path, 'build', 'lvgl_header.h'), False)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/lvgl_micropython/gen/fixed_gen_json.py", line 115, in run
ast.setup_docs(temp_directory)
File "/home/lvgl_micropython/lib/lvgl/scripts/gen_json/pycparser_monkeypatch.py", line 589, in setup_docs
docs = doc_builder.XMLSearch(temp_directory)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/lvgl_micropython/lib/lvgl/docs/doc_builder.py", line 1230, in __init__
sys.stderr.write(err)
TypeError: write() argument must be str, not bytes
OUTPUT:
RESULT: 1
CMake Error at /home/lvgl_micropython/ext_mod/lvgl/micropython.cmake:43 (message):
Failed to generate
/home/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC-SPIRAM/lv_mp.c
Call Stack (most recent call first):
/home/lvgl_micropython/ext_mod/micropython.cmake:6 (include)
/home/lvgl_micropython/lib/micropython/py/usermod.cmake:46 (include)
esp32_common.cmake:22 (include)
main_esp32/CMakeLists.txt:11 (include)
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: Leaving directory '/home/lvgl_micropython/lib/micropython/ports/esp32'
cmake failed with exit code 1, output of the command is in the /home/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC-SPIRAM/log/idf_py_stderr_output_3367 and /home/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC-SPIRAM/log/idf_py_stdout_output_3367
make: *** [Makefile:71: all] Error 1

```

```shell
python3 make.py esp32 BOARD=ESP32_GENERIC BOARD_VARIANT=SPIRAM DISPLAY=ili9341 INDEV=xpt2046
```

kdschlosser, 2024-12-23:
you need to install doxygen.

kdschlosser, 2024-12-23:
I just pushed a commit that fixes the issue with doxygen needing to be installed, It is now no longer needed.

Bigmaxmafia, 2024-12-23:
@kdschlosser good job!

closed 2024-12-23 by Bigmaxmafia