lvgl-micropython/lvgl_micropython #235: esp32 build error

bh1rio, 2024-12-24:
just run 'python make.py esp32'
and error info like:

[7/8] Completed 'bootloader'
Project build complete. To flash, run:
idf.py flash
or
idf.py -p PORT flash
or
python -m esptool --chip esp32 -b 460800 --before default_reset --after no_reset write_flash --flash_mode dio --flash_size 4MB --flash_freq 40m 0x1000 build-ESP32_GENERIC/bootloader/bootloader.bin 0x8000 build-ESP32_GENERIC/partition_table/partition-table.bin 0x10000 build-ESP32_GENERIC/micropython.bin
or from the "/root/temp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC" directory
python -m esptool --chip esp32 -b 460800 --before default_reset --after no_reset write_flash "@flash_args"
bootloader  @0x001000    23264  (    5408 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  2460608  (    1088 remaining)
total                  2526144
make: Leaving directory '/root/temp/lvgl_micropython/lib/micropython/ports/esp32'

Traceback (most recent call last):
  File "/root/temp/lvgl_micropython/make.py", line 239, in <module>
    mod.compile(*extra_args)
    ~~~~~~~~~~~^^^^^^^^^^^^^
  File "/root/temp/lvgl_micropython/builder/esp32.py", line 1333, in compile
    raise RuntimeError(
        'unable to locate python version used in the ESP-IDF'
    )
RuntimeError: unable to locate python version used in the ESP-IDF
[root@t-02 lvgl_micropython]#

hemonu, 2024-12-29:
I had the same problem, when I tried to compile on my new MacBook. Problem was the version of python3. In line 1325 of builder/esp32.py versions 3.8 to 3.12 are hard-coded. 
Check with 'python3 --version' on your machine. If it's 3.13 as on my MacBook, just add that version in line 1325. After that the build ran without any problem.

bh1rio, 2024-12-29:
> I had the same problem, when I tried to compile on my new MacBook. Problem was the version of python3. In line 1325 of builder/esp32.py versions 3.8 to 3.12 are hard-coded. Check with 'python3 [...]

I've given up on using lvgl with Micropython. Because I found that lvgl has turned the entire esp-idf into a Micropython module, so I decided to abandon lvgl_micropython.

 It's just too ugly!Python should be elegant.

closed 2024-12-29 by bh1rio

kdschlosser, 2024-12-29:
ummm... there is only one part of the espidf that has been made into a kind of a module and that is the esp-lcd portion of the espidf. so I am not sure why there is some kind of an issue with that because if that piece didn't exist you would not be able to use the display. There is also a whole lot more to it than that. There are drivers that are already coded into it for quite a few displays and also for touch input. There is a code that reads the LVGL header files to create the binding layer that allows LVGL to be accessed from Micropython,

You also need to remember that you are running a high level programming language in something that has as little as 300K of available memory and less than 1mb of "hdd" space. To be able to do that is a very large feat and to add in a GUI framework omn top of that..

And your comment about it turning the ESP-IDF into a module... Well if that isn't done you would have zero access to any of the hardware. Those modules are the drivers like what you see on a desktop PC. Running CPython is a wrapper around an OS's set of libraries. So the same thing can be said about CPython as well. It just turns the OS's SDK into modules. No different then what is done in MicroPython.

ancebfer, 2025-01-04:
I'm getting the same "unable to locate python version used in the ESP-IDF" error after build is completed.

The problem is that "esp32.py" builder assumes that espressif path is at "~/espressif" but, in my case, this isn't true.

I'm using the docker "espressif/idf" [image](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-docker-image.html) as development environment and real paths are:

```
ESP_IDF_VERSION=5.2
ESP_ROM_ELF_DIR=/opt/esp/tools/esp-rom-elfs/20230320/
IDF_CCACHE_ENABLE=1
IDF_DEACTIVATE_FILE_PATH=/tmp/tmpzn9n2sv7idf_1
IDF_PATH=/opt/esp/idf
IDF_PYTHON_CHECK_CONSTRAINTS=no
IDF_PYTHON_ENV_PATH=/opt/esp/python_env/idf5.2_py3.10_env
IDF_TOOLS_EXPORT_CMD=/opt/esp/idf/export.sh
IDF_TOOLS_INSTALL_CMD=/opt/esp/idf/install.sh
IDF_TOOLS_PATH=/opt/esp
```

Therefore, linking the assumed path to the real path before building solves de problem:

`ln -s "$IDF_TOOLS_PATH" ~/.espressif`

kdschlosser, 2025-01-05:
if the `IDF_PYTHON_ENV_PATH` variable was used consistently I could use that. I have to check but for some reason I do not believe it is.