lvgl-micropython/lvgl_micropython #237: Font library file with bin suffix crashes

wfclhuhu, 2024-12-31:
I used this to compile into firmware, but when I tried to load the font file, the system crashed. The script is: try: with open("./font/ze51.bin", "rb") as f: font_data = f.read() self.chinese_font = lv.binfont_create(font_data) The system crashed when this was executed. When using a TTF - format font library file, I couldn't find an API that supports this TTF font library. Could you give me some guidance? Thank you.

kdschlosser, 2025-01-01:
binfont_create only takes a file path so you need to load the file system driver and pass a path to the binfont_create function.

wfclhuhu, 2025-01-03:
chinese_font = lv.binfont_create("./font/ze5.bin")， Why does loading also crash？

kdschlosser, 2025-01-03:
```
import lvgl as lv
import fs_driver

fs_drv = lv.fs_drv_t()

fs_driver.fs_register(fs_drv, 'S')

chinese_font = lv.binfont_create("S:font/ze5.bin")
```

wfclhuhu, 2025-01-03:
When running to the statement fs_deriver.fs_register (fs_derv,'S'), it crashes. Below is the specific information about my Python IDE:>>> %Run -c $EDITOR_CONTENT

MPY: soft reboot
run fs_drv 

A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

LVGL MPY version : 1.23.0 on 2024-12-02
IDF version : 11eaf41b
Machine     : Generic ESP32S3 module with Octal-SPIRAM with ESP32S3

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42073645  PS      : 0x00060a30  A0      : 0x82073379  A1      : 0x3fcbe2d0  
A2      : 0x00000174  A3      : 0x3c1dc098  A4      : 0x00000001  A5      : 0x3c254098  
A6      : 0x00000001  A7      : 0x3fcbe2d0  A8      : 0x00000174  A9      : 0x3c1dc098  
A10     : 0x3c253f00  A11     : 0x00005922  A12     : 0x00000001  A13     : 0x3c253f64  
A14     : 0x00000001  A15     : 0x3fcbe290  SAR     : 0x00000002  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000174  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x42073642:0x3fcbe2d0 0x42073376:0x3fcbe310 0x42026b2a:0x3fcbe350 0x42006a8d:0x3fcbe390 0x420b8ed5:0x3fcbe3d0 0x420b8f9d:0x3fcbe3f0 0x40379ad2:0x3fcbe410 0x420b233f:0x3fcbe4b0 0x420b8ed5:0x3fcbe510 0x420b8f9d:0x3fcbe530 0x40379ad2:0x3fcbe550 0x420b233f:0x3fcbe5f0 0x420b8ed5:0x3fcbe650 0x420b8f9d:0x3fcbe670 0x420b7bb1:0x3fcbe690 0x420b7a19:0x3fcbe6f0 0x420b8ed5:0x3fcbe710 0x40379a2f:0x3fcbe730 0x420b233f:0x3fcbe7d0 0x420b8ed5:0x3fcbe820 0x420b8eea:0x3fcbe840 0x420c6e77:0x3fcbe860 0x420c6fba:0x3fcbe8f0 0x420ce652:0x3fcbe940

ELF file SHA256: f9fb5429b

Rebooting...
���ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382d16
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x105c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbd8
load:0x403cc700,len:0x2e30
entry 0x403c989c
LVGL MicroPython 1.23.0 on 2024-12-02; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
>>>

yusuhua, 2025-01-05:
I also encountered the same problem, maybe there is some problem with dynamic font loading.
Previous versions of lvgl used
```
font = lv.font_load('S:./font/font-PHT-cn-16.bin')
```
and did not encounter this problem.