lvgl-micropython/lvgl_micropython #238: RuntimeError: Incorrect chip id (0xb5)

paksung, 2024-12-31:
**Describe the bug**
When I initialize cst816s.CST816S, an error occurs.

**Expected behavior**
A clear and concise description of what you expected to happen.

**If you are having a problem compiling paste the build output below. (NO SCREENSHOTS)**

```shell
Chip ID: 0xff
Touch version: 255
Proj ID: 0xff
FW Version: 0xff
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "initialize.py", line 76, in <module>
  File "cst816s.py", line 204, in __init__
RuntimeError: Incorrect chip id (0xb5)
```

**If your problem is running the binding please past the code you are using below.***
I need to see ALL of the setup code exactly as it is being ran.

```python
from machine import SPI
from micropython import const

import lcd_bus

_WIDTH = const(240)
_HEIGHT = const(240)

_SPI_HOST = const(1)
_SPI_SCK = const(10)
_SPI_MOSI = const(11)
_SPI_MISO = const(12)

_LCD_FREQ = const(80_000_000)
_LCD_DC = const(8)
_LCD_CS = const(9)
_LCD_RST = const(14)
_LCD_BACKLIGHT = const(2)

spi_bus = SPI.Bus(
    host=_SPI_HOST,
    # _LCD_FREQ,
    mosi=_SPI_MOSI,
    miso=_SPI_MISO,
    sck=_SPI_SCK
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_LCD_DC,
    cs=_LCD_CS
)
try:
    fb1 = display_bus.allocate_framebuffer(int(240 * 240 * 2 / 10), lcd_bus.MEMORY_SPIRAM)
except MemoryError:
    raise RuntimeError('Failing on first buffer')
try:
    fb2 = display_bus.allocate_framebuffer(int(240 * 240 * 2 / 10), lcd_bus.MEMORY_SPIRAM)
except MemoryError:
    raise RuntimeError('Failing on second buffer')

import gc9a01
import lvgl as lv

lv.init()

display = gc9a01.GC9A01(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    frame_buffer1=fb1,
    frame_buffer2=fb2,
    reset_pin=_LCD_RST,
    backlight_pin=_LCD_BACKLIGHT,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

import task_handler  # NOQA
import cst816s  # NOQA

display.set_power(True)
display.init()
display.set_backlight(100)

touch_dev = SPI.Device(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    cs=_LCD_CS
)

indev = cst816s.CST816S(touch_dev)

# print(indev.who_am_i())

th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)
```

paksung, 2024-12-31:
> **Describe the bug** When I initialize cst816s.CST816S, an error occurs.
>
> **Expected behavior** A clear and concise description of what you expected to happen.
>
> **If you are having a problem [...]

my device: ESP32-S3-Touch-LCD-1.28

kdschlosser, 2025-01-01:
That touch IC is an I2C interface not an SPI interface. 

This is the data sheet for that touch IC

https://www.waveshare.com/w/upload/5/51/CST816S_Datasheet_EN.pdf?page=2

under the section titled "Communication interface"

paksung, 2025-01-01:
> That touch IC is an I2C interface not an SPI interface.
> 
> This is the data sheet for that touch IC
> 
> https://www.waveshare.com/w/upload/5/51/CST816S_Datasheet_EN.pdf?page=2
> 
> under the section titled "Communication interface"

I modified the code, but it throws another error. Additionally, I commented out the code related to cst816s to get the program running, but the screen did not display the content as expected.

```
from machine import SPI
from micropython import const

import lcd_bus

_WIDTH = const(240)
_HEIGHT = const(240)

_SPI_HOST = const(1)
_SPI_SCK = const(10)
_SPI_MOSI = const(11)
_SPI_MISO = const(12)

_LCD_FREQ = const(80_000_000)
_LCD_DC = const(8)
_LCD_CS = const(9)
_LCD_RST = const(14)
_LCD_BACKLIGHT = const(2)

spi_bus = SPI.Bus(
    host=_SPI_HOST,
    # _LCD_FREQ,
    mosi=_SPI_MOSI,
    miso=_SPI_MISO,
    sck=_SPI_SCK
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_LCD_DC,
    cs=_LCD_CS
)
try:
    fb1 = display_bus.allocate_framebuffer(int(240 * 240 * 2 / 10), lcd_bus.MEMORY_SPIRAM)
except MemoryError:
    raise RuntimeError('Failing on first buffer')
try:
    fb2 = display_bus.allocate_framebuffer(int(240 * 240 * 2 / 10), lcd_bus.MEMORY_SPIRAM)
except MemoryError:
    raise RuntimeError('Failing on second buffer')

import gc9a01
import lvgl as lv

display = gc9a01.GC9A01(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    frame_buffer1=fb1,
    frame_buffer2=fb2,
    reset_pin=_LCD_RST,
    backlight_pin=_LCD_BACKLIGHT,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

import task_handler  # NOQA
import cst816s  # NOQA
import i2c

display.set_power(True)
display.init()
display.set_backlight(10)

i2c_bus = i2c.I2C.Bus(host=0, scl=7, sda=6)
touch_i2c = i2c.I2C.Device(
    bus=i2c_bus,
    dev_id=cst816s.I2C_ADDR,
    reg_bits=cst816s.BITS
)
indev = cst816s.CST816S(device=touch_i2c)

th = task_handler.TaskHandler()

scr = lv.screen_active()
btn = lv.button(scr)
btn.align(lv.ALIGN.CENTER, 0, 0)
label = lv.label(btn)
label.set_text("Button")
label.center()
```

The error reported in the terminal:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "initialize.py", line 83, in <module>
  File "cst816s.py", line 190, in __init__
  File "cst816s.py", line 160, in _read_reg
  File "i2c.py", line 95, in write_readinto
  File "i2c.py", line 111, in read_mem
  File "i2c.py", line 77, in readfrom_mem_into
OSError: [Errno 19] ENODEV
```

kdschlosser, 2025-01-01:
if you can provide me a link to the datasheet for the display it would be very helpful.

paksung, 2025-01-01:
> if you can provide me a link to the datasheet for the display it would be very helpful.

https://www.waveshare.net/w/upload/5/5e/GC9A01A.pdf

kdschlosser, 2025-01-01:
that's for the display IC not the board you are using.

paksung, 2025-01-01:
> that's for the display IC not the board you are using.

https://www.waveshare.net/w/upload/2/20/ESP32-S3-Touch-LCD-1.28-Sch.pdf
https://www.waveshare.net/w/upload/1/11/Esp32-s3_technical_reference_manual_en.pdf

Are these the ones?
Thank you for taking the time to address this issue.

0ut4t1m3, 2025-01-01:
I have the same board and it's working for me, a couple of things you can try:

my display setup looks like this, adding the reset state should get the screen working for you:
```
display = gc9a01.GC9A01(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    reset_pin=_RST,
    reset_state=gc9a01.STATE_LOW,
    backlight_pin=_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=gc9a01.BYTE_ORDER_BGR,
    rgb565_byte_swap=True
)
```
then to initiliase touch I'm doing:
```
I2C_BUS = i2c.I2C.Bus(host=0, scl=7, sda=6, freq=400000, use_locks=False)
touch_dev = i2c.I2C.Device(bus=I2C_BUS, dev_id=cst816s.I2C_ADDR, reg_bits=cst816s.BITS)
indev = cst816s.CST816S(touch_dev)
```
I don't have this board on the latest build so can't confirm if there's an issue but it did work pre uP v1.24, I'll do a build and see if I can still get it working.

0ut4t1m3, 2025-01-01:
The below is working on the latest build:
```
import lcd_bus
import machine
from micropython import const

# display settings
_WIDTH = const(240)
_HEIGHT = const(240)
_BL = const(2)
_RST = const(14)
_DC = const(8)

_MOSI = const(11)
_MISO = const(12)
_SCK = const(10)
_HOST = const(1)  # SPI2

_LCD_CS = const(9)
_LCD_FREQ = const(80000000)

SPI_BUS = machine.SPI.Bus(host=_HOST,mosi=_MOSI,miso=_MISO,sck=_SCK)

display_bus = lcd_bus.SPIBus(spi_bus=SPI_BUS,freq=_LCD_FREQ,dc=_DC,cs=_LCD_CS)

import gc9a01  # NOQA
import lvgl as lv  # NOQA

lv.init()

display = gc9a01.GC9A01(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    reset_pin=_RST,
    reset_state=gc9a01.STATE_LOW,
    backlight_pin=_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=gc9a01.BYTE_ORDER_BGR,
    rgb565_byte_swap=True
)

display.set_power(True)
display.init()
display.set_backlight(100)

import cst816s  # NOQA
import i2c  # NOQA

I2C_BUS = i2c.I2C.Bus(host=0, scl=7, sda=6, freq=400000, use_locks=False)
touch_dev = i2c.I2C.Device(bus=I2C_BUS, dev_id=cst816s.I2C_ADDR, reg_bits=cst816s.BITS)
indev = cst816s.CST816S(touch_dev)

import task_handler  # NOQA

th = task_handler.TaskHandler()

scrn = lv.screen_active()
#scrn.set_style_bg_color(lv.color_hex(0xff0000), 0)

slider = lv.slider(scrn)
slider.set_width(200)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)
```

paksung, 2025-01-01:
> The below is working on the latest build:
>
> ```
> import lcd_bus
> import machine
> from micropython import const
>
>
> # display settings
> _WIDTH = const(240)
> _HEIGHT = const(240)
> _BL = const(2)
> _RST = [...]

My screen is displaying correctly now! Thank you for your help. But the touch driver still gives errors. Could you send me the commands you used to build the firmware?
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "initialize.py", line 83, in <module>
  File "cst816s.py", line 190, in __init__
  File "cst816s.py", line 160, in _read_reg
  File "i2c.py", line 95, in write_readinto
  File "i2c.py", line 111, in read_mem
  File "i2c.py", line 77, in readfrom_mem_into
OSError: [Errno 19] ENODEV
```

0ut4t1m3, 2025-01-01:
I built it with this command:
`python3 make.py esp32 BOARD=ESP32_GENERIC_S3 DISPLAY=gc9a01 INDEV=cst816s --flash-size=16`
Don't forget to wipe and re-clone the repo if you're not too by the way.

paksung, 2025-01-01:
> I built it with this command: `python3 make.py esp32 BOARD=ESP32_GENERIC_S3 DISPLAY=gc9a01 INDEV=cst816s --flash-size=16` Don't forget to wipe and re-clone the repo if you're not too by the way.

It's still the same, there's an issue with the touch driver. But thanks for your help anyway.

0ut4t1m3, 2025-01-01:
The error you are getting is hinting that there was no response from the touch chip, do you have anything else connected to pins 6 and 7 and can you see the chip if you run a scan on the i2c bus?

paksung, 2025-01-01:
> The error you are getting is hinting that there was no response from the touch chip, do you have anything else connected to pins 6 and 7 and can you see the chip if you run a scan on the i2c bus?

[image]

The hardware is properly connected, and I can successfully obtain data using other CST816S drivers.

[image]

0ut4t1m3, 2025-01-01:
do you get the same issue if you put this somewhere above where you init touch?

`tch_rst = machine.Pin(13, machine.Pin.OUT, value=1)`

I just noticed that I don't control the reset pin anywhere so maybe you're just unlucky and it's floating a bit lower than mine. I'm pretty sure Kevin has a way to specify this in the init for touch but the above will work for a quick test while I work out what needs to be added.

paksung, 2025-01-01:
still the same.
```
tch_rst = machine.Pin(13, machine.Pin.OUT, value=1)
I2C_BUS = i2c.I2C.Bus(host=0, scl=7, sda=6, freq=400000, use_locks=False)
touch_dev = i2c.I2C.Device(bus=I2C_BUS, dev_id=cst816s.I2C_ADDR, reg_bits=cst816s.BITS)
indev = cst816s.CST816S(device=touch_dev,reset_pin=tch_rst)
```

paksung, 2025-01-01:
> I don't know what else it could be sorry. You can get this board running so don't give up hope.

At least my display is working properly now, thank you.

kdschlosser, 2025-01-01:
ok so you can pass the reset pin to the touch driver constructor. 

```python
I2C_BUS = i2c.I2C.Bus(host=0, scl=7, sda=6, freq=400000, use_locks=False)
touch_dev = i2c.I2C.Device(bus=I2C_BUS, dev_id=cst816s.I2C_ADDR, reg_bits=cst816s.BITS)
indev = cst816s.CST816S(touch_dev, reset_pin=13)
```

don't pass a pin instance pass the number instead.

The other thing I am not seeing are pull-up resistors for the SDA and SCL lines. It does state that the touch IC can be set to use either internal pullups or open drain but it doesn't go into how that is done.

I have modified the i2c.py file and attached it. 

[i2c.py.txt](https://github.com/user-attachments/files/18286983/i2c.py.txt)

remove the .txt extension and upload it to your ESP.

change the I2c bus initilization so it reads

```python
I2C_BUS = i2c.I2C.Bus(host=0, scl=7, sda=6, freq=400000, use_locks=False, pullup=True)
```

see if that solves the problem.

paksung, 2025-01-02:
> ok so you can pass the reset pin to the touch driver constructor.
>
> ```python
> I2C_BUS = i2c.I2C.Bus(host=0, scl=7, sda=6, freq=400000, use_locks=False)
> touch_dev = i2c.I2C.Device(bus=I2C_BUS, [...]

It should not be a problem with the pull-up resistors.I tried your code, but it's still the same.

I referred to other CST816S drivers and found that resetting the CST816S during initialization and disabling auto-sleep mode allows it to operate normally.

[cst816.py.txt](https://github.com/user-attachments/files/18287578/cst816.py.txt)

 I modified the code based on this driver, and it runs normally now.
```
import cst816s
import i2c

# reset
reset_pin = machine.Pin(13, machine.Pin.OUT)
reset_pin(0)
time.sleep_ms(1)
reset_pin(1)
time.sleep_ms(50)

I2C_BUS = i2c.I2C.Bus(host=0, scl=7, sda=6, freq=400000, use_locks=False)
touch_dev = i2c.I2C.Device(bus=I2C_BUS, dev_id=cst816s.I2C_ADDR, reg_bits=cst816s.BITS)

# disable auto sleep
touch_dev.write(bytes([0xFE, 0x01]))

indev = cst816s.CST816S(device=touch_dev,reset_pin=reset_pin)
```

closed 2025-01-03 by paksung

kdschlosser, 2025-01-04:
I updated the driver to perform the reset and to turn off the automatic sleep.

isZeroA, 2025-04-01:
I encountered the same problem and raised the following error: `RuntimeError：Incorrect chip id （0xb5）`. I am using screen drivers st7789 and cst816s, but following the above steps still does not solve the problem.

Finally, I attempted to remove the following statement from the cst816.spy file:
```
if chip_id != _ ChipIDValue:
    raise RuntimeError(f'Incorrect chip id ({hex(_ChipIDValue)})')
```
Surprisingly, the screen is running normally and the touch is also functioning properly. Although I don't know the reason, I hope it can help others.

(My English proficiency is not good, the above is based on machine translation)

kdschlosser, 2025-07-24:
I just updated the code so it has the other chip id