lvgl-micropython/lvgl_micropython #240: Questions about PNG decoding

center789, 2025-01-01:
When I tried to decode and display a PNG file, there were no errors, but the screen didn’t display it correctly. I thought it might be an issue with my PNG image, so I checked some related examples and used their PNG files as well, but they didn’t display either. Am I using the wrong method?
```
import usys as sys
sys.path.append('') 
import machine 
import lcd_bus
import lvgl as lv
import ili9488

spi_bus = machine.SPI.Bus(
    host=2,  
    mosi=11,  
    miso=-1,  
    sck=12  
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=13,  
    cs=10,  
    freq=60000000  
)

display = ili9488.ILI9488(
    data_bus=display_bus,
    display_width=320,
    display_height=480,
    reset_pin=9,
    reset_state=ili9488.STATE_LOW,
    backlight_pin=15,
    backlight_on_state=ili9488.STATE_PWM,
    color_space=lv.COLOR_FORMAT.RGB888,
    color_byte_order=ili9488.BYTE_ORDER_RGB,
    rgb565_byte_swap=False, 
)

# display.set_color_inversion(True)
display.set_power(True)
display.init()
display.set_backlight(10)
display.set_rotation(1)

lv.init()
scr = lv.screen_active()
try:
    script_path = __file__[:__file__.rfind('/')] if __file__.find('/') >= 0 else '.'
except NameError: 
    script_path = ''
    
with open('%s/png_decoder_test.png' % script_path, 'rb') as f:
  png_data = f.read()
#   print(png_data)

png_image_dsc = lv.image_dsc_t({
    'data_size': len(png_data),
    'data': png_data 
})

image1 = lv.image(scr)
image1.set_src(png_image_dsc)
image1.set_pos(100,50)

# Create an image from a symbol

image2 = lv.image(scr)
image2.set_src(lv.SYMBOL.OK + " Accept")
image2.set_pos(100,200)

# Drag handler

def drag_event_handler(e):
    self = e.get_target_obj()
    indev = lv.indev_get_act()
    vect = lv.point_t()
    indev.get_vect(vect)
    x = self.get_x() + vect.x
    y = self.get_y() + vect.y
    self.set_pos(x, y)

# Register drag handler for images

for image in [image1, image2]:
    image.add_flag(image.FLAG.CLICKABLE)
    image.add_event_cb(drag_event_handler, lv.EVENT.PRESSING, None)
```
[image]

While checking if the PNG decoding feature was enabled, I noticed ```LV_USE_LIBPNG```. I looked it up and saw that it offers more advanced features, but enabling it (setting it to 1) causes a compilation error. Which one is recommended to enable?
```
/*LODEPNG decoder library*/
#define LV_USE_LODEPNG 1

/*PNG decoder(libpng) library*/
#define LV_USE_LIBPNG 0
```

kdschlosser, 2025-01-01:
libpng is an external library. getting it to work would require writing a cmake build script so it can be compiled using the ESP32 C compiler. It's quite an involved process.

LODEPNG is built into LVGL so it gets compiled with LVGL.

I recommend using LVGL's internal file system handling when dealing with PNG files.

```
import lvgl as lv
import fs_driver

fs_drv = lv.fs_drv_t()

fs_driver.fs_register(fs_drv, 'S')

image1 = lv.image(scr)
image1.set_src('S:png_decoder_test.png')
image1.set_pos(100,50)
```

center789, 2025-01-02:
I’ve also tried the methods of fs_driver, but the situation remains the same. I even attempted using binary-format image files, but none of them could be displayed.

kdschlosser, 2025-01-02:
is it only images that don't work? can you create a slider and is that visible?

center789, 2025-01-02:
Yes, only the images don’t work. Creating sliders works fine, and using built-in widgets is also fine. The issue only occurs when displaying custom images.

center789, 2025-01-02:
Happy New Year! Wishing you good health and a cheerful mood every day.

kdschlosser, 2025-01-02:
you don't have the task handler running...

I just noticed that...

if the task handler is not running it's not going to render the display.

The file system driver in LVGL is also pretty damned slow, so if you are dealing with small images or you have enough ram to load an image in it's entirety then I would go this route.

```python
import machine
import lcd_bus
import lvgl as lv
import ili9488

spi_bus = machine.SPI.Bus(
    host=2,
    mosi=11,
    miso=-1,
    sck=12
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=13,
    cs=10,
    freq=60000000
)

display = ili9488.ILI9488(
    data_bus=display_bus,
    display_width=320,
    display_height=480,
    reset_pin=9,
    reset_state=ili9488.STATE_LOW,
    backlight_pin=15,
    backlight_on_state=ili9488.STATE_PWM
)

# display.set_color_inversion(True)
display.set_power(True)
display.init()
display.set_backlight(10)
display.set_rotation(1)

scr = lv.screen_active()

with open('png_decoder_test.png', 'rb') as f:
    png_data = bytearray(f.read())

png_data_mv = memoryview(png_data)

png_image_dsc = lv.image_dsc_t(dict(
    data_size=len(png_data),
    data=png_data_mv
))

image1 = lv.image(scr)
image1.set_src(png_image_dsc)
image1.set_pos(100, 50)

import task_handler

task_handler.TaskHandler()
```

that is going to consume a fairly large amount of memory because if needing to decode the PNG file into RGB data. so you end up having both in in memory when you do that. You can reduce the amount of memory used by rendering the PNG to a buffer using the canvas widget and then collecting the buffer of RGB data and using that to pass to the image widget. You can then delete the loaded PNG data.

center789, 2025-01-02:
Yes, the PNG image is displaying correctly now, thanks for your guidance. I want to learn LVGL to complete some of my own small projects, but I currently find it too difficult because I don't know C programming. I feel really exhausted reading the official documentation, and there are too few example tutorials for LVGL in MicroPython. Right now, it feels like I'm completely guessing. GPT's LVGL knowledge base hasn't been updated to version 9 yet, and many of the interfaces are different and unusable. I think I need to find a better way to learn. If possible, could you provide me with an example of displaying this image on a canvas? Thank you very much.

closed 2025-01-02 by center789

kdschlosser, 2025-01-02:
If you have any questions feel free to ask away. There is the discussion area on this repository where you can ask questions. I am more than willing to help you out with it. 

The canvas stuff is all new in version 9.2.2. it has actually changed from 9.0.0. I am willing to learn how to use it with you because I really don't know how to fully use it myself. I have just a idea of how it works.

learning to use the MicroPython binding for LVGL can be a little bit more challenging because you need to know the best ways to handle things in Python. specifically keeping a close eye on keeping things in scope so they don't get garbage collected. 

If we use PNG images as an example..

```python
def display_image()
    scr = lv.screen_active()

    with open('png_decoder_test.png', 'rb') as f:
        png_data = bytearray(f.read())

    png_data_mv = memoryview(png_data)

    png_image_dsc = lv.image_dsc_t(dict(
        data_size=len(png_data),
        data=png_data_mv
    ))

    image1 = lv.image(scr)
    image1.set_src(png_image_dsc)
    image1.set_pos(100, 50)

display_image()
```

The above code is going to fail. The reason why it fails is because LVGL makes heavy use of pointers. I don't know how versed you are with C code but when you look at the LVGL functions you are going to see this...

this is the LVGL function in C code for creating an image widget. There are 2 things to note with this function. The function takes a single parameter and that parameter is a pointer to an `lv_obj_t` object. That means what you are passing is the memory address to where the `lv_obj_t` object is located. You are not passing the actual `lv_obj_t` object.

What is returns is the memory address to the `lv_obj_t` object that gets created. when you create the image widget or any other widget the widget actually is not created in Python code, it gets created in C code and LVGL keeps reference to it. because of that happening you do not need to worry about keeping reference to it in Python code because that is taken care of in C code.

```c
lv_obj_t * lv_image_create(lv_obj_t * parent);
```

But when we look at the function for setting the PNG to the new widget the function is expecting 2 parameters to be passed to it. Both of the parameters are expecting to get passed memory locations. 

```c
void lv_image_set_src(lv_obj_t * obj, const void * src);
```

The first parameter is actually taken care of for you. think of it as "self"

```c
====> image1.set_src(png_image_dsc)
```

`image1` is `self` and gets passed as the first parameter. the second parameter you have to pass which is `png_image_dsc`. That object doesn't get created in C code You have to create that object in Python code. That is seen here..

```python
    png_image_dsc = lv.image_dsc_t(dict(
        data_size=len(png_data),
        data=png_data_mv
    ))
```

How you know the difference is because of the `_t` seen here ` lv.image_dsc_t`. That is telling you that the object that is being created is the actual object and not a pointer to the object. when you pass that object to `set_src` the binding code collects the memory address to that object and then passes that to the underlying LVGL function `lv_image_set_src`. A pointer is just like it sounds. it points to where something is located and nothing more. So if the function exits and the `png_image_dsc` object goes out of scope meaning it is no longer accessible it gets flagged for deletion because it is no longer able to be accessed in the eyes of Python. when it gets flagged for deletion it will get garbage collected. when it gets garbage collected the pointer becomes invalid. It no longer points to that object because the object has been deleted. It becomes an invalid pointer.

When dealing with images the image data needs to be able to be accessed at all times. it has to stay resident in memory for the time that it is being used.

When we deal with PNG files you have the PNG data from the file. That data is encoded. each time LVGL redraws the image the PNG data gets decoded into the RAW RGB data to be rendered to the frame buffer. that RGB data is apart of the stack and is temporary use. There is only so much stack memory that is available. It is able to be adjusted at compile time so we can make it larger but too much stack is not a good thing because that large amount of memory is always going to be allocated when you may only be using it once in a blue moon. It is better to not have a huge stack allocation as it leaves memory available for use outside the stack. It's a balancing act.

To overcome some of this at the cost of using more heap allocation what we do is decode the PNG image a single time and then use that RAW RGB data as the data to be rendered...

This is pseudo code and has not been tested. It is more than likely not going to work and is to be used as an idea or reference as to what needs to be done. There is more than likely a better more efficient way of doing this which we will figure out. I am learning just as much as you are as LVGL 9.2.2 has changed a lot of things. 

[... 91 lines cut ...]
    del png_data 
    
    # garbage collect to free the resources immediately 
    gc.collect()

    # the draw buffer is just like the image descriptor where it contains a pointer to the data.
    # I would have to check the code in LVGL but I believe that the image descriptor holds a 
    # pointer to the draw buffer and the draw buffer holds a pointer to where the actual image data is.
[... cut ...]  line so the actual buffer data doesn't get 
    # garbage collected when the function exits. 
    return draw_buf, decoded_image 

# call the function to load the PNG and hold references to the RAW RGB data and the image descriptor
png_decoder_test_data, png_decoder_test_image = decode_image('png_decoder_test.png')

image1 = lv.image(scr)
# set the image descriptor to the image. 
image1.set_src(png_decoder_test_image)
image1.set_pos(100, 50)

import task_handler

task_handler.TaskHandler()

```