lvgl-micropython/lvgl_micropython #242: Error compiling esp32s3 target

glueckm, 2025-01-03:
Hi,

When I try to compile the bindings for an ESP32S3 target, even with the supply command from the readme, I get the following error:

Command:
$ python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=st7796 INDEV=gt911

-- Configuring incomplete, errors occurred!
See also "/data/tools/lvgl_mp_esp32/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/CMakeFiles/CMakeOutput.log".
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: Leaving directory '/data/tools/lvgl_mp_esp32/lib/micropython/ports/esp32'
cmake failed with exit code 1, output of the command is in the /data/tools/lvgl_mp_esp32/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_1487003 and /data/tools/lvgl_mp_esp32/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_1487003

And the content of stderr file:
CMake Error at CMakeLists.txt:62 (file):
  file failed to open for reading (No such file or directory):

    /data/tools/lvgl_mp_esp32/lib/micropython/ports/esp32/../../../../build/sdkconfig.board

Including User C Module(s) from ../../../../../ext_mod/micropython.cmake
Found User C Module(s): usermod_lcd_bus, usermod_lvgl, lvgl_interface, usermod_lcd_utils

I'm building on a debian 12.8 box

Thanks
Martin

yusuhua, 2025-01-05:
The latest build error is:
```
/Users/y/Documents/github/lvgl_micropython/lib/micropython/ports/esp32/../../../../micropy_updates/common/mp_i2c_common.h:34:24: error: field 'lock' has incomplete type
34 |         i2c_bus_lock_t lock;
|                        ^~~~
```

kdschlosser, 2025-01-05:
I know, I am working on it now. I hopefully have it fixed with the last commit.

kdschlosser, 2025-01-05:
the OP's error should also be fixed as well..

closed 2025-01-05 by kdschlosser