lvgl-micropython/lvgl_micropython #244: esp32s3 psram not available

yusuhua, 2025-01-04:
My board is https://github.com/Xinyuan-LilyGO/T-CameraPlus-S3 esp32s3 flash=16M psram=8M
I built it using the following command, and everything went fine after compiling and burning.

```shell
python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16 DISPLAY=st7789 INDEV=cst816s
```
But when I run the following command, I find that the psram size is wrong.

```shell
>>> import micropython
>>> micropython.mem_info()
stack: 736 out of 15360
GC: total: 127936, used: 104016, free: 23920, max new split: 49152
 No. of 1-blocks: 1003, 2-blocks: 142, max blk sz: 1837, max free sz: 1270
```
I don't know why.

yusuhua, 2025-01-04:
I found that it was not fixed, and after re-burning it became like this:
```
>>> import micropython
>>> micropython.mem_info()
stack: 736 out of 15360
GC: total: 64000, used: 14128, free: 49872, max new split: 139264
 No. of 1-blocks: 332, 2-blocks: 47, max blk sz: 32, max free sz: 3105
```
I guess the GC total should be 8M.

kdschlosser, 2025-01-04:
at the bottom of of the repo for that board..

https://github.com/Xinyuan-LilyGO/T-CameraPlus-S3

It states

```
Flash Mode | QIO 80MHz
Flash Size | 16MB (128Mb)
PSRAM | QSPI PSRAM
```

The flash is quad SPI, you can tell by the "QIO". the RAM is quad as well because it states "QSPI"
The SPIRAM only really has 2 settings. the SPI connection and the speed. The size is automatically detected so we have no control over that. You do not want to use a `BOARD_VARIANT` because the SPIRAM on your board is quad not octal. the `BOARD_VARIANT` is what controls the settings for the SPIRAM.

You also cannot really use the `micropython.mem_info` to determine the size of the memory. This is because the heap is no longer a single allocated amount. It is split. I believe that GC shows the correct total amount but `micropython.mem_info` returns the amount for the heaps that have been allocated and not the actual amount that is available. 

use `gc.mem_alloc() + gc.mem_free()` to get the total amount that is available.

yusuhua, 2025-01-05:
Thank you very much for your reply, gc.mem_alloc() + gc.mem_free() results in 203264
```
>>> import gc
>>> gc.mem_alloc() + gc.mem_free()
203264
```
So, it is not supported yet, if it was supported I would choose BOARD_VARIANT=SPIRAM instead of SPIRAM_OCT.

kdschlosser, 2025-01-05:
it is supported. I have many boards that have 8mb of PSRAM and 32mb of flash and they work without an issue.

I have gotten some boards that were sold as espressif devkits that turned out to be counterfeit. could never get the ram to work properly much like what is happening to you. You may have gotten a junk counterfeit board.

kdschlosser, 2025-01-05:
what is the error you get when the compilation fails from removing the ` BOARD_VARIANT=SPIRAM_OCT`. That option is not needed to compile the firmware.

yusuhua, 2025-01-05:
After seeing your reply, I tried again and it is indeed supported. Maybe there was something wrong with the command I wrote last time.
Thank you so much for your support, it's great to finally have enough space to customize the fonts.

closed 2025-01-05 by yusuhua