lvgl-micropython/lvgl_micropython #245: feature request

F4HSL-Yves, 2025-01-04:
**Describe the bug**
A clear and concise description of what the bug is.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Images**

If it is a visual issue then take a picture of it or a video and add it here.

**If you are having a problem compiling paste the build output below. (NO SCREENSHOTS)**

```shell
PASTE OUTPUT HERE
```

**If your problem is running the binding please past the code you are using below.***
I need to see ALL of the setup code exactly as it is being ran.

```python
PASTE CODE HERE
```

**Exact make and model number of the MCU that you are compiling for or the firmware is running on. **

* Make:
* Model:

**For ESP32 MCU's The PSRAM and FLASH SPI type, quad SPI or octal SPI.**

* PSRAM: quad OR octal
* FLASH:  quad OR octal

I need to know the OS and OS version of the machine that compiled the binary. If using a VM then I need to know the OS and OS version the VM is running. (WSL == VM)
 - OS: [e.g. iOS]

**Build Command**

```shell
BUILD COMMAND HERE
```

F4HSL-Yves, 2025-01-04:
Hi, thanks for your great work. I use the waveshare ESP32-S3-Touch-LCD-2.8 which use the st7789 for LCD and cts328 for the touch device. Is this touch device will be supported ?
Best regards
Yves
best

F4HSL-Yves, 2025-01-04:
CST328, not cts328, sorry !

kdschlosser, 2025-01-05:
I should be able to add it pretty quickly.

Give me a few hours and I will get it added.

kdschlosser, 2025-01-05:
I added the driver.

F4HSL-Yves, 2025-01-05:
Tnx a lot, i will try, i have to find how to  update your example (hello world with slider) which is working very well without the touch driver.
Regards
Yves

kdschlosser, 2025-01-05:
There is a compile error I am trying to sort out. IDK if the last commit I made fixed it. gotta wait a couple more minutes before I find out.

kdschlosser, 2025-01-05:
It is compiling properly now.

F4HSL-Yves, 2025-01-05:
Fine !

F4HSL-Yves, 2025-01-06:
New firmware compilation ok ! By the way i cannot test the touch driver because i have no idea how to use this driver cst328 of my esp32 lcd touch. Do you know when i can find some examples of micropython code in purpose of testing it just like your example test the screen with the slider and print the text hello world ?

kdschlosser, 2025-01-06:
yeah in the readme.md file in the root of the repo.

F4HSL-Yves, 2025-01-07:
Your example use a touch device with SPI protocol, the CST328 use I2C...So i am a little bit confusing and i don't know how to use your test example with my device :-)

kdschlosser, 2025-01-07:
https://github.com/lvgl-micropython/lvgl_micropython#i8080-display-with-i2c-touch-input

That example shows how to use the I2C touch

F4HSL-Yves, 2025-01-07:
I get an error when executing the test program :

Traceback (most recent call last):
  File "<stdin>", line 53, in <module>
  File "st7789.py", line 58, in __init__
  File "display_driver_framework.py", line 185, in __init__
MemoryError: memory allocation failed, allocating 1009072024 bytes

The error raises when i invoke the indev device : indev = cst328.CST328(touch_dev)
All is working fine without using the indev device
Tnx again for your help !

Here is the full code :

import lcd_bus
from micropython import const
import machine

# display settings
_WIDTH = const(240)
_HEIGHT = const(320)
_BL = const(5)
_RST = const(39)
_DC = const(41)

_MOSI = const(45)
_MISO = const(46)
_SCK = const(40)
_HOST = const(1)  # SPI2

_LCD_CS = const(42)
_LCD_FREQ = const(80000000)

_SDA = const(1)
_SCL = const(3)
_TP_FREQ = const(100000)

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)

# we are going to let the display driver sort out the best freame buffer size and where to allocate it to.
# fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
# fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import st7789  # NOQA
import lvgl as lv  # NOQA
import i2c
import cst328  # NOQA

display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_RGB,
    rgb565_byte_swap=True,
)

import task_handler  # NOQA

display.set_power(True)
display.init()
display.set_backlight(100)

i2c_bus = i2c.I2C.Bus(host=0, scl=_SCL, sda=_SDA, freq=_TP_FREQ, use_locks=False)
touch_dev = i2c.I2C.Device(bus=i2c_bus, dev_id=cst328.I2C_ADDR, reg_bits=cst328.BITS)

indev = cst328.CST328(touch_dev)

if not indev.is_calibrated:
    display.set_backlight(100)
    indev.calibrate()

th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(100, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO YVES!')
label.align(lv.ALIGN.CENTER, 0, -50)

F4HSL-Yves, 2025-01-07:
Tnx Kevin, now another error with your code :

Traceback (most recent call last):
  File "<stdin>", line 38, in <module>
TypeError: can't convert float to int

F4HSL-Yves, 2025-01-07:
Done that to correct the conv error 
fb1 = display_bus.allocate_framebuffer(int(_WIDTH * _HEIGHT * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
fb2 = display_bus.allocate_framebuffer(int(_WIDTH * _HEIGHT * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
but still not working, screen is blurred

F4HSL-Yves, 2025-01-07:
hum, seems the driver cst328 make some problems...did you test it ?

F4HSL-Yves, 2025-01-07:
Here is the error in cst328.py :
Traceback (most recent call last):
  File "<stdin>", line 74, in <module>
  File "cst328.py", line 98, in __init__
  File "cst328.py", line 61, in _read_reg
OSError: [Errno 116] ETIMEDOUT
Pb with I2C  parameters ?

kdschlosser, 2025-01-07:
How am I going to test a touch screen that I do not own? I can't. All I did was add the driver based on what little information there is available for that specific touch screen. So we are going to have to work our way through it to get it to work. So now at least we are no longer getting a memory error and we are actually getting an I2C related Error now. So that's at least a step in the right direction.

F4HSL-Yves, 2025-01-07:
ok Kevin, the memory error was my fault (wrong utilisation of thonny). But now i think the init error is true, i was wondering if all i2C parameters are true (ie adress) could explain the pb to read register ?

kdschlosser, 2025-01-08:
OK with I2C there are actually 2 addresses to talk to the device, one is for read and the other is for write.

Those addresses are derived from a base address and we need to look at it in binary to understand what is happening..

The address 0x1A in binary is `00011010`

I don't recall off hand which one is read and which one is write but for example lets say read works like this...
```
(0x1A << 1) | 0x1
```

and write is like this..
```
(0x1A << 1) | 0x0
```
The read binary address ends up being this.
```
00110101
```

and the write is this
```
00110100
```

wanna guess what the hex value is for the read and write binary?
read is `0x35`
write is `0x34`

That's the same addresses as what you listed.

The I2C address I have set for the device is correct.

F4HSL-Yves, 2025-01-08:
Ok tnx Kevin ! I still get the timeout error, i looked to the demo example from waveshare (i think you know about it)
https://files.waveshare.com/wiki/ESP32-S3-Touch-LCD-2.8/ESP32-S3-Touch-LCD-2.8-Demo.zip
trying to understand the touchdriver stuff written in c , but it is little bit too hard for me :-)
i tried to change the i2C clock value to 400000 as they do in their example but no success, same error
so i don't know what to do...
Anyway it is a game for me even if  i would like to find out this challenge ! so it is not a real problem !
By the way what the slider is supposed to do ? in case of the touch driver working ?
Tnx again for your work !

F4HSL-Yves, 2025-01-08:
Good news and well done Kevin, i have build a new firm with your last code, and....it works perfect !
So i think the issue can be closed
tnx !!

kdschlosser, 2025-01-08:
so everything is good to go now?

closed 2025-01-08 by kdschlosser

kdschlosser, 2025-01-08:
I have to do some more testing with it to see where it is getting mucked up.

F4HSL-Yves, 2025-01-08:
Ok Kevin be free to ask me if you want to test on my device !