lvgl-micropython/lvgl_micropython #246: QSPI PSRAM esp32s3 board screen not working

yusuhua, 2025-01-05:
When I build using this command
```
python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3  --flash-size=16 DISPLAY=st7789 INDEV=cst816s
```
gc.mem_alloc() + gc.mem_free() print 8M, but the screen not working.

The same code, built with the addition of BOARD_VARIANT=SPIRAM_OCT screen works, but the PSRAM size is incorrect（For the reason of "QSPI"）.

board: https://github.com/Xinyuan-LilyGO/T-CameraPlus-S3

```
import lcd_bus
from micropython import const
import machine

# display settings
_WIDTH = const(240)
_HEIGHT = const(240)
_BL = const(46)
_DC = const(45)

_MOSI = const(35)
_MISO = const(-1)
_SCK = const(36)
_HOST = const(1)

_LCD_CS = const(34)
_LCD_FREQ = const(80000000)

_TOUCH_CS = const(18)
_TOUCH_FREQ = const(10000000)

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)

import st7789
import lvgl as lv

display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)

import task_handler
import cst816s
import i2c
import time

display.set_power(True)
display.init()
display.set_backlight(100)

I2C_BUS = i2c.I2C.Bus(host=0, scl=2, sda=1, freq=400000, use_locks=False)
touch_dev = i2c.I2C.Device(bus=I2C_BUS, dev_id=cst816s.I2C_ADDR, reg_bits=cst816s.BITS)

indev = cst816s.CST816S(device=touch_dev, reset_pin=48)

th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(180, 30)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)

```

kdschlosser, 2025-01-05:
delete your clone of the repo and start from scratch and clone it again. I want to make sure that your build is clean and using the latest code.

do not use the BOARD_VARIANT in your build command... There also should only be a single bin file in the build folder unless you have tried multiple times to compile using different build parameters.

yusuhua, 2025-01-05:
I just saw that you submitted the code, deleted the clone, and encountered a build problem.
```
/Users/yusuhua/Documents/github/lvgl_micropython/lib/micropython/ports/esp32/machine_i2c.c: In function 'i2c_bus_deinit_internal':
/Users/yusuhua/Documents/github/lvgl_micropython/lib/micropython/ports/esp32/machine_i2c.c:75:59: error: 'mp_machine_hw_i2c_bus_obj_t' {aka 'struct _mp_machine_hw_i2c_bus_obj_t'} has no member named 'lock'
75 |     #define I2C_BUS_LOCK_ACQUIRE(self) xSemaphoreTake(self->lock.handle, portMAX_DELAY)
|                                                           ^~
/Users/yusuhua/Documents/github/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/semphr.h:279:76: note: in definition of macro 'xSemaphoreTake'
279 | #define xSemaphoreTake( xSemaphore, xBlockTime )    xQueueSemaphoreTake( ( xSemaphore ), ( xBlockTime ) )
|                                                                            ^~~~~~~~~~
/Users/yusuhua/Documents/github/lvgl_micropython/lib/micropython/ports/esp32/machine_i2c.c:114:17: note: in expansion of macro 'I2C_BUS_LOCK_ACQUIRE'
114 |                 I2C_BUS_LOCK_ACQUIRE(self);
|                 ^~~~~~~~~~~~~~~~~~~~
/Users/yusuhua/Documents/github/lvgl_micropython/lib/micropython/ports/esp32/machine_i2c.c:77:59: error: 'mp_machine_hw_i2c_bus_obj_t' {aka 'struct _mp_machine_hw_i2c_bus_obj_t'} has no member named 'lock'
77 |     #define I2C_BUS_LOCK_RELEASE(self) xSemaphoreGive(self->lock.handle)
|                                                           ^~
/Users/yusuhua/Documents/github/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/semphr.h:428:80: note: in definition of macro 'xSemaphoreGive'
428 | #define xSemaphoreGive( xSemaphore )    xQueueGenericSend( ( QueueHandle_t ) ( xSemaphore ), NULL, semGIVE_BLOCK_TIME, queueSEND_TO_BACK )
|                                                                                ^~~~~~~~~~
/Users/yusuhua/Documents/github/lvgl_micropython/lib/micropython/ports/esp32/machine_i2c.c:115:17: note: in expansion of macro 'I2C_BUS_LOCK_RELEASE'
115 |                 I2C_BUS_LOCK_RELEASE(self);
|                 ^~~~~~~~~~~~~~~~~~~~
/Users/yusuhua/Documents/github/lvgl_micropython/lib/micropython/ports/esp32/machine_i2c.c:76:60: error: 'mp_machine_hw_i2c_bus_obj_t' {aka 'struct _mp_machine_hw_i2c_bus_obj_t'} has no member named 'lock'
76 |     #define I2C_BUS_LOCK_DELETE(self) vSemaphoreDelete(self->lock.handle)
|                                                            ^~
/Users/yusuhua/Documents/github/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/semphr.h:1042:77: note: in definition of macro 'vSemaphoreDelete'
1042 | #define vSemaphoreDelete( xSemaphore )    vQueueDelete( ( QueueHandle_t ) ( xSemaphore ) )
|                                                                             ^~~~~~~~~~
/Users/yusuhua/Documents/github/lvgl_micropython/lib/micropython/ports/esp32/machine_i2c.c:116:17: note: in expansion of macro 'I2C_BUS_LOCK_DELETE'
116 |                 I2C_BUS_LOCK_DELETE(self);
|                 ^~~~~~~~~~~~~~~~~~~
/Users/yusuhua/Documents/github/lvgl_micropython/lib/micropython/ports/esp32/machine_i2c.c: In function 'i2c_bus_transfer':
/Users/yusuhua/Documents/github/lvgl_micropython/lib/micropython/ports/esp32/machine_i2c.c:75:59: error: 'mp_machine_hw_i2c_bus_obj_t' {aka 'struct _mp_machine_hw_i2c_bus_obj_t'} has no member named 'lock'
75 |     #define I2C_BUS_LOCK_ACQUIRE(self) xSemaphoreTake(self->lock.handle, portMAX_DELAY)
|                                                           ^~
/Users/yusuhua/Documents/github/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/semphr.h:279:76: note: in definition of macro 'xSemaphoreTake'
279 | #define xSemaphoreTake( xSemaphore, xBlockTime )    xQueueSemaphoreTake( ( xSemaphore ), ( xBlockTime ) )
|                                                                            ^~~~~~~~~~
/Users/yusuhua/Documents/github/lvgl_micropython/lib/micropython/ports/esp32/machine_i2c.c:154:39: note: in expansion of macro 'I2C_BUS_LOCK_ACQUIRE'
154 |             if (self->use_locks == 1) I2C_BUS_LOCK_ACQUIRE(self);
|                                       ^~~~~~~~~~~~~~~~~~~~
/Users/yusuhua/Documents/github/lvgl_micropython/lib/micropython/ports/esp32/machine_i2c.c:77:59: error: 'mp_machine_hw_i2c_bus_obj_t' {aka 'struct _mp_machine_hw_i2c_bus_obj_t'} has no member named 'lock'
77 |     #define I2C_BUS_LOCK_RELEASE(self) xSemaphoreGive(self->lock.handle)
|                                                           ^~
/Users/yusuhua/Documents/github/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/semphr.h:428:80: note: in definition of macro 'xSemaphoreGive'
428 | #define xSemaphoreGive( xSemaphore )    xQueueGenericSend( ( QueueHandle_t ) ( xSemaphore ), NULL, semGIVE_BLOCK_TIME, queueSEND_TO_BACK )
|                                                                                ^~~~~~~~~~
/Users/yusuhua/Documents/github/lvgl_micropython/lib/micropython/ports/esp32/machine_i2c.c:190:39: note: in expansion of macro 'I2C_BUS_LOCK_RELEASE'
190 |             if (self->use_locks == 1) I2C_BUS_LOCK_RELEASE(self);
|                                       ^~~~~~~~~~~~~~~~~~~~
/Users/yusuhua/Documents/github/lvgl_micropython/lib/micropython/ports/esp32/machine_i2c.c: In function 'i2c_bus_print':
[... cut ...]
/Users/yusuhua/Documents/github/lvgl_micropython/lib/micropython/ports/esp32/machine_i2c.c:626:60: error: 'machine_i2c_init_obj' undeclared here (not in a function); did you mean 'machine_hw_i2c_obj'?
626 |         { MP_ROM_QSTR(MP_QSTR_init),           MP_ROM_PTR(&machine_i2c_init_obj)          },
|                                                            ^~~~~~~~~~~~~~~~~~~~
/Users/yusuhua/Documents/github/lvgl_micropython/lib/micropython/py/obj.h:348:24: note: in definition of macro 'MP_ROM_PTR'
348 | #define MP_ROM_PTR(p) (p)
|                        ^
cc1: some warnings being treated as errors

```

yusuhua, 2025-01-05:
Maybe I should try again later, the version without the BOARD_VARIANT screen not working is your last commit on Jan 4, 2025.

kdschlosser, 2025-01-05:
Just hold on for a few. I am ironing out the last few kinks in it right now. I had to shuffle some things around because of an issue with soft resetting. once I get the last couple of things sorted out I will let you know. It should only take me a few more minutes. I am checking right now to see if it compiles properly. once it compiles hopefully it will run without any issue.

I am specifically messing around with the I2C driver.

kdschlosser, 2025-01-05:
If you click on the actions  tab at the top of the repo here you should be able to watch it compile. There are 3 that run each commit. one for unix, one  for macOS and the last is Windows which doesn't do anything currently.

kdschlosser, 2025-01-05:
It is compiling properly now.

yusuhua, 2025-01-05:
I did a fresh git clone to a new directory, the screen still cannot display.

yusuhua, 2025-01-05:
For comparison, I built a firmware with BOARD_VARIANT=SPIRAM_OCT and found that the screen display was fine, but the touch was still the same as without BOARD_VARIANT=SPIRAM_OCT. The last version with BOARD_VARIANT=SPIRAM_OCT yesterday also had no problem with touch.

kdschlosser, 2025-01-05:
do a hard reset and see if the error persists. I don't know if I have everything sorted out yet with the soft resets.

kdschlosser, 2025-01-05:
I think I found the problem. Go ahead and give it another try and see if it works.

yusuhua, 2025-01-06:
Unfortunately that didn't change anything, I already did a hard reboot.

kdschlosser, 2025-01-06:
I am going to have to add some debugging code so we can see what is happening.

kdschlosser, 2025-01-08:
That's because you don't have octal spiram you only have quad spiram.