lvgl-micropython/lvgl_micropython #247: Compile Error for ESP32S3 target

glueckm, 2025-01-05:
Hi,
First of all, thanks for all the work and effort you put into maintaining the bindings for this lvgl. We all really appreciate that!

I did a fresh git clone to a new directory.

when I try to compile for the ESP32S3 target withe the following command 
`python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=ili9488 INDEV=ft6x36 --ota --flash-size=16`
I get this compile error:
```
[ 66%] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/network_lan.c.obj
/data/tools/esp32_lvgl_mp/lib/micropython/ports/esp32/machine_i2c.c: In function 'i2c_bus_make_new':
/data/tools/esp32_lvgl_mp/lib/micropython/ports/esp32/machine_i2c.c:236:25: warning: comparison is always true due to limited range of data type [-Wtype-limits]
236 |         if (!(I2C_NUM_0 <= i2c_id && i2c_id < I2C_NUM_MAX)) {
|                         ^~
/data/tools/esp32_lvgl_mp/lib/micropython/ports/esp32/machine_i2c.c: In function 'i2c_device_write_readinto':
/data/tools/esp32_lvgl_mp/lib/micropython/ports/esp32/machine_i2c.c:429:64: warning: dereferencing 'void *' pointer
429 |             memaddr |= (uint32_t)(((uint8_t *)write_bufinfo.buf[i]) << ((~i + (self->reg_bits / 8)) * 8));
|                                                                ^
/data/tools/esp32_lvgl_mp/lib/micropython/ports/esp32/machine_i2c.c:429:36: error: invalid use of void expression
429 |             memaddr |= (uint32_t)(((uint8_t *)write_bufinfo.buf[i]) << ((~i + (self->reg_bits / 8)) * 8));
|                                    ^
make[3]: *** [esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/build.make:3757: esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/machine_i2c.c.obj] Error 1
```

kdschlosser, 2025-01-05:
I am in the process of fixing the errors from the code I just changed. I should be down to the last error I believe (I hope). The CI is compiling right now and I will know in a few minutes if it compiles properly. Then the testing begins.

I am not too far off from being able to do an official release. The last bunch of changes I made were done to fix a problem with soft reboots of the ESP32. The hardware was not getting reset to it's unused state when doing a soft reset. I have come up with a way to get it to work correctly and what I am just finishing up with is the last piece.

glueckm, 2025-01-05:
Great to hear. Again, thanks for your work and effort!

kdschlosser, 2025-01-05:
no worries m8.

I had to fix something else. it's never ending. This last code update was pretty large and problems rippled out all over the place.

kdschlosser, 2025-01-05:
It is compiling properly now.

glueckm, 2025-01-05:
Yes, it compiles fine now. Thanks!

closed 2025-01-05 by glueckm