lvgl-micropython/lvgl_micropython #248: How to select ILI9341 type 1 or type2 and what's the difference

ste7anste7an, 2025-01-05:
I build the firmware without any issues (I only had to choose a building direcotry in the root of my home folder, because bash had too long pathes).

hardware:
- ESP32-WROVER (4MByte of PSRAM, 4Mbyte FLASH)
- ILI9341 SPI display
- XPT2046 touch on the same SPI bus

```
python3 make.py esp32 BOARD=ESP32_GENERIC BOARD_VARIANT=SPIRAM DISPLAY=ili9341 INDEV=xpt2046
```

I got an error when running the folowing test program:
<details>

<summary>test_lvgl.py</summary>

```python
import lcd_bus
from micropython import const
import machine

# display settings
_WIDTH = const(240)
_HEIGHT = const(320)
_BL = const(33)
_RST = const(32)
_DC = const(27)

_MOSI = const(13)
_MISO = const(12)
_SCK = const(14)
_HOST = const(1)  # SPI2

_LCD_CS = const(15)
_LCD_FREQ = const(40000000)

_TOUCH_CS = const(26)
_TOUCH_FREQ = const(1000000)

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)

# we are going to let the display driver sort out the best freame buffer size and where to allocate it to.
# fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
# fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import ili9341  # NOQA
import lvgl as lv  # NOQA

display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
     reset_pin=_RST,
    reset_state=ili9341.STATE_LOW,
    backlight_pin=_BL,
    backlight_on_state=ili9341.STATE_HIGH,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=ili9341.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)

import task_handler  # NOQA
import xpt2046  # NOQA

display.set_power(True)
display.init()
display.set_rotation(lv.DISPLAY_ROTATION._180)
display.set_backlight(100)

touch_dev = machine.SPI.Device(
    spi_bus=spi_bus,
    freq=_TOUCH_FREQ,
    cs=_TOUCH_CS
)

# I had to rotate the touch relative to the screen, as the touch was mirrored in first instance.
indev = xpt2046.XPT2046(touch_dev,debug=True,startup_rotation=lv.DISPLAY_ROTATION._180)

th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(200, 20)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)
```
</details>

```
_ili9341_init.py not found
```

As a work around, in `~/lvgl_micropython/api_drivers/common_api_drivers/display/ili9341` I copied:
```
cp _ili9341_init_type1.py _ili9341_init.py
```

I have two questions:
1) how can I choose between type1 and type2 of ILI9341? Should that be donde when building, or is it a parameter in my Python code?
2) I do see some differnces in the initialization when comparing`_ili9341_init_type1.py` to `_ili9341_init_type2.py`. But how can I see which version I have?

PS. keep up the good work! It is for the first time I get MicroPython 1.24 working with LVGL. The lv_binding got only to 1.20.

kdschlosser, 2025-01-06:
you need to supply a 1 or a 2 to the `init` method of the display driver. The difference is the config settings that are used to set up the display. I don't know off the top of my head exactly what the differences are. Try type 1 and if it works then that is the one you use. if it doesn't then try type 2. if that doesn't work then let me know and I will do some digging to find out exactly what the parameters are for your specific display.

The config settings that get sent to the display IC are what instruct the display IC on what need to be done to produce an image on the display that the IC is attached to. Stating a display is an ILI9341 display is a misnomer. the display is not an ILI9341 display. the IC that is being used as the bridge between the display and the MCU is an ILI9341. the display can be manufactured by anyone and can have different specs on how data needs to be sent to it in order for it to show a proper image.

ste7anste7an, 2025-01-06:
Thanks, that worked, but I do not see any difference between type1 and type2. 

The colors are very dark, even if I comment the line `scrn.set_style_bg_color(lv.color_hex(0x000000), 0)`. Is there a global way for setting ligher colors, as I got with the old LVGL8: white background and blue buttons?

kdschlosser, 2025-01-07:
You display might not be setting spicific with what is being set in type 1 and type 2....

Can you take a photo of the screen while it has something rendered to it and attach it here so I can see what you mean by dark. Can you also attach a link to who makes the display so I can check and see what the settings are supposed to be for it.

ste7anste7an, 2025-01-08:
This is how my screen looks when I run:
<details>
<summary>test_lvgl.py</summary>

```python
import lcd_bus
from micropython import const
import machine

# display settings
_WIDTH = const(240)
_HEIGHT = const(320)
_BL = const(33)
_RST = const(32)
_DC = const(27)

_MOSI = const(13)
_MISO = const(12)
_SCK = const(14)
_HOST = const(1)  # SPI2

_LCD_CS = const(15)
_LCD_FREQ = const(40000000)

_TOUCH_CS = const(26)
_TOUCH_FREQ = const(1000000)

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)

# we are going to let the display driver sort out the best freame buffer size and where to allocate it to.
# fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
# fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import ili9341  # NOQA
import lvgl as lv  # NOQA

display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
     reset_pin=_RST,
    reset_state=ili9341.STATE_LOW,
    backlight_pin=_BL,
    backlight_on_state=ili9341.STATE_HIGH,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=ili9341.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)

import task_handler  # NOQA
import xpt2046  # NOQA

display.set_power(True)
display.init(1)
display.set_rotation(lv.DISPLAY_ROTATION._180)
display.set_backlight(100)

touch_dev = machine.SPI.Device(
    spi_bus=spi_bus,
    freq=_TOUCH_FREQ,
    cs=_TOUCH_CS
)

indev = xpt2046.XPT2046(touch_dev,debug=True,startup_rotation=lv.DISPLAY_ROTATION._180)

th = task_handler.TaskHandler()

scrn = lv.screen_active()
#scrn.set_style_bg_color(lv.color_hex(0xffffff), 0)

btnm = lv.buttonmatrix(scrn)
btnm.add_event_cb(lambda e: btnm_event_handler(e,scrn),lv.EVENT.VALUE_CHANGED, None)

btn = lv.button(scrn)
btn.center()
btn.set_size(100,50)
o = 1

def btnm_event_handler(e,ta):
    global o
    obj = e.get_target()
    o=obj
    print("Toggled")

```
</details>

[image]

This is the back of my display:
[image]

I believe it is best described here: ([2.4inch_SPI_Module_ILI9341](http://www.lcdwiki.com/2.4inch_SPI_Module_ILI9341_SKU:MSP2402)).

kdschlosser, 2025-01-08:
do me a favor... use this code for the UI stuff and snap a photo of it so I am able to see what it looks like..

```python
# your screen setup code above this
scrn = lv.screen_active()

width = scrn.get_width()
height = scrn.get_height()

center_x = int(width / 2)
center_y = int(height / 2)

obj_width = int(width / 5)
obj_height = int(height / 5)

obj1 = lv.obj(scrn)
obj1.set_pos(center_x - 10 - obj_width, center_y - 10 - obj_height)
obj1.set_size(obj_width, obj_height)
obj1.set_style_bg_color(lv.color_hex(0xFF0000), 0)

obj2 = lv.obj(scrn)
obj2.set_pos(center_x  + 10, center_y - 10 - obj_height)
obj2.set_size(obj_width, obj_height)
obj2.set_style_bg_color(lv.color_hex(0x00FF00), 0)

obj3 = lv.obj(scrn)
obj3.set_pos(center_x  - int(obj_width / 2), center_y + 10)
obj3.set_size(obj_width, obj_height)
obj3.set_style_bg_color(lv.color_hex(0x0000FF), 0)

import task_handler

task_handler.TaskHandler()

```

ste7anste7an, 2025-01-08:
The three basic color looks OK, aren't they?

[image]

This is with
```
scrn.set_style_bg_color(lv.color_hex(0xffffff), 0)
```
[image]

kdschlosser, 2025-01-08:
the colors are inverted...

you need to call `display.set_color_inversion(True)` after you call `display.init(...)`

ste7anste7an, 2025-01-08:
No, I don't think the colors are inverted. If I call `display.set_color_inversion(True)`, Blue becomes Yellow, Gren becomes Purple and Red becomes Turcqoise. So I think, setting the background style to 0xffffff and use the noremal settings, it's just fine. Thanks for your help.

Kalrkloss, 2025-04-11:
I had the same problem with the colors. 
You have to use color_byte_order=ili9341.BYTE_ORDER_RGB, byteswap and display.set_color_inversion(True)

`
import ili9341
display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
     reset_pin=_RST,
    reset_state=ili9341.STATE_LOW,
    backlight_pin=_BL,
    backlight_on_state=ili9341.STATE_HIGH,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=ili9341.BYTE_ORDER_RGB,
    rgb565_byte_swap=True,
)
display.set_power(True)
display.init(1)
display.set_color_inversion(True)
display.set_backlight(100)
`

With this, I get the right colors.