lvgl-micropython/lvgl_micropython #249: Feature request: adds ESP32 TWAI CAN bus driver

tetopik, 2025-01-07:
Hi, Kevin. I see that you were active in some of the MicroPython ESP32 CAN bus forum discussions, and in this repo I know that you're doing great with coding, especially with MicroPython.

So if it's possible, since micropython canbus currently only implemented in `pyb.CAN`, can you make my dream come true to use such as `machine.CAN` module in harmony with this superb lvgl repo on an ESP32 without the need of another module such as MCP2515?

Doing some car projects would be so much fun for so many people due to your great skills and efforts.

tetopik, 2025-01-07:
It's already there in circuitpython as `canio`, but lacks of 'lvgl'.

kdschlosser, 2025-01-07:
I can add the canbus related stuff. it's not that hard to add it.

Once I get the I2C back up and running I will work on adding it.

tetopik, 2025-01-08:
What a big news! Awesome, Kevin. Can't wait to see a big changes around micropython.

This whose time struggling to find a native use of canbus on esp32 running micropython, got some but outdated and only generic esp32 available. 

Big thanks, really appreciate it.

kdschlosser, 2025-01-08:
As it turns out one of the users of this binding has written a CAN extension module for the ESP32... You have to thank @straga for writing it

https://github.com/straga/micropython-esp32-twai

I forked the repo and I am making some changes to it. I will submit a PR for the changes any maybe he will merge them. we will see.

You should be able to use it if you use the following commands.

```
cd ext_mod
git clone https://github.com/straga/micropython-esp32-twai
cd ..
export CFLAGS="-DMODULE_CAN_ENABLED"
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 USER_C_MODULE={/absolute/path/to/lvgl_micropython}/ext_mod/micropython-esp32-twai/src/micropython.cmake
```

replacing `{/absolute/path/to/lvgl_micropython}` with the absolute path to lvgl_micropython

tetopik, 2025-01-08:
> I am making some changes to it. I will submit a PR for the changes any maybe he will merge them.

Should I wait for it or it's already usable right now?

tetopik, 2025-01-08:
Compiling for C3 returns this error:

```
[969/1935] cd /home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/esp-idf/main_esp32c3 && /usr/bin/cmake -E make_directory /home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/genhdr && /home/tetopik/.espressif/python_env/idf5.2_py3.12_env/bin/python /home/tetopik/micropython/lvgl_micropython/lib/micropython/py/makeversionhdr.py /home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/genhdr/mpversion.h
GEN /home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/genhdr/mpversion.h
[972/1935] Building C object esp-idf/espressif__mdns/CMakeFiles/__idf_espressif__mdns.dir/mdns_networki[973/1935] Generating ../../pins.c, ../../genhdr/pins.h                                                [978/1935] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/manag[979/1935] Generating /home/tetopik/micropython/lvgl_micropython/lib/micropython/mpy-cross/build/mpy-cross     
make[1]: Entering directory '/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/esp-idf/main_esp32c3'
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make[1]: Leaving directory '/home/tetopik/micropython/lvgl_micropython/lib/micropython/mpy-cross'
[983/1935] Building C object esp-idf/espressif__mdns/CMakeFiles/__idf_espressif__mdns.dir/mdns_console.[984/1935] Linking C static library esp-idf/esp_rom/libesp_rom.a                                       [1026/1935] Generating ../../genhdr/qstr.i.last                                                        
FAILED: genhdr/qstr.i.last /home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/genhdr/qstr.i.last
/bin/sh esp-idf/main_esp32c3/CMakeFiles/qstr.i.last-93c470f.sh 8b4b281f05216edb
/home/tetopik/micropython/lvgl_micropython/ext_mod/micropython-esp32-twai/src/modcan.c:39:10: fatal error: soc/dport_reg.h: No such file or directory
39 | #include "soc/dport_reg.h"
|          ^~~~~~~~~~~~~~~~~
compilation terminated.
Command '['/home/tetopik/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-gcc', '-E', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/config', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/berkeley-db-1.xx/include', '-I/home/tetopik/micropython/lvgl_micropython/ext_mod/micropython-esp32-twai/src', '-I/home/tetopik/micropython/lvgl_micropython/ext_mod/spi3wire', '-I/home/tetopik/micropython/lvgl_micropython/ext_mod/spi3wire/include', '-I/home/tetopik/micropython/lvgl_micropython/ext_mod/lcd_bus', '-I/home/tetopik/micropython/lvgl_micropython/ext_mod/lcd_bus/esp32_include', '-I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_lcd/include', '-I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_lcd/interface', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython', '-I/home/tetopik/micropython/lvgl_micropython/lib', '-I/home/tetopik/micropython/lvgl_micropython/lib/lvgl', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/boards/ESP32_GENERIC_C3', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/config', '-I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/config', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/managed_components/espressif__mdns/include', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/managed_components/espressif__mdns/private_include', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/config', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/config', '-I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/app_trace/include', '-I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/app_trace/private_include', '-I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/app_trace/port/include', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/config', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/config', '-I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/app_update/include', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/config', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/config', '-I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/config', '-I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/bootloader_support/include', '-I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/bootloader_sup [... cut ...]
ninja failed with exit code 1, output of the command is in the /home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/log/idf_py_stderr_output_33281 and /home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/log/idf_py_stdout_output_33281
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: Leaving directory '/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32'
make: *** [Makefile:71: all] Error 1
```

kdschlosser, 2025-01-08:
replace the code in `/home/tetopik/micropython/lvgl_micropython/ext_mod/micropython-esp32-twai/src/modcan.c` with the code from the attached file.

[scratch_1489.c.txt](https://github.com/user-attachments/files/18347912/scratch_1489.c.txt)

closed 2025-01-08 by tetopik

tetopik, 2025-01-10:
Thanks for the update @straga, will try it out soon!

tetopik, 2025-01-10:
@straga Hi, Viktor. I just want to let you know that your last updated repo doesn't successfully compiled for esp32s2, but it does using @kdschlosser's [modcan.c](https://github.com/user-attachments/files/18347912/scratch_1489.c.txt) above.

straga, 2025-01-10:
Oh, I remember about S2

tetopik, 2025-01-10:
> Oh, I remember about S2. Checkout git added esp32s2

Awesome! Appreciate it.

kdschlosser, 2025-01-10:
@straga 

Here is the MCU list with how many TWAI channels each one has...

* C6: 2
* C3: 1
* C61: 2
* S3: 1
* S2: 1
* H2: 1
* ESP32: 1
* P4: 3

I started to do some work with the TWAI driver at the esp-idf level and I then realized there is a MAJOR flaw in the design of it. There is a constant flooding of ISR's made to notify the software when the TX buffer is empty. The transmits happen within the context of the ISR based off of receiving an event for the buffer being empty. So even if the user has nothing to send the ISR will occur and the transmit queue is checked to see if there is anything to send. 

That is one crazy way of designing something. by design it is a fantastic waste of resources. How it should have been written is when someone calls the function to transmit it should check to see if the buffer is empty. IF the transmit buffer is empty then it should send the frame. If it is not empty then it should place the frame to be sent into the queue to be sent. The event for transmit buffer being empty should only occur when the state of the buffer changed from being used to being empty. When that happens the ISR can check to see if there are any messages in the queue and if their are then send one. If the transmit queue is empty then nothing is done and no more ISR's occur for the transmit buffer being empty. There is ZERO wasted resources for ISR's made when there is no work to be done.

How did someone fumble that one? I don't know enough of the ESP32 low level stuff to be able to tweak the portion of the driver that handles the ISR for the TX buffer being empty. This has got to be a bug in the design of it. I cannot see it as being anything other than that. I am going to open an issue over at Espressif and see what they have to say about it.

tetopik, 2025-01-17:
Hi, Kevin @kdschlosser.

Recloned current repo, I got this following error:

### ESP32_GENERIC
```
[1308/1972] Building C object esp-idf/main_esp32/CMakeFiles/__idf_main_esp32.dir/__/machine_hw_spi.c.obj
FAILED: esp-idf/main_esp32/CMakeFiles/__idf_main_esp32.dir/__/machine_hw_spi.c.obj
/home/tetopik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc -DBERKELEY_DB_CONFIG_FILE=\"/home/tetopik/micropython/lvgl_micropython/lib/micropython/extmod/berkeley-db/berkeley_db_config_port.h\" -DESP_PLATFORM -DFFCONF_H=\"/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/oofatfs/ffconf.h\" -DIDF_VER=\"c9763f62\" -DLFS1_NO_ASSERT -DLFS1_NO_DEBUG -DLFS1_NO_ERROR -DLFS1_NO_MALLOC -DLFS1_NO_WARN -DLFS2_NO_ASSERT -DLFS2_NO_DEBUG -DLFS2_NO_ERROR -DLFS2_NO_MALLOC -DLFS2_NO_WARN -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMICROPY_ESP_IDF_4=1 -DMICROPY_MODULE_FROZEN_MPY="(1)" -DMICROPY_PY_BTREE=1 -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_VFS_FAT=1 -DMICROPY_VFS_LFS2=1 -DNDEBUG -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/config -I/home/tetopik/micropython/lvgl_micropython/lib/micropython -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/berkeley-db-1.xx/include -I/home/tetopik/micropython/lvgl_micropython/ext_mod/spi3wire -I/home/tetopik/micropython/lvgl_micropython/ext_mod/spi3wire/include -I/home/tetopik/micropython/lvgl_micropython/ext_mod/lcd_bus -I/home/tetopik/micropython/lvgl_micropython/ext_mod/lcd_bus/esp32_include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_lcd/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_lcd/interface -I/home/tetopik/micropython/lvgl_micropython/lib -I/home/tetopik/micropython/lvgl_micropython/lib/lvgl -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32 -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/boards/ESP32_GENERIC -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/newlib/platform_include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/config/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/config/include/freertos -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/config/xtensa/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/esp_additions/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include/soc -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/port/esp32/. -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/heap/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/log/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/soc/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/soc/esp32 -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/soc/esp32/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/hal/platform_port/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/hal/esp32/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/hal/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_rom/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_rom/include/esp32 -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32 -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_common/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_system/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_system/port/soc -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_system/port/include/private -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/xtensa/esp32/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/xtensa/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/xtensa/deprecated_include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/lwip/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/lwip/include/apps -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/lwip/include/apps/sntp -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/lwip/lwip/src/include -I/home/tetopik/mic [... cut ...] i.c.obj -c /home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/machine_hw_spi.c
/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/machine_hw_spi.c: In function 'machine_hw_spi_device_make_new':
/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/machine_hw_spi.c:645:38: error: statement with no effect [-Werror=unused-value]
645 |     if (dual || quad || octal) flags != SPI_DEVICE_HALFDUPLEX;
cc1: some warnings being treated as errors
```

tetopik, 2025-01-17:
I've managed to spot the typo:

`645 |     if (dual || quad || octal) flags != SPI_DEVICE_HALFDUPLEX;`

It should be:

`645 |     if (dual || quad || octal) flags |= SPI_DEVICE_HALFDUPLEX;`

tetopik, 2025-01-18:
Hi, Kevin, now this is the error for **ESP32_GENERIC_S3**:

```
[1396/2050] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/home/tetopik/micropython/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus.c.obj
FAILED: esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/home/tetopik/micropython/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus.c.obj
/home/tetopik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc -DBERKELEY_DB_CONFIG_FILE=\"/home/tetopik/micropython/lvgl_micropython/lib/micropython/extmod/berkeley-db/berkeley_db_config_port.h\" -DCFG_TUSB_MCU=OPT_MCU_ESP32S3 -DESP_PLATFORM -DFFCONF_H=\"/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/oofatfs/ffconf.h\" -DIDF_VER=\"c9763f62\" -DLFS1_NO_ASSERT -DLFS1_NO_DEBUG -DLFS1_NO_ERROR -DLFS1_NO_MALLOC -DLFS1_NO_WARN -DLFS2_NO_ASSERT -DLFS2_NO_DEBUG -DLFS2_NO_ERROR -DLFS2_NO_MALLOC -DLFS2_NO_WARN -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMICROPY_ESP_IDF_4=1 -DMICROPY_HW_BOARD_NAME="\"Generic ESP32S3 module with Octal-SPIRAM\"" -DMICROPY_MODULE_FROZEN_MPY="(1)" -DMICROPY_PY_BTREE=1 -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_VFS_FAT=1 -DMICROPY_VFS_LFS2=1 -DNDEBUG -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/config -I/home/tetopik/micropython/lvgl_micropython/lib/micropython -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/berkeley-db-1.xx/include -I/home/tetopik/micropython/lvgl_micropython/ext_mod/spi3wire -I/home/tetopik/micropython/lvgl_micropython/ext_mod/spi3wire/include -I/home/tetopik/micropython/lvgl_micropython/ext_mod/lcd_bus -I/home/tetopik/micropython/lvgl_micropython/ext_mod/lcd_bus/esp32_include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_lcd/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_lcd/interface -I/home/tetopik/micropython/lvgl_micropython/lib -I/home/tetopik/micropython/lvgl_micropython/lib/lvgl -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/lib/tinyusb/src -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/shared/tinyusb -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32 -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/boards/ESP32_GENERIC_S3 -I/home/tetopik/micropython/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/newlib/platform_include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/config/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/config/include/freertos -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/config/xtensa/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/freertos/esp_additions/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include/soc -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_hw_support/port/esp32s3/. -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/heap/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/log/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/soc/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/soc/esp32s3 -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/soc/esp32s3/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/hal/platform_port/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/hal/esp32s3/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/hal/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_rom/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_rom/include/esp32s3 -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32s3 -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_common/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_system/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_system/port/soc -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/esp_system/port/include/private -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/xtensa/esp32s3/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/xtensa/include -I/home/tetopik/micropython/lvgl_micropython/lib/esp-idf/components/ [... cut ...]
12 |     static inline rgb565_dither_pixel(uint8_t treshold_id, uint16_t *pixel)
|                   ^~~~~~~~~~~~~~~~~~~
/home/tetopik/micropython/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c: At top level:
/home/tetopik/micropython/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c:145:17: warning: 'copy_pixels' used but never defined
145 |     static void copy_pixels(void *dst, void *src, uint32_t x_start, uint32_t y_start,
|                 ^~~~~~~~~~~
cc1: some warnings being treated as errors
```