lvgl-micropython/lvgl_micropython #251: NV3401A-Display on ESP32-S3 MCU JC4827543R

JGelfert1, 2025-01-10:
**Describe the bug**
I want to use LVGL and MicroPython on a JC4827543R. I tried out with ESP32-2432S028r2. It works with the modified Example code "SPI bus with SPI touch (same SPI bus)". Touchpad needed to use SoftSPI then it worked fine.

LVGL works on JC4827543R with arduino-ide, see [(https://github.com/lsdlsd88/JC4827W543)]

Is there a way to get it work with JC4827543R?

**Expected behavior**
Have a Example main.py script that works for JC4827543R, display NV3401a and resistive touch with XPT2046.

kdschlosser, 2025-01-10:
build command
```
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=4 DISPLAY=nv3041a INDEV=xpt2046
```

example script

```python
import lcd_bus
from micropython import const
import machine

# display settings
_WIDTH = const(480)
_HEIGHT = const(272)
_BUFFER_SIZE = const(26112)  # _WIDTH * _HEIGHT * 2 / 10

_RTP_CS = const(10)
_RTP_MISO = const(11)
_RTP_CLK = const(12)
_RTP_MOSI = const(13)
_RTP_IRQ = const(3)
_RTP_FREQ = const(10000000)
_RTP_HOST = const(1)

_SPECK_DIN = const(41)
_SPECK_BCLK = const(42)
_SPECK_LRCLK = const(2)

_LCD_TE = const(0)  # not used
_LCD_BL = const(1)
_LCD_CS = const(45)
_LCD_CLK = const(47)
_LCD_A0 = const(21)
_LCD_A1 = const(48)
_LCD_A2 = const(40)
_LCD_A3 = const(39)
_LCD_HOST = const(2)
_LCD_FREQ = const(40000000)

disp_spi_bus = machine.SPI.Bus(
    host=_LCD_HOST,
    mosi=_LCD_A0,
    miso=_LCD_A1,
    sck=_LCD_CLK,
    quad_pins=(_LCD_A2, _LCD_A3)
)

display_bus = lcd_bus.SPIBus(
    spi_bus=disp_spi_bus,
    dc=-1,
    freq=_LCD_FREQ,
    cs=_LCD_CS,
    quad=True
)

fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import nv3041a  # NOQA
import lvgl as lv  # NOQA

display = nv3041a.NV3041A(
    data_bus=display_bus,
    frame_buffer1=fb1,
    frame_buffer2=fb2,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_LCD_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

import task_handler  # NOQA
import xpt2046  # NOQA

display.init()

rtp_spi_bus = machine.SPI.Bus(
    host=_RTP_HOST,
    mosi=_RTP_MOSI,
    miso=_RTP_MISO,
    sck=_RTP_CLK
)

rtp_spi_dev = machine.SPI.Device(
    spi_bus=rtp_spi_bus,
    cs=_RTP_CLK,
    freq=_RTP_FREQ
)

indev = xpt2046.XPT2046(rtp_spi_dev)

if not indev.is_calibrated:
    display.set_backlight(100)
    indev.calibrate()

# if you want to change from portrait to landscape or from landscape to portrait
# display.set_rotation(lv.DISPLAY_ROTATION._90)

display.set_backlight(100)

th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)
```

JGelfert1, 2025-01-11:
Thank you. I didn't see Display NV3401A in sources. In README.md NV3401A is not included. 

With the example script the device shows:
[image]

kdschlosser, 2025-01-12:
i have to do some research on the display. the driver is new and untested which is why it is not seen in the readme.

JGelfert1, 2025-01-12:
I tried to change the example code. For touchpad I changed MOSI and MISO, then touchpad works. For display i tried other combinations for buffersize and allocating other types of Memory for framebuffer. With some Combinations I saw sometimes the slider and the HelloWorld label. The slider don't reacts on touch. 

touch works only on calibration on the edge.  Which values for Framebuffer allocation may work?

https://github.com/user-attachments/assets/260decc0-700a-4f12-a8ab-79a6f30d54c9

modified main.py:

```python

import lcd_bus
from micropython import const
import machine

# display settings
_WIDTH = const(480)
_HEIGHT = const(272)
_BUFFER_SIZE = const(26112)  # _WIDTH * _HEIGHT * 2 / 10

_RTP_CS = const(10)
_RTP_MISO = const(13)
_RTP_CLK = const(12)
_RTP_MOSI = const(11)
_RTP_IRQ = const(3)
_RTP_FREQ = const(10000000)
_RTP_HOST = const(1)

_SPECK_DIN = const(41)
_SPECK_BCLK = const(42)
_SPECK_LRCLK = const(2)

_LCD_TE = const(0)  # not used
_LCD_BL = const(1)
_LCD_CS = const(45)
_LCD_CLK = const(47)
_LCD_A0 = const(21)
_LCD_A1 = const(48)
_LCD_A2 = const(40)
_LCD_A3 = const(39)
_LCD_HOST = const(2)
_LCD_FREQ = const(40000000)

disp_spi_bus = machine.SPI.Bus(
    host=_LCD_HOST,
    mosi=_LCD_A0,
    miso=_LCD_A1,
    sck=_LCD_CLK,
    quad_pins=(_LCD_A2, _LCD_A3)
)

display_bus = lcd_bus.SPIBus(
    spi_bus=disp_spi_bus,
    dc=0,
    freq=_LCD_FREQ,
    cs=_LCD_CS,
    quad=True
)

fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_DEFAULT)
fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_DEFAULT)

import nv3041a  # NOQA
import lvgl as lv  # NOQA

display = nv3041a.NV3041A(
    data_bus=display_bus,
    frame_buffer1=fb1,
    frame_buffer2=fb2,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_LCD_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

import task_handler  # NOQA
import xpt2046  # NOQA

display.init()

rtp_spi_bus = machine.SPI.Bus(
    host=_RTP_HOST,
    mosi=_RTP_MOSI,
    miso=_RTP_MISO,
    sck=_RTP_CLK
)

rtp_spi_dev = machine.SPI.Device(
    spi_bus=rtp_spi_bus,
    cs=_RTP_CS,
    freq=_RTP_FREQ
)

indev = xpt2046.XPT2046(rtp_spi_dev)

#if not indev.is_calibrated:
display.set_backlight(100)
indev.calibrate()

# if you want to change from portrait to landscape or from landscape to portrait
# display.set_rotation(lv.DISPLAY_ROTATION._90)

display.set_backlight(100)

th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO, WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)

```

JGelfert1, 2025-01-13:
Without DMA there is a way to see more on the display. But then there is a problem with _flush_cb() from NV3401A.

[image]

```
LVGL (9.2.2) MicroPython (1.24.1) Binding compiled on 2025-01-132S3 module with Octal-SPIRAM with ESP32S3

Type "help()" for more information.

>>> Traceback (most recent call last):

  File "task_handler.py", line 136, in _task_handler

  File "nv3041a.py", line 205, in _flush_cb
OSError: 257(lcd_panel_io_tx_color)
LVGL (9.2.2) Mice i
```

main.py: 

```python
import lcd_bus
from micropython import const
import machine
import lvgl as lv  # NOQA

# display settings
_WIDTH = const(480)
_HEIGHT = const(272)
_BUFFER_SIZE = _WIDTH * _HEIGHT * 16

_RTP_CS = const(10)
_RTP_MISO = const(13)
_RTP_CLK = const(12)
_RTP_MOSI = const(11)
_RTP_IRQ = const(3)
_RTP_FREQ = const(10000000)
_RTP_HOST = const(1)

_SPECK_DIN = const(41)
_SPECK_BCLK = const(42)
_SPECK_LRCLK = const(2)

_LCD_TE = const(0)  # not used
_LCD_BL = const(1)
_LCD_CS = const(45)
_LCD_CLK = const(47)
_LCD_A0 = const(21)
_LCD_A1 = const(48)
_LCD_A2 = const(40)
_LCD_A3 = const(39)
_LCD_HOST = const(2)
_LCD_FREQ = const(32000000)

disp_spi_bus = machine.SPI.Bus(
    host=_LCD_HOST,
    mosi=_LCD_A0,
    miso=_LCD_A1,
    sck=_LCD_CLK,
    quad_pins=(_LCD_A2, _LCD_A3)
)

display_bus = lcd_bus.SPIBus(
    spi_bus=disp_spi_bus,
    dc=-1,
    freq=_LCD_FREQ,
    cs=_LCD_CS,
    quad=True
)

fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM)
fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM)

import nv3041a  # NOQA

display = nv3041a.NV3041A(
    data_bus=display_bus,
    frame_buffer1=fb1,
    frame_buffer2=fb2,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_LCD_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    #rgb565_byte_swap=False,
    offset_y = 0,
    offset_x = 0,
)

import task_handler  # NOQA
import xpt2046  # NOQA

rtp_spi_bus = machine.SPI.Bus(
    host=_RTP_HOST,
    mosi=_RTP_MOSI,
    miso=_RTP_MISO,
    sck=_RTP_CLK
)

rtp_spi_dev = machine.SPI.Device(
    spi_bus=rtp_spi_bus,
    cs=_RTP_CS,
    freq=_RTP_FREQ
)

indev = xpt2046.XPT2046(rtp_spi_dev)

#if not indev.is_calibrated:
#display.set_backlight(100)
#indev.calibrate()

# if you want to change from portrait to landscape or from landscape to portrait
# display.set_rotation(lv.DISPLAY_ROTATION._90)
display.set_power(True)
display.set_backlight(100)
display.init()

lv.init()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO, WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)

th = task_handler.TaskHandler()

```

kdschlosser, 2025-01-14:
I believe I just found what the problem is.

```python
import lcd_bus
from micropython import const
import machine
import lvgl as lv  # NOQA

# display settings
_WIDTH = const(480)
_HEIGHT = const(272)

# this was the problem. The buffers need to be full frame buffers and not partials.
# you had it set so it was 16 bytes per pixel. That's far too large and what was 
# causing the problem with how it was being displayed. I had it set to 1/10th the 
# size of a full frame buffer so that is the reason why it was not displaying properly 
# with what I had written it as initially. This is how it should be done.
_BUFFER_SIZE = _WIDTH * _HEIGHT * 2   

_RTP_CS = const(10)
_RTP_MISO = const(13)
_RTP_CLK = const(12)
_RTP_MOSI = const(11)
_RTP_IRQ = const(3)
_RTP_FREQ = const(10000000)
_RTP_HOST = const(1)

_SPECK_DIN = const(41)
_SPECK_BCLK = const(42)
_SPECK_LRCLK = const(2)

_LCD_TE = const(0)  # not used
_LCD_BL = const(1)
_LCD_CS = const(45)
_LCD_CLK = const(47)
_LCD_A0 = const(21)
_LCD_A1 = const(48)
_LCD_A2 = const(40)
_LCD_A3 = const(39)
_LCD_HOST = const(2)
_LCD_FREQ = const(32000000)

disp_spi_bus = machine.SPI.Bus(
    host=_LCD_HOST,
    mosi=_LCD_A0,
    miso=_LCD_A1,
    sck=_LCD_CLK,
    quad_pins=(_LCD_A2, _LCD_A3)
)

display_bus = lcd_bus.SPIBus(
    spi_bus=disp_spi_bus,
    dc=-1,
    freq=_LCD_FREQ,
    cs=_LCD_CS,
    quad=True
)

fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM | lcd_bus.MEMORY_DMA)
fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM | lcd_bus.MEMORY_DMA)

import nv3041a  # NOQA

display = nv3041a.NV3041A(
    data_bus=display_bus,
    frame_buffer1=fb1,
    frame_buffer2=fb2,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_LCD_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    #rgb565_byte_swap=False,
    offset_y = 0,
    offset_x = 0,
)

import task_handler  # NOQA
import xpt2046  # NOQA

rtp_spi_bus = machine.SPI.Bus(
    host=_RTP_HOST,
    mosi=_RTP_MOSI,
    miso=_RTP_MISO,
    sck=_RTP_CLK
)

rtp_spi_dev = machine.SPI.Device(
    spi_bus=rtp_spi_bus,
    cs=_RTP_CS,
    freq=_RTP_FREQ
)

indev = xpt2046.XPT2046(rtp_spi_dev)

#if not indev.is_calibrated:
#display.set_backlight(100)
#indev.calibrate()

# if you want to change from portrait to landscape or from landscape to portrait
# display.set_rotation(lv.DISPLAY_ROTATION._90)
display.set_power(True)
display.set_backlight(100)
display.init()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO, WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)

th = task_handler.TaskHandler()

```

give that a try and see if it works.

JGelfert1, 2025-01-14:
Thanks for the suggestion.

I tried out with _BUFFER_SIZE = _WIDTH * _HEIGHT * 2 .  Incomplete Display and error are the same.

```
LVGL (9.2.2) MicroPython (1.24.1) Binding compiled on 2025-01-132S3 module with Octal-SPIRAM with ESP32S3

Type "help()" for more information.
>>> Traceback (most recent call last):
  File "task_handler.py", line 136, in _task_handler
  File "nv3041a.py", line 205, in _flush_cb
OSError: 257(lcd_panel_io_tx_color)
LVGL (9.2.2) Mic
```

JGelfert1, 2025-01-14:
I get it. I changed _flush_cb() in nv3041a.py. Now only touch is inverted. 

Many thanks for your support. 

[image]

https://github.com/user-attachments/assets/334fdb0d-380f-4aa8-99ac-079052bbc225

nv3041a.py:
```python
...
def _flush_cb(self, _, area, color_p):
        x1 = area.x1 + self._offset_x
        x2 = area.x2 + self._offset_x

        y1 = area.y1 + self._offset_y
        y2 = area.y2 + self._offset_y

        self._set_memory_location(x1, y1, x2, y2)

        width = x2 - x1 + 1
        height = y2 - y1 + 1
        size = width * height * lv.color_format_get_size(self._color_space)

        data_view = color_p.__dereference__(size)
        
        #Divide buffer in 2 chunks:
        first_chunk = int(size / 2)

        self._data_bus.tx_color(self.__ramwr, data_view[:first_chunk], x1, y1, x2, y2, self._rotation, False)
        self._data_bus.tx_color(self.__ramwrc, data_view[first_chunk:], x1, y1, x2, y2, self._rotation, self._disp_drv.flush_is_last())
```
main.py
```python
import lcd_bus
from micropython import const
import machine
import lvgl as lv  # NOQA

# display settings
_WIDTH = const(480)
_HEIGHT = const(272)
_BUFFER_SIZE = _WIDTH * _HEIGHT * 2

_RTP_CS = const(10)
_RTP_MISO = const(13)
_RTP_CLK = const(12)
_RTP_MOSI = const(11)
_RTP_IRQ = const(3)
_RTP_FREQ = const(10000000)
_RTP_HOST = const(1)

_SPECK_DIN = const(41)
_SPECK_BCLK = const(42)
_SPECK_LRCLK = const(2)

_LCD_TE = const(0)  # not used
_LCD_BL = const(1)
_LCD_CS = const(45)
_LCD_CLK = const(47)
_LCD_A0 = const(21)
_LCD_A1 = const(48)
_LCD_A2 = const(40)
_LCD_A3 = const(39)
_LCD_HOST = const(2)
_LCD_FREQ = const(32000000)

disp_spi_bus = machine.SPI.Bus(
    host=_LCD_HOST,
    mosi=_LCD_A0,
    miso=_LCD_A1,
    sck=_LCD_CLK,
    quad_pins=(_LCD_A2, _LCD_A3)
)

display_bus = lcd_bus.SPIBus(
    spi_bus=disp_spi_bus,
    dc=-1,
    freq=_LCD_FREQ,
    cs=_LCD_CS,
    quad=True
)

fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM)
fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM)

import nv3041a  # NOQA

display = nv3041a.NV3041A(
    data_bus=display_bus,
    frame_buffer1=fb1,
    frame_buffer2=fb2,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_LCD_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    #rgb565_byte_swap=False,
    offset_y = 0,
    offset_x = 0,
)

import task_handler  # NOQA
import xpt2046  # NOQA

rtp_spi_bus = machine.SPI.Bus(
    host=_RTP_HOST,
    mosi=_RTP_MOSI,
    miso=_RTP_MISO,
    sck=_RTP_CLK
)

rtp_spi_dev = machine.SPI.Device(
    spi_bus=rtp_spi_bus,
    cs=_RTP_CS,
    freq=_RTP_FREQ
)

indev = xpt2046.XPT2046(rtp_spi_dev)

#if not indev.is_calibrated:
#display.set_backlight(100)
#indev.calibrate()

# if you want to change from portrait to landscape or from landscape to portrait
#display.set_rotation(lv.DISPLAY_ROTATION._90)
display.set_power(True)
display.set_backlight(100)
display.init()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

label = lv.label(scrn)
label.set_text('Hallo, Welt!')
label.align(lv.ALIGN.CENTER, 0, -50)

th = task_handler.TaskHandler()

```

JGelfert1, 2025-01-14:
Here' s a main.py mit working touch:
```python
import lcd_bus
from micropython import const
import machine
import lvgl as lv  # NOQA

# display settings
_WIDTH = const(480)
_HEIGHT = const(272)
_BUFFER_SIZE = _WIDTH * _HEIGHT * 2

_RTP_CS = const(10)
_RTP_MISO = const(13)
_RTP_CLK = const(12)
_RTP_MOSI = const(11)
_RTP_IRQ = const(3)
_RTP_FREQ = const(10000000)
_RTP_HOST = const(1)

_SPECK_DIN = const(41)
_SPECK_BCLK = const(42)
_SPECK_LRCLK = const(2)

_LCD_TE = const(0)  # not used
_LCD_BL = const(1)
_LCD_CS = const(45)
_LCD_CLK = const(47)
_LCD_A0 = const(21)
_LCD_A1 = const(48)
_LCD_A2 = const(40)
_LCD_A3 = const(39)
_LCD_HOST = const(2)
_LCD_FREQ = const(32000000)

disp_spi_bus = machine.SPI.Bus(
    host=_LCD_HOST,
    mosi=_LCD_A0,
    miso=_LCD_A1,
    sck=_LCD_CLK,
    quad_pins=(_LCD_A2, _LCD_A3)
)

display_bus = lcd_bus.SPIBus(
    spi_bus=disp_spi_bus,
    dc=-1,
    freq=_LCD_FREQ,
    cs=_LCD_CS,
    quad=True
)

fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM)
fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_SPIRAM)

import nv3041a  # NOQA

display = nv3041a.NV3041A(
    data_bus=display_bus,
    frame_buffer1=fb1,
    frame_buffer2=fb2,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_LCD_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    #rgb565_byte_swap=False,
    offset_y = 0,
    offset_x = 0,
)

import task_handler  # NOQA
import xpt2046  # NOQA

rtp_spi_bus = machine.SPI.Bus(
    host=_RTP_HOST,
    mosi=_RTP_MOSI,
    miso=_RTP_MISO,
    sck=_RTP_CLK
)

rtp_spi_dev = machine.SPI.Device(
    spi_bus=rtp_spi_bus,
    cs=_RTP_CS,
    freq=_RTP_FREQ
)

display.set_power(True)
display.set_backlight(100)
display.init()
display.set_rotation(lv.DISPLAY_ROTATION._180)

indev = xpt2046.XPT2046(
    rtp_spi_dev)

if not indev.is_calibrated:
    indev.calibrate()

# if you want to change from portrait to landscape or from landscape to portrait
#display.set_rotation(lv.DISPLAY_ROTATION._90)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

label = lv.label(scrn)
label.set_text('Hallo, Welt!')
label.align(lv.ALIGN.CENTER, 0, -50)

th = task_handler.TaskHandler()

```

kdschlosser, 2025-01-14:
sweet you got it working, FANTASTIC.. I am however going to alter your changes a bit so we don't need to use 2 huge frame buffers. It should be pretty easy to do based on the code you have provided.

kdschlosser, 2025-01-15:
Do me a favor and test this code to see if it works.. leave the changes you made to the flush callback in place and use the code below for your main.py script. 

```python
import lcd_bus
from micropython import const
import machine
import lvgl as lv  # NOQA

# display settings
_WIDTH = const(480)
_HEIGHT = const(272)
_BUFFER_SIZE = int(_WIDTH * _HEIGHT * 2 / 10)

_RTP_CS = const(10)
_RTP_MISO = const(13)
_RTP_CLK = const(12)
_RTP_MOSI = const(11)
_RTP_IRQ = const(3)
_RTP_FREQ = const(10000000)
_RTP_HOST = const(1)

_SPECK_DIN = const(41)
_SPECK_BCLK = const(42)
_SPECK_LRCLK = const(2)

_LCD_TE = const(0)  # not used
_LCD_BL = const(1)
_LCD_CS = const(45)
_LCD_CLK = const(47)
_LCD_A0 = const(21)
_LCD_A1 = const(48)
_LCD_A2 = const(40)
_LCD_A3 = const(39)
_LCD_HOST = const(2)
_LCD_FREQ = const(32000000)

disp_spi_bus = machine.SPI.Bus(
    host=_LCD_HOST,
    mosi=_LCD_A0,
    miso=_LCD_A1,
    sck=_LCD_CLK,
    quad_pins=(_LCD_A2, _LCD_A3)
)

display_bus = lcd_bus.SPIBus(
    spi_bus=disp_spi_bus,
    dc=-1,
    freq=_LCD_FREQ,
    cs=_LCD_CS,
    quad=True
)

fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import nv3041a  # NOQA

display = nv3041a.NV3041A(
    data_bus=display_bus,
    frame_buffer1=fb1,
    frame_buffer2=fb2,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_LCD_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    #rgb565_byte_swap=False,
    offset_y = 0,
    offset_x = 0,
)

import task_handler  # NOQA
import xpt2046  # NOQA

rtp_spi_bus = machine.SPI.Bus(
    host=_RTP_HOST,
    mosi=_RTP_MOSI,
    miso=_RTP_MISO,
    sck=_RTP_CLK
)

rtp_spi_dev = machine.SPI.Device(
    spi_bus=rtp_spi_bus,
    cs=_RTP_CS,
    freq=_RTP_FREQ
)

display.set_power(True)
display.set_backlight(100)
display.init()
display.set_rotation(lv.DISPLAY_ROTATION._180)

indev = xpt2046.XPT2046(
    rtp_spi_dev)

if not indev.is_calibrated:
    indev.calibrate()

# if you want to change from portrait to landscape or from landscape to portrait
#display.set_rotation(lv.DISPLAY_ROTATION._90)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

label = lv.label(scrn)
label.set_text('Hallo, Welt!')
label.align(lv.ALIGN.CENTER, 0, -50)

th = task_handler.TaskHandler()

```

JGelfert1, 2025-01-15:
The skript results in the first pictures. Drawing in a tenth of display on the edge.

https://github.com/user-attachments/assets/538f5741-eafa-4da0-af7b-47dcd7d64a74

kdschlosser, 2025-01-15:
OK so it does need to have the entire display rendered each time. Funny thing is that is not mentioned at all in the datasheet for the display IC nor is there any mention of having to split the buffer when transmitting using 2 different registers when performing the transmit...

JGelfert1, 2025-01-21:
Thank you for your support. The display works but if I try a little bit more there are problems with less memory.

microPython is a great tool but there is a little bit to much overhead for the comfort of a great programming language. 

Please continue your work. You do a great job.

closed 2025-01-21 by JGelfert1