lvgl-micropython/lvgl_micropython #253: Pb with SPI

F4HSL-Yves, 2025-01-11:
**Describe the bug**
A clear and concise description of what the bug is.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Images**

If it is a visual issue then take a picture of it or a video and add it here.

**If you are having a problem compiling paste the build output below. (NO SCREENSHOTS)**

```shell
PASTE OUTPUT HERE
```

**If your problem is running the binding please past the code you are using below.***
I need to see ALL of the setup code exactly as it is being ran.

```python
PASTE CODE HERE
```

**Exact make and model number of the MCU that you are compiling for or the firmware is running on. **

* Make:
* Model:

**For ESP32 MCU's The PSRAM and FLASH SPI type, quad SPI or octal SPI.**

* PSRAM: quad OR octal
* FLASH:  quad OR octal

I need to know the OS and OS version of the machine that compiled the binary. If using a VM then I need to know the OS and OS version the VM is running. (WSL == VM)
 - OS: [e.g. iOS]

**Build Command**

```shell
BUILD COMMAND HERE
```

F4HSL-Yves, 2025-01-11:
I tried to test the SDcard on my ESP32LCDTOUCH 2.8 from waveshare  : 

import machine, sdcard, os
from machine import SPI
spi = SPI(0, baudrate=100000)  
sd = sdcard.SDCard(spi, machine.Pin(21))
os.mount(sd, '/sd')
os.listdir('/')

I get always this error : 
Traceback (most recent call last):
  File "<stdin>", line 5, in <module>
TypeError: can't create 'SPI' instances

When i use a standard ESP32S3 board with micropython (without LVGL) i don't get this error

An idea ?

Regards
Yves

F4HSL-Yves, 2025-01-11:
hi, i have tested SD features  with the standard micripython firmware on my touch device, no problem. Seems LVGL locks something about nex SPI instances...

kdschlosser, 2025-01-11:
The SD Card works fine, It actually works better than the vanilla MicroPython SDCard works because you are able to use it with other devices attached to the bus. You cannot do this with vanilla MicroPython..

To to use the SDCard this is what6 you need to do..

```python
spi_bus = machine.SPI.Bus(host=2, miso9, mosi=11, sck=12)
sdcard = machine,SDCard(spi_bus=spi_bus, cs=14)
```

F4HSL-Yves, 2025-01-11:
Hi Kevin, tnx a lot, it is working now, no more errors ! I have to test full feature of vfs but i think we can close the issue (which was not...)

kdschlosser, 2025-01-12:
You can reopen the issue if there is a problem. It has been tested and it does work.

With vanilla MicroPython you are unable to share an SPI bus with displays and SDCards which is the reason why the change was made. Now you can have an SDCard reader, Display and touch panel all on the same SPI Bus. You can actually have a quad SPI SDCard reader, dual SPI display and a traditional SPI touch panel all operating at different speeds all on the same bus.

closed 2025-01-12 by kdschlosser

F4HSL-Yves, 2025-01-12:
II works fine, tnx again Kevin for your work and réactivity to questions !
By the way i wish you and your family a very happy new year !
PS (ponctuation correction for readers :-), gpio numbers to be ajusted for the board used of course :
spi_bus = machine.SPI.Bus(host=2, miso=9, mosi=11, sck=12)
sdcard = machine.SDCard(spi_bus=spi_bus, cs=14)

> Le 11 janv. 2025 à 21:03, Kevin Schlosser ***@***.***> a écrit :
> 
> spi_bus = machine.SPI.Bus(host=2, miso9, mosi=11, sck=12) sdcard = machine,SDCard(spi_bus=spi_bus, cs=14)

kdschlosser, 2025-01-12:
It got the idea across punctuation errors and all.