lvgl-micropython/lvgl_micropython #254: White screen

Bigmaxmafia, 2025-01-12:
I'm using esp32 s3 octal. When running the code there are no errors, you can see in the console that the program has worked. But on the st7796 there is just a white screen, and upon restart you can see that the screen backlight turns off and turns on again. I tried other pins - the same result.  
I tried to do esp32-wrover-ie spiram  - also a white screen.

```shell
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1108
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbdc
load:0x403cc700,len:0x2f4c
entry 0x403c989c
END
LVGL (9.2.2) MicroPython (1.24.1) Binding compiled on 2025-01-11; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
```

```python
import lcd_bus
from micropython import const
import machine

# display settings
_WIDTH = const(320)
_HEIGHT = const(480)
_BL = const(6)
_RST = const(4)
_DC = const(5)

_MOSI = const(11)
_MISO = const(12)
_SCK = const(13)
_HOST = const(1)  # SPI2

_LCD_CS = const(10)
_LCD_FREQ = const(80000000)

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)

# we are going to let the display driver sort out the best freame buffer size and where to allocate it to.
# fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
# fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import st7796  # NOQA
import lvgl as lv  # NOQA

display = st7796.ST7796(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7796.BYTE_ORDER_RGB,
    rgb565_byte_swap=True,
)

import task_handler  # NOQA

display.set_power(True)
display.init()
display.set_backlight(100)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)
print('END')
```

**Build Command**

```shell
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=st7796 INDEV=xpt2046
```

kdschlosser, 2025-01-12:
I see you have a reset pin but you are not supplying it to the display driver... When a reset pin is connected then you need to pass it to the display driver. You also may need to reverse the polarity that the reset occurs in. Some displays will be a negitive polarity and some will be a positive and the display could be interpreting a float condition of the pin as being a reset so the display is staying latched in a reset state.

Bigmaxmafia, 2025-01-12:
> I see you have a reset pin but you are not supplying it to the display driver... When a reset pin is connected then you need to pass it to the display driver. You also may need to reverse the [...]

@kdschlosser im add     
reset_pin=_RST,
reset_state=st7796.STATE_HIGH,

If HIGH - nothing changes, white screen
if LOW - entire screen in RGB dots

kdschlosser, 2025-01-12:
do you have a link to the datasheet for the display?

Bigmaxmafia, 2025-01-12:
> do you have a link to the datasheet for the display?

@kdschlosser 
[image]
[image]

kdschlosser, 2025-01-12:
I am seeing this display but with all kinds of different display IC's being listed as being used. Even that image you posted has conflicting information In one place it says it is an ST7796 and in another it states it is a ILI9486. so which is it?

Here is a list of some of the IC's I have found for that display....

ILI9341 
ILI9488
ILI9486
ST7796

kdschlosser, 2025-01-12:
Lower your LCD frequency down to 40mhz from 80. You are never going to be able to get 80 since you are using jumper wires to the display. That's if you are using hardware defined SPI pins and host 2 (host = 1).

SPI host 2 (host = 1) you MUST use the hardware pins and those are as follows..

MOSI: 11
MISO: 13
CS: 10
SCK: 12

If you use SPI3 (host = 2) then you can use whatever pins you want. There is a speed limitation when using SPI3 because of the routing through the GPIO Matrix on the ESP32. This has a maximum speed of 27mhz, that's what the datasheet states.

If you are not going to be using a pin set it to `-1`

Bigmaxmafia, 2025-01-13:
@kdschlosser Thanks for your help! In my esp documentation the SCK/MISO pins were mixed up.

closed 2025-01-13 by Bigmaxmafia

kdschlosser, 2025-01-13:
Ohhhh sucks when that happens doesn't it.