lvgl-micropython/lvgl_micropython #255: esp32-S3 + st7789 or ili9341: No errors, but not rendering anything

kdmukai, 2025-01-12:
First off, THANK YOU SO MUCH FOR WORKING ON THIS PROJECT!!!!!

Getting lvgl into MicroPython in a SANE way for devs is just an INCREDIBLE gift!

---

## Issue Overview
Compiling and running on two different esp32-S3 units.

Trying to drive st7789 or ili9341.

For the st7789: During initialization, screen goes black for a moment, then the backlight comes back. But nothing is rendered to the screen (not even the background color).

For the ili9341: Screen initializes and goes full white, but nothing is rendered.

Confirmed that the st7789 screen and physical wiring / pin selection work by running a pure python library (micropython-nano-gui with an older MP v1.21 release .bin).

Also spent a little time with an esp32-S2 WROVER board + st7789 and couldn't get anything to render there, either.

---

esp32-s3 DevKitC-1 + st7789:
```python
from micropython import const
import machine

_WIDTH = const(240)
_HEIGHT = const(240)

import lcd_bus

# Tried host 1 and 2; w/hard resets in between
_HOST = const(1)
_MOSI_PIN = const(11)
_MISO_PIN = const(-1)  # Not connected
_SCLK_PIN = const(12)

_DC_PIN = const(7)
_CS_PIN = const(10)
_RESET_PIN = const(6)
_POWER_PIN = const(-1)
_BACKLIGHT_PIN = const(5)

_FREQ = const(40_000_000)

_WP_PIN = const(-1)
_HD_PIN = const(-1)

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI_PIN,
    miso=_MISO_PIN,
    sck=_SCLK_PIN,
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_FREQ,
    dc=_DC_PIN,
    cs=_CS_PIN,
)

import st7789  # NOQA
import lvgl as lv  # NOQA

display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BACKLIGHT_PIN,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_BGR,
    rgb565_byte_swap=True
)

display.init()

display.set_backlight(100)

import task_handler
th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0xff0000), 0)

slider = lv.slider(scrn)
slider.set_size(180, 30)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)
```

UM Feather S3 + ili9341:
```python
from micropython import const
import machine

# Tried swapping dimensions as well
_WIDTH = const(240)
_HEIGHT = const(320)

import lcd_bus

_HOST = const(1)
_MOSI_PIN = const(35)
_MISO_PIN = const(37)  # Not connected
_SCLK_PIN = const(36)

_DC_PIN = const(5)
_CS_PIN = const(6)
_RESET_PIN = const(-1)
_POWER_PIN = const(-1)
_BACKLIGHT_PIN = const(12)

_FREQ = const(40_000_000)

_WP_PIN = const(-1)
_HD_PIN = const(-1)

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI_PIN,
    miso=_MISO_PIN,
    sck=_SCLK_PIN,
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_FREQ,
    dc=_DC_PIN,
    cs=_CS_PIN,
)

import ili9341  # NOQA
import lvgl as lv  # NOQA

display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BACKLIGHT_PIN,
    color_space=lv.COLOR_FORMAT.RGB565
)

# Tried both types 1 and 2 w/hard resets inbetween
display.init(type=2)

display.set_backlight(100)

import task_handler
th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0xff0000), 0)

slider = lv.slider(scrn)
slider.set_size(180, 30)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)
```

---

**Exact make and model number of the MCU that you are compiling for or the firmware is running on. **

* Make:  Espressif
* Model: [ESP32-S3-DevKitC-1-N8R8](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-devkitc-1/user_guide_v1.0.html)

* Make: Unexpected Maker
* Model: [Feather S3](https://esp32s3.com/feathers3.html)

**For ESP32 MCU's The PSRAM and FLASH SPI type, quad SPI or octal SPI.**

S3 DevKit C:
* PSRAM: octal
* FLASH:  quad

UM Feather S3:
* PSRAM: quad
* FLASH:  quad

---

 - OS: Ubuntu 22.04.5 (via WSL2)

---

ST7789:
[Waveshare 240x240, 1.3inch IPS LCD display HAT for Raspberry Pi](https://www.waveshare.com/1.3inch-lcd-hat.htm) (using a gpio breakout to breadboard just the display pins)

ILI9341: Generic/Ubiquitous red board 2.4" TFT SPI 240x320 (touch pins ignored for now).

---

**Build Command**

Current commit: 19dedff

For the DevKit C board, tried running both:
```shell
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 DISPLAY=st7789 DISPLAY=ili9341 DISPLAY=ili9486

python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=st7789 DISPLAY=ili9341 DISPLAY=ili9486
```

For the UM Feather S3:
```shell
python3 make.py esp32 BOARD=UM_FEATHERS3 DISPLAY=st7789 DISPLAY=ili9341 DISPLAY=ili9486
```

kdschlosser, 2025-01-12:
for the devkit-c your build command is wrong.

it should be...

```
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=st7789 DISPLAY=ili9341 DISPLAY=ili9486 --flash-size=8 
```

For the Feather you need to use SPI host 3 which is the number 2.

Have you checked the mode to see what is needed for your displays? I have seen some that need to have the mode set to '3' How you do that is by setting the polarity and phase both to 1

kdmukai, 2025-01-12:
BOTH DISPLAYS ARE WORKING!!

Updating the DevKit-C build line didn't make a difference, but flipping the RESET pin to LOW did the trick:

```python
display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    reset_pin=_RESET_PIN,
    reset_state=st7789.STATE_LOW,  # <-----
    backlight_pin=_BACKLIGHT_PIN,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_BGR,
    rgb565_byte_swap=True
)
```

---

Same for the Feather S3 + ili9341. On this one I initially didn't have the RESET pin wired. Added that wire and set it to LOW.

```python
_RESET_PIN = const(14)
display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    reset_pin=_RESET_PIN,  # <-----
    reset_state=ili9341.STATE_LOW,  # <-----
    backlight_pin=_BACKLIGHT_PIN,
    color_space=lv.COLOR_FORMAT.RGB565
)
```

closed 2025-01-12 by kdmukai

kdschlosser, 2025-01-12:
Glad you got it figured out. It all depends on how the manufacturer set the IC up. That's the reason why I coded in the ability to change the reset state. I did the same thing for the power pin and also the backlight pin. This is so you know for later on down the road should you have an issue.

kdschlosser, 2025-01-12:
also I didn't mention the reset pin state because you did not have the pin being passed to the display driver..

This is from your initial code example.
```python
display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BACKLIGHT_PIN,
    color_space=lv.COLOR_FORMAT.RGB565
)
```

This is the rule of thumb so you know in the future. If the display you are using has an MCU built in and you see the pins being connected in the diagram then you MUST use them in your code. If the display doesn't have an MCU built in and you are wiring the 2 together then you do not have to connect things you do not need to control.

Here is an example.

Say you have a display with no MCU and that display is SPI and it has these pins broken out.

MISO
MOSI
SCK
CS
BL
RST
DC
PWR

You do not have to connect the following pins
MISO
CS
BL
RST
PWR

But you DO have to supply the signal those pins are expecting to see.
If the reset is a positive reset then the RST needs to be connected to ground. The PWR and BL would need to be connected to the supply voltage of the display. The CS would need to be connected to ground and the MISO can be left alone.

That will save some pin connections. but you loose the ability to do things programmatically like power on and off the display, setting the brightness level of the backlight (if supported) and having other devices attached to the same SPI bus as the display.