lvgl-micropython/lvgl_micropython #257: pinging Kevin

asheroto, 2025-01-14:
Hey @kdschlosser,

Sorry to contact you over here off-topic, but I noticed you had commits here just a few minutes ago.

People over here are worried about you:
`https://github.com/EventGhost/EventGhost/issues/417#issuecomment-1383238472`

Some have regenerated some of the content:
`https://github.com/EventGhost/EventGhost/issues/417#issuecomment-1384919971`

There's talk of domain transfers and project continuation plans. I definitely don't want to pressure you but since there is no official word we aren't sure what's going on with EventGhost. If you're done with the project I think everyone would understand, we just weren't even sure you were still on GitHub until this or if you had named someone to replace or assist you as repo owner.

Links provided in markdown as code to prevent tagging on the other topic to protect your privacy. 😊

Hope you're alright, man. Feel free to hide or delete this.

kdschlosser, 2025-01-14:
I don't own the domains, never have. The person that originally authored EventGhost owns them and I have never been able to get in contact with that person. I tried many times and never got a response.

The code is hosted on Github so it is free and open source so anyone is able to download it and modify it as they see fit. If someone wants to start developing it again then they can fork the repo and once I see that there is work being done it it on a consistent basis for a duration of time I will be more than happy to transfer ownership of the repo to that person.

asheroto, 2025-01-14:
Gotcha. Sorry for the confusion! Glad you're okay! 😊

I think the confusion was created because if you look at EventGhost Project (org) on GitHub, you're the only member listed, at least from what I can tell. Maybe the other members left the GitHub organization? Do you have administrative control over the org or repo still?

https://github.com/orgs/EventGhost/people

[image]

I'm pinging @dequi, @Ian-Zz, and @DanEdens as they have been actively involved in helping with this. One of them would likely be the candidates to carry the torch. As far as the domain goes, I believe @Ian-Zz may currently own the domain and can comment further.

Again, sorry for the off-topic here, just trying to help sort this out. I appreciate your time.

asheroto, 2025-01-14:
Does anyone know who may have had access to the EventGhost forum database? Who was the webmaster?

Ian-Zz, 2025-01-14:
I have no access to database. But concerning the eventghost.net domain at domain.com that was in my name. Months back I removed my payment info from it and also stopped auto renew. Just checked at domain com and here is the stats... eventghost.net
Was protected from expiring but requires renewal
Domain Expiration Protection (eventghost.net)
Expired
01/18/2024
Domain Privacy + Protection (eventghost.net)
Expires Soon
01/16/2025

If anybody wants it transfered to them. We have a few days it looks like. I know before when I tried to share the eg.net domain.com admin privileges with others they made it damn near impossible. I am willing to do what's needed to hand it over.

Ian

kdschlosser, 2025-01-15:
the forum database is gone. The company that was doing the hosting up and went bankrupt and the database was lost when that happened. I have an old computer that should have a copy of the database on it but it is an older copy of it. I am not even 100% sure if the database is even on that machine or not....

The actual domain for eventghost was "eventghost.com" and apparently the original author finally let it expire and some holding company bought the domain.... I do have the domain "eventghost.rocks" that just renewed automatically last month. I just saw it yesterday in my charge card bill.

asheroto, 2025-01-15:
@Ian-Zz I volunteer to pay for the domain transfer to help keep the site alive and avoid scammers from grabbing it. Our company has its own web servers and WAF, so am extremely familiar with web hosting & domain management. You can [chat with me](https://tawk.to/asheroto) or email me asheroto [at] proton mail [dot] com.

Regarding `eventghost.com` indeed I saw that it appears another company has it, but will expire March 5. I will put in an order to catch it once it expires and pay for the renewal. We have enough domains in our account that we can get it for somewhat of a discount once it becomes available.

@kdschlosser If you are able to find any backup of the database, I'll volunteer to recreate the forum and get it back online using our web servers as the host.

Ian-Zz, 2025-01-15:
Email sent asheroto.

kdschlosser, 2025-01-15:
May I make a suggestion for a host?.... I recommend using [Joe's Datacenter](https://joesdatacenter.com/) (It's now called Patmos) out of Kansas City. They usually have a long list of servers you can lease from them. At the moment the list is pretty small but they do have dual processor intel box that is 12 cores at 2.67ghz per processor for 40 a month. They do have deals where I have seen this same machine for as low as 25 a month. It's a 30 day contract so you are not locked into anything long term. I don't remember off hand what the cap is for the data, It's some very large number like 4 or 8 TB (up and down combined). It's a 1000mbit link and you actually get 1000mbit up and down. I believe you can get up to 10 static ipv4 addresses. It's a bare metal server so you are not sharing the machine with anyone else. It's a great rate if you get a machine when they are on special.

kdschlosser, 2025-01-15:
It's going to be a while before I am able to get that other computer connected and up and running. I am int he process of doing a down to the stud remodel of my office and I have stuff in storage and stuff stacked in every space I had available in my home. I have no idea where the computer is stashed at. I am not going to even try to locate the thing. With my luck it's in the storage unit and it will be all the way in the back of it. I have 1/2 the office sheet rocked, joints taped and primed. The trim is cut for 2 of the 3 windows, I still have 2 doors and the closet to do the trim for. I am going to tear out the Pergo floor and install 2/4" solid bamboo flooring (not that engineered crap) 1 more wall still needs to be rocked, taped and skimmed with mud.

asheroto, 2025-01-15:
Sounds like you have a lot of things on your plate at the moment! Hope all goes as planned. :-)

DanEdens, 2025-01-16:
In the meantime for anyone interested; 
Come debate Architecture with me for a Rust implementation I'm working on. 
I tried for a long time to get python3.5 to work, and couldn't do it, but have some hope again with Rust channels.

(Maybe once python release a more stable version running without GIL, but last i checked, it's broken AF lol.) 
It's been a lot of fun trying to learn it inside and out. 
https://github.com/DanEdens/EventGhost-Rust/blob/master/ARCHITECTURE.md

asheroto, 2025-01-17:
@kdschlosser you mentioned you never had access to the domains/website. Who did? I may be able to reach out to them and get more info.

asheroto, 2025-01-17:
Another user emailed me a 482 MB archive of the entire website using a Wayback machine downloader. Looks like I can restore the website itself and get it back up and running. The forum will require some work since I only having the generated HTML and not the underlying database.

kdschlosser, 2025-01-17:
The original author had the domain. Apparently he has finally let it go and now it has been bought by some holding company. So basically unless you want to spend some $$ you are not going to be able to buy it from the holding company.

I did have the .net domain but that has long since expired.

asheroto, 2025-01-17:
That's okay. Do you think the original author had a backup or forum database backup? What happened with them? I'm willing to contact them.

kdschlosser, 2025-01-17:
He has not done anything with it for at least m12 years. so no he is not going to have a backup. 

As I have already said. I am pretty sure that I have a backup it's just on a different computer that I am not able to get to very easily and it is going to require some time.

asheroto, 2025-01-17:
Ah, understood. I'll work with the Wayback Machine backup I have for now and whenever you get to it will work with that to see if anything can be salvaged. Thanks for your help.

kdschlosser, 2025-01-17:
personally I would not bother with a website and the forum. I would use GitHub to handle everything. It is able to do everything you need it to do. I would not compile an executable and add Eventghost to the python package index (pypi).

kdschlosser, 2025-01-17:
You are going to have a hell of a time keeping the spam bots out of the forum. It was a really large headache to deal with. I was spending a decent amount of money every month just to keep that from happening. There were tens of thousands per month that would occur or at least try to.

asheroto, 2025-01-17:
Makes sense. I think GitHub Pages may suffice for the documentation. I think I can probably port the documentation over there, but I would need contributor access, I believe. Otherwise I can just host them on our servers. The documentation is basically text-only so not much data being transferred.

I think the "shock" for everyone was that everything vanished without warning. The forums contained a lot of helpful information that may still be indexed by Google, or at least linked in other forums/Reddit/etc. Even if there was a page that said "due to data loss the forum is now inactive" I think that'd be better than nothing.

Rebuilding the forums is probably unlikely as I only have the generated HTML. I might be able to do a read-only copy if I fix the links, but the forum itself won't be active where someone could reply, unless I get a backup of the database itself. Preventing spam is easier now with technology such as Cloudflare Turnstile and its WAF products that we use. It doesn't block everything but quite a bit.

closed 2025-01-27 by asheroto

asheroto, 2025-01-27:
@kdschlosser, got the invitation, thank you. Check out this PR when you are free...

https://github.com/EventGhost/EventGhost/pull/429#issuecomment-2616718572