lvgl-micropython/lvgl_micropython #259: Compile error with esp32-s3

Liuzachary, 2025-01-14:
i was use t-display s3
```python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16 DISPLAY=ili9341```
```
ir/__/machine_hw_spi.c.obj -c /home/liuzc/lvgl_micropython/lib/micropython/ports/esp32/machine_hw_spi.c
/home/liuzc/lvgl_micropython/lib/micropython/ports/esp32/machine_hw_spi.c: In function 'machine_hw_spi_device_transfer':
/home/liuzc/lvgl_micropython/lib/micropython/ports/esp32/machine_hw_spi.c:327:53: error: expected ';' before '_'
327 |             transaction.flags |= SPI_TRANS_MODE_OCT _ SPI_TRANS_MULTILINE_ADDR;
|                                                     ^
/home/liuzc/lvgl_micropython/lib/micropython/ports/esp32/machine_hw_spi.c: In function 'machine_hw_spi_device_make_new':
/home/liuzc/lvgl_micropython/lib/micropython/ports/esp32/machine_hw_spi.c:645:23: error: 'm' undeclared (first use in this function)
645 |     if (dual || quad |m| octal) flags != SPI_DEVICE_HALFDUPLEX;
|                       ^
/home/liuzc/lvgl_micropython/lib/micropython/ports/esp32/machine_hw_spi.c:645:23: note: each undeclared identifier is reported only once for each function it appears in
/home/liuzc/lvgl_micropython/lib/micropython/ports/esp32/machine_hw_spi.c:645:39: error: statement with no effect [-Werror=unused-value]
645 |     if (dual || quad |m| octal) flags != SPI_DEVICE_HALFDUPLEX;
/home/liuzc/lvgl_micropython/lib/micropython/ports/esp32/machine_hw_spi.c:652:18: error: expected expression before ',' token
652 |         .flags = ,
|                  ^
cc1: some warnings being treated as errors
```
help please

EU1KY, 2025-01-14:
+1, the same happens here after today's pull from the 'main'.

jbalague, 2025-01-16:
I guess operator shoud be `|=`.

When changed I get other compile errors though:

```
In file included from /home/jbalague/esp32/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus.c:36:
/home/jbalague/esp32/lvgl_micropython/ext_mod/lcd_bus/esp32_include/rgb565_dither.h:12:19: error: return type defaults to 'int' [-Werror=implicit-int]
12 |     static inline rgb565_dither_pixel(uint8_t treshold_id, uint16_t *pixel)
|                   ^~~~~~~~~~~~~~~~~~~
/home/jbalague/esp32/lvgl_micropython/ext_mod/lcd_bus/esp32_include/rgb565_dither.h: In function 'rgb565_dither_pixel':
/home/jbalague/esp32/lvgl_micropython/ext_mod/lcd_bus/esp32_include/rgb565_dither.h:14:29: error: invalid operands to binary >> (have 'uint16_t *' {aka 'short unsigned int *'} and 'int')
14 |         *pixel = (((((pixel >> 8) & 0xF8) + red_thresh[treshold_id]) << 8) |
|                             ^~
/home/jbalague/esp32/lvgl_micropython/ext_mod/lcd_bus/esp32_include/rgb565_dither.h:15:29: error: invalid operands to binary >> (have 'uint16_t *' {aka 'short unsigned int *'} and 'int')
15 |                   ((((pixel >> 3) & 0xFC) + green_thresh[treshold_id]) << 3) |
|                             ^~
/home/jbalague/esp32/lvgl_micropython/ext_mod/lcd_bus/esp32_include/rgb565_dither.h:16:29: error: invalid operands to binary & (have 'uint16_t *' {aka 'short unsigned int *'} and 'int')
16 |                   ((((pixel & 0x1F) << 3) + blue_thresh[treshold_id]) >> 3)) ;
|                             ^
cc1: some warnings being treated as errors
```

I guess operator shoud be `|=`.

When changed I get other compile errors though:

```
In file included from /home/jbalague/esp32/lvgl_micropython/ext_mod/lcd_bus/esp32_src/rgb_bus.c:36:
/home/jbalague/esp32/lvgl_micropython/ext_mod/lcd_bus/esp32_include/rgb565_dither.h:12:19: error: return type defaults to 'int' [-Werror=implicit-int]
12 |     static inline rgb565_dither_pixel(uint8_t treshold_id, uint16_t *pixel)
|                   ^~~~~~~~~~~~~~~~~~~
/home/jbalague/esp32/lvgl_micropython/ext_mod/lcd_bus/esp32_include/rgb565_dither.h: In function 'rgb565_dither_pixel':
/home/jbalague/esp32/lvgl_micropython/ext_mod/lcd_bus/esp32_include/rgb565_dither.h:14:29: error: invalid operands to binary >> (have 'uint16_t *' {aka 'short unsigned int *'} and 'int')
14 |         *pixel = (((((pixel >> 8) & 0xF8) + red_thresh[treshold_id]) << 8) |
|                             ^~
/home/jbalague/esp32/lvgl_micropython/ext_mod/lcd_bus/esp32_include/rgb565_dither.h:15:29: error: invalid operands to binary >> (have 'uint16_t *' {aka 'short unsigned int *'} and 'int')
15 |                   ((((pixel >> 3) & 0xFC) + green_thresh[treshold_id]) << 3) |
|                             ^~
/home/jbalague/esp32/lvgl_micropython/ext_mod/lcd_bus/esp32_include/rgb565_dither.h:16:29: error: invalid operands to binary & (have 'uint16_t *' {aka 'short unsigned int *'} and 'int')
16 |                   ((((pixel & 0x1F) << 3) + blue_thresh[treshold_id]) >> 3)) ;
|                             ^
cc1: some warnings being treated as errors
```