lvgl-micropython/lvgl_micropython #260: Errors initilising SPI bus on ESP32-C6

0ut4t1m3, 2025-01-14:
Hey Kevin,

I'm having trouble running the port on a C6 board. Everything else seems to be working but I'm getting errors when trying to initialise the SPI bus. 
I know you made some changes to the SPI driver recently so I'm not sure if I'm doing something wrong in the constructors or if it's related to the relative weirdness of the SPI bus setup on the C6, I'm not having trouble on any of my S series devices. Would you mind seeing if you can spot the issue? 

```python
import lvgl as lv

import st7789
import lcd_bus
import machine

spi_bus = machine.SPI.Bus(host=1, mosi=15, miso=16, sck=18)
display_bus = lcd_bus.SPIBus(spi_bus=spi_bus, freq=40000000, dc=19, cs=14)

display = st7789.ST7789(
    data_bus=display_bus,
    display_width=128,
    display_height=128,     
    reset_pin=20,
    reset_state=st7789.STATE_LOW,
    backlight_pin=2,
    backlight_on_state=st7789.STATE_PWM,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_BGR,
    rgb565_byte_swap=True
)
display.init()
``` 

```
Traceback (most recent call last):
  File "<stdin>", line 19, in <module>
  File "st7789.py", line 60, in __init__
  File "display_driver_framework.py", line 231, in __init__
  File "display_driver_framework.py", line 243, in _init_bus
ValueError: SPI(1) invalid configuration
```

0ut4t1m3, 2025-01-14:
Apologies for some reason Github wasn't showing me the changes you've made today when I checked before raising this issue. I'm trying to compile a fresh build to see if this issue is still needed but I'm running into some build errors.

I had to make a couple of changes to micropy_updates/esp32/machine_hw_spi.c in order to get the latest commit to build:
Line 645 `    if (dual || quad || octal) flags |= SPI_DEVICE_HALFDUPLEX;`
Line 652 ```        .flags = flags,```
I'm just wildly guessing on that second line but it did compile without errors after adding that. Still no change to the original issue though, getting `ValueError: SPI(1) invalid configuration` when trying to init the bus.

0ut4t1m3, 2025-01-18:
I managed to fix the SPI error, I changed line 566 of ```lvgl_micropython\micropy_updates\esp32\machine_hw_spi.c``` to this:
```#if CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32S3 || CONFIG_IDF_TARGET_ESP32C3 || CONFIG_IDF_TARGET_ESP32C6``` and the bus is working fine now.

I am having some trouble with touch mirrored in both X and Y only in 0 and 180 orientations, 90 and 270 work fine.
I'm using the st7789 and cst816s drivers but can't see a way to set the touch directions. I don't want to mess with MADCTL values as the screen orientation is working properly. I tried rotating the screen before initilising the indev driver but it didn't change the mirrored touch.

0ut4t1m3, 2025-01-18:
I can get everything working properly by flipping the 0 and 180 values in the orientation table and then setting the screen rotation before I init touch. This gets everything aligned and I can rotate to any orientation normally.

``` python
### display setup here ###

display._ORIENTATION_TABLE = (192, 96, 0, 160)
display.set_rotation(lv.DISPLAY_ROTATION._180)

### touch setup here ###

display.set_rotation(lv.DISPLAY_ROTATION._0)
```
I'm happy to leave it here but please let me know if there's a better way of going about getting the two aligned.

kdschlosser, 2025-01-18:
The issue with orientation is all in the display that is being used and the way the data is being sent from the Display IC to the display panel.

you can have 4 different displays that are all attached to the same display IC and the orientation will behave differently for each one of them. This is the reason why the ability to override the default orientation table exists. to do exactly what you are doing.

kdschlosser, 2025-01-18:
Your orientation values are not correct which is the reason why you are having to apply the rotation before calling the init method and then having to call rotation again after the init. 

it looks like the 192 and the 0 needs to be swapped in their position in the table is all that needs to be done.

0ut4t1m3, 2025-01-19:
Thanks Kevin, you got me thinking more and noticed I was trying to set the orientation table after calling display.init(). I moved it above the init call and now everything works without needing the two rotation calls.

closed 2025-01-19 by 0ut4t1m3

kdschlosser, 2025-01-19:
good deal!! Having the ability to override the orientation table was a very important thing to have available due to the differences in display manufacturers. If you wanted to you could write a new initialization for your display. it's easy to do.

This is what you would do.

Create a new file named `_st7789_init__type_1.py` and in that file you put the following code.

```python
import sys

def init(self):
    self._ORIENTATION_TABLE = (192, 96, 0, 160)
    
    import _st7789_init
    _st7789_init.init(self)
    del _st7789_init
    del sys.modules['_st7789_init']

```

now when you call the init method from your main script you pass a `1`.

```
display.init(1)
```

and this will load the new init you made and set the orientation table properly.

The reason why ti is coded this way is so that the initialization commands do not stay resident in memory. No need to consume RAM for code that is run once and done. Simply having the commands typed up in the source file consumes memory when the source file is loaded. The code doesn't even need to run for some memory to be consumed. By having those lines of code in a module that gets unloaded from memory the ram gets freed. So once the orientation table gets set the code that sets it doesn't need to occupy memory any longer. The display driver will take care of importing the init module you make and then deleting it once it has run.

0ut4t1m3, 2025-01-20:
A very useful feature, thanks for the example.