lvgl-micropython/lvgl_micropython #261: Can I update ui in pin.irq callback？

hyansuper, 2025-01-15:
I notice that TaskHandler uses a timer to update lvgl tick. can I update UI in `Pin.irq` callback?
so timer and input pins are both interrupt sources, do they have priority and one can preempt another?

I did a very simple test and it seems fine. but not sure would it cause concurrency issue. 
```
th=task_handler.TaskHandler()
ui_init()

button = machine.Pin( 0, Pin.IN, Pin.PULL_UP)
button.irq(lambda _: label.set_y(0) , Pin.IRQ_FALLING)
```
I am aware that lvgl has button indev, but for simplicity and other reasons, I just want to use simpler approach if possible.
If it's wrong doing so, what modification can be done?

I really appreciate if you help me clearify

kdschlosser, 2025-01-15:
No you cannot update the UI from a PIN IRQ callback. The reason why is the inability to allocate memory from within an interrupt.

The best way to handle this kind of a thing is as follows.

```python
def pin_main_task_callback(pin):
    print('This is called in the context of the main task and not in an interrupt')
    print('State:', pin.value())

def pin_interrupt_callback(pin):
    machine.schedule(pin_main_task_callback_reference, pin_reference)
    

pin = machine.Pin(10, machine.Pin.INPUT)

# these next 2 must be done to keep from allocating memory
pin_reference = pin  
pin_main_task_callback_reference = pin_main_task_callback

# setup interrupt for pin here. Provide the pin_interrupt_callback function when doing that. 
```

In the example above you are now free to do whatever you want inside of the `pin_main_task_callback` function because the code in that function is not running within the context of the ISR. It is running in the main task context.

There is also another way to handle updating the UI based on input from a pin and that is to use LVGL's built in timer. The limitation with this is the duration of the input from the pin  must exceed 10 milliseconds. You would also want to pass a 10 when constructing the `task_handler.TaskHandler` instance. If you go lower than 10 milliseconds which you can do it could effect performance depending on how long it takes for the LVGL code to run including any timers that you have added. This is something that I have been working on in order to improve the performance. I used the RGB bus driver as a test and it is working. I have to update the rest of the lcd bus drivers to use the same kind of mechanics which would offload a lot of work being done to the second CPU core on the ESP32. This would free up more of the core that is running LVGL and a users application code.

kdschlosser, 2025-01-16:
the main task context is what the REPL loop runs in. It is basically what most of the python code runs in. I say most of because IRQ functions don't and if you start a thread the code supplied in the function for the thread doesn't either.

LVGL is not using any timers from MicroPython. It is unaware that MicroPython even exists...

With the ESP32 the only memory you can allocate is IRAM. You cannot allocate say PSRAM. The reason why is the access to the PSRAM will cause an interrupt to occur and you cannot cause an interrupt to be trigger from inside of an interrupt. In MicroPython you have zero control over what memory is being allocated. PLUS there is typically not a whole lot of IRAM left to be allocated. LVGL does not direct where memory allocation occurs. it is not able to. What gets used most of the time is PSRAM and that is where you end up with issues. 

Think of a FreeRTOS task as a thread on a PC. It behaves very much in the same manner. You do not want to access the same space of memory from multiple threads at the same time. That will lead to corruption. ISR's also have the same rules regarding memory access as threads do. So you need to also think of an ISR like a thread. An ISR literally tells the processor to stop doing what it is going and to run the code for the ISR. You cannot cause another ISR to be triggered from inside the code that an already running ISR is running. That means no memory allocation because the majority of the time allocation ends up in PSRAM and that is going to cause another ISR to get triggered. 

Go ahead and test it out.. create an ISR and inside of that ISR allocate some memory create a bytearray that is 100 characters long and place that bytearray into a global and see what happens.

```python

some_variable = None

def isr_handler(_):
    global some_variable
    some_variable = bytearray(list(range(100)))

# attach handler to pin code here...

```