lvgl-micropython/lvgl_micropython #266: Add option for --factory

dcmcshan, 2025-01-20:
I'd like to have a factory partition in addition to ota_0 and ota_1.  Is this something you'd like to add as command line option, or should I hack builder/esp32.py?

kdschlosser, 2025-01-21:
the partitions are broken up as

For an ota setup you have the following partitions. 
nvs
otadata
phy_init
ota_0
ota_1
vfs

and without OTA you have the following..
nvs
phy_init
factory
vfs

Those are only names.

nvs = data
otadata = data
phy_init = data
ota_0 = app
ota_1= app
vfs = data

nvs = data
phy_init = data
factory = app
vfs = data

so with an ota setup you are 2 partitions that are "factory" partitions.

app partitions cannot be loaded inside of MicroPython. The data that is written to them can only be done when flashing. They do not contain a file system so you would not be able to access the information that is on it if you add it. There really is no point to having it. You cannot write data to it other than when it is flashed.

I already built in a feature that you can use. Compile the firmware the way you want and then go into the `lib/micropython/ports/esp32/build-*` folder then locate the folder `config` and go into that. send me the `sdkconfig.h` file.

I also need to know what your build command is. Once I have those things I will set you up with a way to make all kinds of custom changes without needing to modify anything in the build routine.

dcmcshan, 2025-01-24:
So the idea in having the "factory" partition is that it is a "factory tested" firmware that customer can always revert to, if ota corrupts in some fashion.  I have sufficient flash to do this, and seems like a useful thing.  I am currently running a script that sed's into the builder/esp32.py file to make some modifications (spiffs, etc) so I can just add factory to that, but thought it might be a useful command line option.

kdschlosser, 2025-01-24:
with an OTA update that's what the second OTA partition is for. It automatically plays round robin between the 2 partitions in case of a failure when flashing it can always revert back to the last known good state.

You want a 3rd partition for the purposes of actually holding the original firmware that came with the device.
 As I have said I made a modification that will allow you to do this.  it is the --custom-board-path build argument. I am still working on the documentation for it.