lvgl-micropython/lvgl_micropython #267: Rotary encoder and push button as INDEV?

tetopik, 2025-01-22:
Hi, Kevin. 

This is not an issue, just asking a way on how to set a rotary enc and a button as the input device of the display.

Thanks!

kdschlosser, 2025-01-22:
I have not yet fully implemented this as I do not own one to test with.

AqeeAqee, 2025-10-01:
We can read A/B with machine.Pin and with it's IRQ in micropython code. And further optimise with asyncio. 
But may become a little jaggy when lvgl program get heavier.

Good new!
MicroPython has supported encoder feature inside recently, from v1.26.0, as machine.Encoder.
[https://github.com/micropython/micropython/releases/tag/v1.26.0](https://github.com/micropython/micropython/releases/tag/v1.26.0)

I tried with v1.26.1, looks very fast. But didn't test with lvgl-micropython togather, cause there are some API incompatible error when compiling it with MP v1.26.1

@kdschlosser Hope lvgl-micropython update to MP v1.26+ soon, thanks in advance!

AqeeAqee, 2025-10-01:
Ah, sorry for misunderstood.

The EncoderDriver class of lvgl-micropython is what you need?
[https://github.com/lvgl-micropython/lvgl_micropython/blob/main/api_drivers/py_api_drivers/frozen/indev/encoder_framework.py#L7](https://github.com/lvgl-micropython/lvgl_micropython/blob/main/api_drivers/py_api_drivers/frozen/indev/encoder_framework.py#L7)
which is based on LV_INDEV_TYPE_ENCODER indev in LVGL.
[https://docs.lvgl.io/master/details/main-modules/indev.html#encoder](https://docs.lvgl.io/master/details/main-modules/indev.html#encoder)

tetopik, 2025-10-01:
Yes, it is.
But as Kevin stated above, it is not fully implemented yet right now.

The encoder itself indeed working pretty well in C lvgl. Do you have any information regards if it's really work here?

AqeeAqee, 2025-10-01:
Actually, I haven't try this class, none of my project needed to navigate widgets or switch to edit mode, etc.
After a quit browse, it seems easy to use, it wrapped stuffs to adapt to LVGL indev. 
Basicly, create a class inherite it, deal with hardware to provide current encoder value & btn state, return as (enc, key), in _get_enc() override.