lvgl-micropython/lvgl_micropython #268: Screen Display Issue - Garbled Output on ESP32C3 with ST7789 Display

kk8900, 2025-01-22:
Hello, thank you for developing this project!
I successfully compiled the firmware using the following command:
`python3 make.py esp32 clean BOARD=ESP32_GENERIC_C3 --flash-size=8 DISPLAY=st7789 INDEV=ft6x36`

The development board I am using is the LC ESP32C3（https://lckfb.com/project/detail/lckfb-esp32-c3-va）. The test code is as follows:
`import lcd_bus
from micropython import const
import machine

# display settings
_WIDTH = const(240)
_HEIGHT = const(320)
_BL = const(2)
_RST = const(-1) #no such pin
_DC = const(6)

_MOSI = const(5)
_MISO = const(-1)#no such pin
_SCK = const(3)
_HOST = const(1)  # SPI2

_LCD_CS = const(4)
_LCD_FREQ = const(20000000)

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)

import st7789  # NOQA
import lvgl as lv  # NOQA

display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    #reset_pin=_RST,
    #reset_state=st7789.STATE_LOW,
    backlight_pin=_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_RGB,
    rgb565_byte_swap=True,
)

display.set_power(True)
display.init()
display.set_backlight(0)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)
`

However, I am unable to successfully drive the screen. The screen can light up, but it remains in a garbled state. How should I adjust the code?

tetopik, 2025-01-22:
I don't know exactly about your display, but my st7789 display expect the SPI polarity to be 1. You could try..

kk8900, 2025-01-23:
> I don't know exactly about your display, but my st7789 display expect the SPI polarity to be 1.我不确切知道你的显示器，但我的st7789显示器期望SPI极性为1。
> 
> See [#236 (comment)](https://github.com/lvgl-micropython/lvgl_micropython/issues/236#issuecomment-2573670548) 看到 [#236 (comment)](https://github.com/lvgl-micropython/lvgl_micropython/issues/236#issuecomment-2573670548)

Thank you for your response. The modified code is as follows: 
```
import lcd_bus
from micropython import const
import machine
import lcd_utils

spi_mode = lcd_utils.spi_polarity_phase_to_mode(1, 0)  # (polarity, phase)
polarity, phase = lcd_utils.spi_mode_to_polarity_phase(spi_mode)

# display settings
_WIDTH = const(240)
_HEIGHT = const(320)
_BL = const(2)
_RST = const(-1) #no such pin
_DC = const(6)

_MOSI = const(5)
_MISO = const(-1)#no such pin
_SCK = const(3)
_HOST = const(1)  # SPI2

_LCD_CS = const(4)
_LCD_FREQ = const(40000000)

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK,    
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
    spi_mode=spi_mode
)

import st7789  # NOQA
import lvgl as lv  # NOQA

display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    #reset_pin=_RST,
    #reset_state=st7789.STATE_LOW,
    backlight_pin=_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_RGB,
    rgb565_byte_swap=True,
)

display.set_power(True)
display.init()
display.set_backlight(0)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)

```
I set the polarity to 1, but the screen is still garbled.

kdschlosser, 2025-01-23:
OK so it looks like there is only a single SPI host that is able to be used by the user. That is SPI2 which is `host=1` It looks like that host is IOMUX only which means you need to use the hardware defined pins. Those pins are as follows...

* MOSI: 7
* MISO: 2
* SCK:  6
* CS: 10

You need to have the display connected tho those pins. If you do not have it connected to those pins you are going to need to use a bitbang (soft) driver to transmit the data to the display. I do not have a bitbang SPI driver written into this library. MicroPython does include a a bitbang driver `machine.SoftSPI` and that is able to be used if you write a custom display driver to use it.

kk8900, 2025-01-23:
> OK so it looks like there is only a single SPI host that is able to be used by the user. That is SPI2 which is `host=1` It looks like that host is IOMUX only which means you need to use the hardware [...]

I can run the following code successfully on MicroPython:
```
from machine import Pin
from machine import I2C
import time
from ft6x36 import FT6x36
from machine import Pin,SPI
import st7789,time

spi=SPI(1,baudrate=26666666, polarity=1, sck=Pin(3), mosi=Pin(5))

tft=st7789.ST7789(spi, 240, 320, dc=Pin(6, Pin.OUT), cs=Pin(4, Pin.OUT), inversion=True, rotation=0, options=0)
backlight = Pin(2, Pin.OUT)

tft.init()
backlight.value(0)
tft.fill(st7789.BLACK)

i2c = I2C(0, scl=Pin(1), sda=Pin(0))
print(i2c.scan())
t = FT6x36(i2c)
while True:
    xy = t.get_positions()
    
    if len(xy) > 0:
        if len(xy[0]) > 1:
            xy1 = xy
            
            print(xy)
            tft.fill_circle(xy[0][0], xy[0][1], 5, st7789.RED) 
```
These should be the correct SPI pins, not software SPI.
I have tried swapping the width and height, but it didn't work.

tetopik, 2025-01-23:
Try add this one after you initialize the display:
```
import task_handler
th = task_handler.TaskHandler()
```

kk8900, 2025-01-24:
> Try add this one after you initialize the display:
> 
> ```
> import task_handler
> th = task_handler.TaskHandler()
> ```

Thank you for your response.  It works now.

kdschlosser, 2025-01-24:
Good catch on that @tetopik

kdschlosser, 2025-01-24:
I think I am going to have to make that automatically run. The user would be able to shut it down easy enough if they wanted to do something different. A lot of people seem to miss that task handler.

kk8900, 2025-01-24:
> I think I am going to have to make that automatically run. The user would be able to shut it down easy enough if they wanted to do something different. A lot of people seem to miss that task handler.

It is working properly now. However, due to the performance limitations of the ESP32C3, the touch response is not very sensitive, and memory errors frequently occur. Could the firmware be appropriately optimized for the ESP32C3? 
Currently, the firmware size for the ESP32C3 is around 2.8M, which is twice as large as the MicroPython firmware.

kdschlosser, 2025-01-24:
paste the test file you are using. I want to see how things are set up.

kk8900, 2025-01-25:
> paste the test file you are using. I want to see how things are set up.

The slider cannot slide smoothly, and while it works the first time, it becomes very difficult to slide afterward.
It might be due to insufficient memory, requiring an RST (reset) before each run.
```
import lcd_bus
from micropython import const
import machine
import lcd_utils

spi_mode = lcd_utils.spi_polarity_phase_to_mode(1, 0)  # (polarity, phase)

# display settings
_WIDTH = const(240)
_HEIGHT = const(320)
_BL = const(2)
_RST = const(-1) #no such pin
_DC = const(6)

_MOSI = const(5)
_MISO = const(-1)#no such pin
_SCK = const(3)
_HOST = const(1)  # SPI2

_LCD_CS = const(4)
_LCD_FREQ = const(40000000)

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK,    
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
    spi_mode=spi_mode
)

import st7789  # NOQA
import lvgl as lv  # NOQA

display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)

import i2c  # NOQA
import task_handler  # NOQA
import ft6x36,pointer_framework  # NOQA

display.init()

i2c_bus = i2c.I2C.Bus(host=0, scl=1, sda=0, freq=400000, use_locks=False)
touch_dev = i2c.I2C.Device(bus=i2c_bus, dev_id=0x38, reg_bits=ft6x36.BITS)

indev = ft6x36.FT6x36(touch_dev,startup_rotation=pointer_framework.lv.DISPLAY_ROTATION._90)

# you want to rotate the display after the calibration has been done in order
# to keep the corners oriented properly.
display.set_rotation(lv.DISPLAY_ROTATION._90)

display.set_backlight(0)

th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)
slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)

```

tetopik, 2025-01-25:
> It might be due to insufficient memory, requiring an RST (reset) before each run.

I don't think that's the case. 
I've managed to run on an even larger display such as 480x320 ILI9488 (but the ts is xpt2046) using esp32c3 too, and the slider, toggle, spinner, all runs smoothly.

We'll wait @kdschlosser doing his `things`..

kk8900, 2025-01-25:
> I don't think that's the case. I've managed to run on an even larger display such as 480x320 ILI9488 (but the ts is xpt2046) using esp32c3 too, and the slider, toggle, spinner, all runs smoothly. [...]

Perhaps I should optimize my code. Could you provide some examples for reference?

tetopik, 2025-01-25:
I don't own such a touch driver, I'm sorry.

kdschlosser, 2025-01-25:
I am not understanding why you are doing this...

```
indev = ft6x36.FT6x36(touch_dev,startup_rotation=pointer_framework.lv.DISPLAY_ROTATION._90)

# you want to rotate the display after the calibration has been done in order
# to keep the corners oriented properly.
display.set_rotation(lv.DISPLAY_ROTATION._90)
```

I am confused by the 2 rotations taking place.

You would be better suited to use this code instead...
```
indev = ft6x36.FT6x36(touch_dev)

if not indev.is_calibrated:
    indev.calibrate()

# you want to rotate the display after the calibration has been done in order
# to keep the corners oriented properly.
display.set_rotation(lv.DISPLAY_ROTATION._90)
```

kdschlosser, 2025-01-25:
inherently LVGL has a lag issue with respect to the input drivers. The way they are coded doesn't place priority on input. There is a default refresh time of 33 milliseconds. so once every 33 milliseconds LVGL is going to poll for input and if there is input the changes are made to the UI and then another 33 milliseconds has to pass before those changes are seen on the display. This is not ideal if you are wanting to have the utmost priority place on user input.

I wrote code that works around this problem and places user input at the top in terms of priority. What the code does is the normal mechanics of LVGL works as usual and once there is input detected the UI gets updated based on that input and then instead of letting LVGL do it's normal thing I tell the display to refresh and I loop back around to check for input again. This loop will continue so long as there is user input.

To activate this feature you use the following code.

```python
indev = ft6x36.FT6x36(touch_dev)

if not indev.is_calibrated:
    indev.calibrate()

indev.enable_input_priority()

# you want to rotate the display after the calibration has been done in order
# to keep the corners oriented properly.
display.set_rotation(lv.DISPLAY_ROTATION._90)
```

Once the priority is enable it cannot be programmatically disabled. You would have to remove the line of code. 

Where doing this is the most noticeable is scrolling and how fast the UI responds to things like releasing a button.

kk8900, 2025-01-26:
> inherently LVGL has a lag issue with respect to the input drivers. The way they are coded doesn't place priority on input. There is a default refresh time of 33 milliseconds. so once every 33 milliseconds LVGL is going to poll for input and if there is input the changes are made to the UI and then another 33 milliseconds has to pass before those changes are seen on the display. This is not ideal if you are wanting to have the utmost priority place on user input.
> 
> I wrote code that works around this problem and places user input at the top in terms of priority. What the code does is the normal mechanics of LVGL works as usual and once there is input detected the UI gets updated based on that input and then instead of letting LVGL do it's normal thing I tell the display to refresh and I loop back around to check for input again. This loop will continue so long as there is user input.
> 
> To activate this feature you use the following code.
> 
> indev = ft6x36.FT6x36(touch_dev)
> 
> if not indev.is_calibrated:
>     indev.calibrate()
> 
> indev.enable_input_priority()
> 
> # you want to rotate the display after the calibration has been done in order
> # to keep the corners oriented properly.
> display.set_rotation(lv.DISPLAY_ROTATION._90)
> Once the priority is enable it cannot be programmatically disabled. You would have to remove the line of code.
> 
> Where doing this is the most noticeable is scrolling and how fast the UI responds to things like releasing a button.

Thank you very much for your response! After using your code, the touch sensitivity has improved significantly. However, the left and right directions of the touch are reversed (when I swipe to the right, the slider moves to the left). I tried changing `display.set_rotation` to `lv.DISPLAY_ROTATION._270`, but the issue remains the same. When I do not use `display.set_rotation`, the slider works normally. 

Could you please help me with this?

kdschlosser, 2025-01-26:
It sounds like the rotation table for the display needs to be adjusted for your display...

See if this code works for ya.

```python
import lcd_bus
from micropython import const
import machine
import lcd_utils

spi_mode = lcd_utils.spi_polarity_phase_to_mode(1, 0)  # (polarity, phase)

# display settings
_WIDTH = const(240)
_HEIGHT = const(320)
_BL = const(2)
_RST = const(-1) #no such pin
_DC = const(6)

_MOSI = const(5)
_MISO = const(-1)#no such pin
_SCK = const(3)
_HOST = const(1)  # SPI2

_LCD_CS = const(4)
_LCD_FREQ = const(40000000)

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK,
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
    spi_mode=spi_mode
)

import st7789  # NOQA
import lvgl as lv  # NOQA

st7789.ST7789._ORIENTATION_TABLE = (
    0x0,
    0x20,
    0x40 | 0x80,
    0x20 | 0x40 | 0x80
)

display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)

import i2c  # NOQA
import task_handler  # NOQA
import ft6x36,pointer_framework  # NOQA

display.init()

i2c_bus = i2c.I2C.Bus(host=0, scl=1, sda=0, freq=400000, use_locks=False)
touch_dev = i2c.I2C.Device(bus=i2c_bus, dev_id=0x38, reg_bits=ft6x36.BITS)

indev = ft6x36.FT6x36(touch_dev)

if not indev.is_calibrated:
    indev.calibrate()

indev.enable_input_priority()

# you want to rotate the display after the calibration has been done in order
# to keep the corners oriented properly.
display.set_rotation(lv.DISPLAY_ROTATION._90)

display.set_backlight(0)

th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)
slider = lv.slider(scrn)
slider.set_size(300, 50)
slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)

```

kk8900, 2025-01-26:
> It sounds like the rotation table for the display needs to be adjusted for your display...
> 
> See if this code works for ya.
> 
> import lcd_bus
> from micropython import const
> import machine
> import lcd_utils
> 
> spi_mode = lcd_utils.spi_polarity_phase_to_mode(1, 0)  # (polarity, phase)
> 
> # display settings
> _WIDTH = const(240)
> _HEIGHT = const(320)
> _BL = const(2)
> _RST = const(-1) #no such pin
> _DC = const(6)
> 
> _MOSI = const(5)
> _MISO = const(-1)#no such pin
> _SCK = const(3)
> _HOST = const(1)  # SPI2
> 
> _LCD_CS = const(4)
> _LCD_FREQ = const(40000000)
> 
> spi_bus = machine.SPI.Bus(
>     host=_HOST,
>     mosi=_MOSI,
>     miso=_MISO,
>     sck=_SCK,
> )
> 
> display_bus = lcd_bus.SPIBus(
>     spi_bus=spi_bus,
>     freq=_LCD_FREQ,
>     dc=_DC,
>     cs=_LCD_CS,
>     spi_mode=spi_mode
> )
> 
> 
> import st7789  # NOQA
> import lvgl as lv  # NOQA
> 
> st7789.ST7789._ORIENTATION_TABLE = (
>     0x0,
>     0x20,
>     0x40 | 0x80,
>     0x20 | 0x40 | 0x80
> )
> 
> display = st7789.ST7789(
>     data_bus=display_bus,
>     display_width=_WIDTH,
>     display_height=_HEIGHT,
>     backlight_pin=_BL,
>     color_space=lv.COLOR_FORMAT.RGB565,
>     color_byte_order=st7789.BYTE_ORDER_BGR,
>     rgb565_byte_swap=True,
> )
> 
> import i2c  # NOQA
> import task_handler  # NOQA
> import ft6x36,pointer_framework  # NOQA
> 
> display.init()
> 
> i2c_bus = i2c.I2C.Bus(host=0, scl=1, sda=0, freq=400000, use_locks=False)
> touch_dev = i2c.I2C.Device(bus=i2c_bus, dev_id=0x38, reg_bits=ft6x36.BITS)
> 
> indev = ft6x36.FT6x36(touch_dev)
> 
> if not indev.is_calibrated:
>     indev.calibrate()
> 
> indev.enable_input_priority()
> 
> # you want to rotate the display after the calibration has been done in order
> # to keep the corners oriented properly.
> display.set_rotation(lv.DISPLAY_ROTATION._90)
> 
> display.set_backlight(0)
> 
> th = task_handler.TaskHandler()
> 
> scrn = lv.screen_active()
> scrn.set_style_bg_color(lv.color_hex(0x000000), 0)
> slider = lv.slider(scrn)
> slider.set_size(300, 50)
> slider.center()
> 
> label = lv.label(scrn)
> label.set_text('HELLO WORLD!')
> label.align(lv.ALIGN.CENTER, 0, -50)

The same issue remains unresolved.

kdschlosser, 2025-01-26:
if you are holding the display when it doesn't have any rotation applied and then when it is applied what way do you have to turn the display so it appears correctly, clockwise or counter clockwise??

kk8900, 2025-01-27:
> if you are holding the display when it doesn't have any rotation applied and then when it is applied what way do you have to turn the display so it appears correctly, clockwise or counter clockwise??

When the program is running, the screen rotates 90 degrees clockwise, but the touch rotates 90 degrees counterclockwise. I think there is a problem with the touch rotation, which does not rotate clockwise like the screen.

tetopik, 2025-01-27:
If you already calibrate the display, run those code outside the `if` statement

kdschlosser, 2025-01-27:
Is the touch input working properly and then it goes backwards once the calibration has been run? Is that what's going on? If that is the case then there is something that needs to be fixed in the touch calibration.

When you flash the ESP32 with new firmware it will erase any previously stored calibration if a full erase is performed as part of the flashing. I believe that the command to flash that is given at the end of the build has the switches to erase the memory first.

kk8900, 2025-01-28:
> That flipped x or y touch direction did happen on my display too, it is easier to fix it by toggling these 2 `mirror` params either it's true or false:
> 
> ```
> if not indev.is_calibrated:
>         indev.calibrate()
>         indev._cal.mirrorX = False
>         indev._cal.mirrorY = False
>         indev._cal.save()
> ```
I set indev._cal.mirrorY to True, and the touch is working normally now. Thank you very much.

kk8900, 2025-01-28:
> Is the touch input working properly and then it goes backwards once the calibration has been run? Is that what's going on? If that is the case then there is something that needs to be fixed in the [...]

I have set indev._cal.mirrorY to True, and it is working normally now.
If you update the code, do I need to recompile?

kdschlosser, 2025-01-28:
OK so here is the issue with the touch. The display orientation is more often the problem than the touch drivers are. I am going to write a calibration program that will also handle getting the display orientation correct I have to work out how to mirror some text along both the X and Y axis to make it work the best.

closed 2025-02-01 by kk8900