lvgl-micropython/lvgl_micropython #269: Build failure for macOS

gmeader, 2025-01-22:
**Describe the bug**
1. Followed the compiling for macOS directions.
```
lvgl_micropython % brew install libffi
Warning: libffi 3.4.6 is already installed and up-to-date.
```
Gave the command:
`python3 make.py macOS`

Got the following error:

```
GEN build-standard/genhdr/qstr.i.last
modffi.c:40:10: fatal error: 'ffi.h' file not found
40 | #include <ffi.h>
|          ^~~~~~~
1 error generated.
```

**If you are having a problem compiling paste the build output below. (NO SCREENSHOTS)**

```shell
GEN build-standard/genhdr/qstr.i.last
modffi.c:40:10: fatal error: 'ffi.h' file not found
40 | #include <ffi.h>
|          ^~~~~~~
1 error generated.

make: *** [build-standard/genhdr/qstr.i.last] Error 1
make: *** Deleting file `build-standard/genhdr/qstr.i.last'
```

**Exact make and model number of the MCU that you are compiling for or the firmware is running on. **

* Make: MacBook Pro
* Model:

**For ESP32 MCU's The PSRAM and FLASH SPI type, quad SPI or octal SPI.**

* PSRAM: quad OR octal
* FLASH:  quad OR octal

I need to know the OS and OS version of the machine that compiled the binary. If using a VM then I need to know the OS and OS version the VM is running. (WSL == VM)
 - OS: MacOS Sonoma 14.6.1

**Build Command**

```shell
python3 make.py macOS
```

kdschlosser, 2025-01-22:
If that doesn't work then you need to find out where brew installed FFI. To do this use the following command

```
brew info libffi 
```

kdschlosser, 2025-01-22:
Your build command is also missing a few bits to it. It should be...

```
python3 make.py macOS DISPLAY=sdl_display INDEV=sdl_pointer
```

There is also an option to set the amount of memory that is allowed to by used by MicroPython. The default is 8mb and you can increase it if you like by using the `--heap-size={size in bytes}` build command, replacing `{size in bytes}` with the amount of memory you want to allocate to MicroPython

gmeader, 2025-01-22:
Thanks so much -  Got past the previous issue, now I get:
 ```
LINK build-standard/micropython
Undefined symbols for architecture arm64:
"_SDL_CreateRenderer", referenced from:
_sdl_init in sdl_bus.o
"_SDL_CreateTexture", referenced from:
_sdl_init in sdl_bus.o
_mp_lcd_sdl_set_window_size in sdl_bus.o
"_SDL_CreateWindow", referenced from:
_sdl_init in sdl_bus.o
"_SDL_DestroyRenderer", referenced from:
_sdl_del in sdl_bus.o
"_SDL_DestroyTexture", referenced from:
_sdl_del in sdl_bus.o
_mp_lcd_sdl_set_window_size in sdl_bus.o
"_SDL_DestroyWindow", referenced from:
_sdl_del in sdl_bus.o
"_SDL_GetWindowFlags", referenced from:
_process_event in sdl_bus.o
"_SDL_GetWindowID", referenced from:
_process_event in sdl_bus.o
"_SDL_InitSubSystem", referenced from:
_init_sdl in machine_sdl.o
_init_sdl in machine_sdl.o
_init_sdl in machine_sdl.o
"_SDL_PollEvent", referenced from:
_mp_lcd_sdl_poll_events in sdl_bus.o
_mp_lcd_sdl_poll_events in sdl_bus.o
"_SDL_Quit", referenced from:
_sdl_del in sdl_bus.o
"_SDL_QuitSubSystem", referenced from:
_deinit_sdl in machine_sdl.o
_deinit_sdl in machine_sdl.o
_deinit_sdl in machine_sdl.o
"_SDL_RenderClear", referenced from:
_sdl_tx_color in sdl_bus.o
"_SDL_RenderCopy", referenced from:
_sdl_tx_color in sdl_bus.o
"_SDL_RenderPresent", referenced from:
_sdl_tx_color in sdl_bus.o
"_SDL_SetTextureBlendMode", referenced from:
_sdl_init in sdl_bus.o
_mp_lcd_sdl_set_window_size in sdl_bus.o
"_SDL_SetWindowSize", referenced from:
_sdl_init in sdl_bus.o
_mp_lcd_sdl_set_window_size in sdl_bus.o
"_SDL_StartTextInput", referenced from:
_sdl_init in sdl_bus.o
"_SDL_UpdateTexture", referenced from:
_sdl_tx_color in sdl_bus.o
"_ffi_call", referenced from:
_ffifunc_call in modffi.o
"_ffi_closure_alloc", referenced from:
_mod_ffi_callback in modffi.o
"_ffi_prep_cif", referenced from:
_mod_ffi_callback in modffi.o
_make_func in modffi.o
"_ffi_prep_closure_loc", referenced from:
_mod_ffi_callback in modffi.o
"_ffi_type_double", referenced from:
_char2ffi_type in modffi.o
"_ffi_type_float", referenced from:
_char2ffi_type in modffi.o
"_ffi_type_pointer", referenced from:
_char2ffi_type in modffi.o
"_ffi_type_sint16", referenced from:
_char2ffi_type in modffi.o
"_ffi_type_sint32", referenced from:
_char2ffi_type in modffi.o
"_ffi_type_sint64", referenced from:
_char2ffi_type in modffi.o
"_ffi_type_sint8", referenced from:
_char2ffi_type in modffi.o
"_ffi_type_uint16", referenced from:
_char2ffi_type in modffi.o
"_ffi_type_uint32", referenced from:
_char2ffi_type in modffi.o
"_ffi_type_uint64", referenced from:
_char2ffi_type in modffi.o
"_ffi_type_uint8", referenced from:
_char2ffi_type in modffi.o
"_ffi_type_void", referenced from:
_char2ffi_type in modffi.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [build-standard/micropython] Error 1
```

kdschlosser, 2025-01-23:
I suspect that the x86 emulated version of brew is running and not the arm64 version. Lets find out of that is the case.

run the following commands and paste the output to me.

```
which -a brew
```

and 

```
brew config
```

kdschlosser, 2025-01-23:
You can also manually direct to the correct installation of homebrew by using the following..

```
/opt/homebrew/bin/brew install --force sdl2
/opt/homebrew/bin/brew install --force libffi
```

and then use the following commands to locate the correct paths to the compiled library and the header files..

```
/opt/homebrew/bin/brew info sdl2
/opt/homebrew/bin/brew info libffi 
```

Once you know the paths then you can export the pats into the variables by using..

```
export LDFLAGS="-L{libffi library path} -L{sdl2 library path}"
export CFLAGS="-I{libffi include path} -I{sdl2 include path}"
```
replacing the `{ }` and contents with what the contents says.

gmeader, 2025-01-23:
```
glenn@Glenns-MacBook-Pro lvgl_micropython % which -a brew
/opt/homebrew/bin/brew
glenn@Glenns-MacBook-Pro lvgl_micropython % brew config
HOMEBREW_VERSION: 4.4.17
ORIGIN: https://github.com/Homebrew/brew
HEAD: 55f399c10dc7e6cc07497bdfccdecc96329e3b57
Last commit: 4 days ago
Branch: stable
Core tap HEAD: b508def1f5a7f095edff6095e70b35cb7fa78e5c
Core tap last commit: 21 hours ago
Core tap JSON: 22 Jan 21:23 UTC
Core cask tap JSON: 22 Jan 21:23 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_MAKE_JOBS: 8
Homebrew Ruby: 3.3.7 => /opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.3.7/bin/ruby
CPU: octa-core 64-bit arm_firestorm_icestorm
Clang: 16.0.0 build 1600
Git: 2.39.5 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 8.7.1 => /usr/bin/curl
macOS: 14.6.1-arm64
CLT: 16.2.0.0.1.1733547573
Xcode: 16.2
Rosetta 2: false
```

gmeader, 2025-01-23:
Posting this for completeness:

```
glenn@Glenns-MacBook-Pro lvgl_micropython % /opt/homebrew/bin/brew install --force sdl2
/opt/homebrew/bin/brew install --force libffi
==> Downloading https://formulae.brew.sh/api/formula.jws.json
==> Downloading https://formulae.brew.sh/api/cask.jws.json
Warning: sdl2 2.30.11 is already installed and up-to-date.
To reinstall 2.30.11, run:
  brew reinstall sdl2
Warning: libffi 3.4.6 is already installed, it's just not linked.
To link this version, run:
  brew link libffi

glenn@Glenns-MacBook-Pro lvgl_micropython % brew link libffi
Warning: libffi is keg-only and must be linked with `--force`.
glenn@Glenns-MacBook-Pro lvgl_micropython % brew link libffi --force
Linking /opt/homebrew/Cellar/libffi/3.4.6... 11 symlinks created.

glenn@Glenns-MacBook-Pro lvgl_micropython % brew info sdl2
==> sdl2: stable 2.30.11 (bottled), HEAD
Low-level access to audio, keyboard, mouse, joystick, and graphics
https://www.libsdl.org/
Installed
/opt/homebrew/Cellar/sdl2/2.30.11 (94 files, 6.4MB) *
  Poured from bottle using the formulae.brew.sh API on 2025-01-22 at 13:24:46
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/s/sdl2.rb
License: Zlib
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 68,337 (30 days), 197,543 (90 days), 808,703 (365 days)
install-on-request: 11,893 (30 days), 33,516 (90 days), 128,876 (365 days)
build-error: 82 (30 days)

glenn@Glenns-MacBook-Pro lvgl_micropython % brew info libffi
==> libffi: stable 3.4.6 (bottled), HEAD [keg-only]
Portable Foreign Function Interface library
https://sourceware.org/libffi/
Installed
/opt/homebrew/Cellar/libffi/3.4.6 (18 files, 764.4KB) *
  Poured from bottle using the formulae.brew.sh API on 2025-01-22 at 13:23:45
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/lib/libffi.rb
License: MIT
==> Options
--HEAD
	Install HEAD version
==> Caveats
libffi is keg-only, which means it was not symlinked into /opt/homebrew,
because macOS already provides this software and installing another version in
parallel can cause all kinds of trouble.

For compilers to find libffi you may need to set:
  export LDFLAGS="-L/opt/homebrew/opt/libffi/lib"
  export CPPFLAGS="-I/opt/homebrew/opt/libffi/include"

For pkg-config to find libffi you may need to set:
  export PKG_CONFIG_PATH="/opt/homebrew/opt/libffi/lib/pkgconfig"
==> Analytics
install: 29,194 (30 days), 83,099 (90 days), 432,400 (365 days)
install-on-request: 5,663 (30 days), 17,601 (90 days), 90,694 (365 days)
build-error: 2 (30 days)
```

here are the export commands I used:

```
glenn@Glenns-MacBook-Pro lvgl_micropython %  export LDFLAGS="-L/opt/homebrew/opt/libffi/lib -L/opt/homebrew/Cellar/sdl2/2.30.11/lib"
glenn@Glenns-MacBook-Pro lvgl_micropython % export CPPFLAGS="-I/opt/homebrew/opt/libffi/include -I/opt/homebrew/Cellar/sdl2/2.30.11/include"
```
glenn@Glenns-MacBook-Pro lvgl_micropython % python3 make.py macOS DISPLAY=sdl_display INDEV=sdl_pointer     

SUCCESS:
It completed the build without error.
Running ./lvgl_micropy_macOS gives the expected REPL prompt.
Entering: import demo.py displays a window with a slider! (demo.py is the Python code example for Unix/macOS from the README)

kdschlosser, 2025-01-23:
OK so are you saying you got it up and running?

gmeader, 2025-01-23:
Yes. Running fine.

closed 2025-01-23 by gmeader