lvgl-micropython/lvgl_micropython #270: Unix version demo code displays a blank window.

gmeader, 2025-01-23:
**Describe the bug**
I successfully built lvgl_micropy_unix on a freshly installed Lubuntu 24.10 machine,
`python3 make.py unix DISPLAY=sdl_display INDEV=sdl_pointer`

I copied the Python code example for Unix/macOS into a file named demo.py

running` ./lvgl_micropy_unix demo.py` displays a window for an instant, then it disappears.
Adding a time.sleep(15) at the end of the demo code keeps the window displayed for 15 sec.
However the window is empty with black background, It has a title bar labelled "LVGL MP".

**Expected behavior**
Expected to see a window with a slider displayed in it.

I need to know the OS and OS version of the machine that compiled the binary. If using a VM then I need to know the OS and OS version the VM is running. (WSL == VM)
 - OS: Lubuntu 24.10

**Build Command**

```shell
python3 make.py unix DISPLAY=sdl_display INDEV=sdl_pointer
```

gmeader, 2025-01-23:
SUCCESS Thanks! 
I run `.lvgl_micropy_unix`
I get the expected REPL prompt.
I enter: `import demo.py` 
_(the Python code example for Unix/macOS from the README)_

A window appears with a working slider!

_(I do get an error message: ImportError: no module named 'demo.py')_

kdschlosser, 2025-01-23:
so it's working as it should?

kdschlosser, 2025-01-23:
You need to have the script in the same folder where the binary is located in order to import it.

gmeader, 2025-01-23:
it is working as it should.
The demo.py script is in the same folder as the binary lvgl_micropy_unix (/build) and that is my current working directory.
The demo.py code runs properly when give the import command, but the ImportError message also appears in the REPL. 
It seems to be a spurious message or just a warning or ?