lvgl-micropython/lvgl_micropython #272: ESP32C3 display initialization problems

maro48m, 2025-01-28:
Hi,

After some pause i've tried once again to build and run lvgl_micropython on my device.
This time, after running command: ```python3 make.py --optimize-size --flash-size=4 esp32 BOARD=ESP32_GENERIC_C3 DISPLAY=gc9a01 INDEV=cst816s``` and uploading bin to device micropython is launching and giving prompt.
Now when i try to launch code:
```
import lvgl as lv
import gc9a01
import lcd_bus
from machine import SPI, Pin
from micropython import const
import task_handler  # NOQA
import time  # NOQA
import gc

# display settings

WIDTH = const(240)
HEIGHT = const(240)

SPI_HOST = const(1)
SPI_SCK = const(6)
SPI_MOSI = const(7)
SPI_MISO = const(-1)

LCD_FREQ = const(80000000)
LCD_DC = const(2)
LCD_CS = const(10)
LCD_RST = const(-1)
LCD_BACKLIGHT = const(3)

TOUCH_CS = const(-1)
TOUCH_FREQ = const(10000000)

LCD_PIXEL_CLOCK_HZ    = const (20 * 1000 * 1000)
LCD_BK_LIGHT_ON_LEVEL  = const (1)
LCD_BK_LIGHT_OFF_LEVEL = const (0)

BUFFER_SIZE = WIDTH * HEIGHT // 20

print(1)
# Initialize the SPI bus
spi_bus = SPI.Bus(
    host = SPI_HOST,
    mosi = SPI_MOSI,
    miso = SPI_MISO,
    sck = SPI_SCK
)
print(2)
# Initialize the display bus
display_bus = lcd_bus.SPIBus(
    spi_bus = spi_bus,
    dc = LCD_DC,
    cs = LCD_CS,
    freq=LCD_FREQ ,
    spi_mode = 0
)
print(3)
fb1 = display_bus.allocate_framebuffer(BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
fb2 = display_bus.allocate_framebuffer(BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

gc.collect()
print(4)
display = gc9a01.GC9A01(
    data_bus=display_bus,
    display_width=WIDTH,
    display_height=HEIGHT,
    frame_buffer1=fb1,
    frame_buffer2=fb2,
    backlight_pin=LCD_BACKLIGHT,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

print(5)
display.set_power(True)

display.init()
display.set_backlight(100)
lv.init()

th = task_handler.TaskHandler(500)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)  # 0x000000 , 0x0000FF

label = lv.label(scrn)
label.set_text("HELLO WORLD!!")
label.set_style_text_color(lv.color_hex(0x0000FF), 0)
label.align(lv.ALIGN.CENTER, 0, 0)
```
sometimes code runs and displays red "HELLO WORLD!"
but sometimes it crashes with Core 0panic:
```
A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

LVGL MicroPython 
IDF version : c9763f62
Machine     : ESP32C3 module with ESP32C3

Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42098582  RA      : 0x42097f9e  SP      : 0x3fca76e0  GP      : 0x3fc95400  
TP      : 0x3fc396a8  T0      : 0x4005890e  T1      : 0x0000000f  T2      : 0x3fcaa7d0  
S0/FP   : 0x3fca7710  S1      : 0x3fca93b0  A0      : 0x3fcaae74  A1      : 0x00008522  
A2      : 0x3fca94b0  A3      : 0x3fca94c0  A4      : 0x3fca94b0  A5      : 0x00008528  
A6      : 0x00000000  A7      : 0x0000884a  S2      : 0x3c24cbbc  S3      : 0x3fca9444  
S4      : 0x3c2556e0  S5      : 0x3fca9440  S6      : 0x00000000  S7      : 0x3fc98f30  
S8      : 0x3fca93c4  S9      : 0x3c256446  S10     : 0x00001036  S11     : 0x3c25642e  
T3      : 0x00008d0a  T4      : 0x00000001  T5      : 0x00008d02  T6      : 0x00000007  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00008528  
MHARTID : 0x00000000  

Stack memory:
3fca76e0: 0x3c2108c0 0x3fca94b0 0x00008522 0x3fcaae74 0x000010c8 0x3fca76e4 0x00008528 0x00008528
3fca7700: 0x00000003 0x0000884a 0x3fca7740 0x3fca7740 0x00000000 0x3fca9440 0x00000000 0x3fcaae74
3fca7720: 0x3fca93c8 0x3c256442 0x00001036 0x3fca94b0 0x00008d0a 0x00000001 0x3fca7770 0x42041130
3fca7740: 0x00001036 0x3fca93b0 0x000000f0 0x000000f0 0x00000000 0x3fca9440 0x3c2556e0 0x000000f0
3fca7760: 0x3c2110c4 0x3fca93c4 0x3fca77a0 0x4202bbdc 0x00000000 0x42041110 0x3fca93cc 0x00000002
3fca7780: 0x00000002 0x00000002 0x000000f0 0x000000f0 0x00000000 0x3fca9440 0x3fca77d0 0x42007b60
3fca77a0: 0x3fca93cc 0x00000000 0x00000002 0x3c20f294 0x00000000 0x3fca93cc 0x00000000 0x3c20f294
3fca77c0: 0x3c24cbbc 0x3fca93b0 0x3c256447 0x4038413c 0x3fca93f0 0x3c245504 0x3fca9344 0x00000000
3fca77e0: 0x00008d0a 0x3fca9438 0x00000012 0x00000007 0x3fca7910 0x3c241d85 0x40383082 0x3fca78a0
3fca7800: 0x3fca93b0 0x3c24cbbc 0x3fca9444 0x00000001 0x3fca9440 0x3fca9344 0x3fc98f30 0x3c245000
3fca7820: 0x3fcaad7e 0x00000fca 0x3fcaad75 0x3fca77d0 0x00000000 0x3fca9688 0x3fca9560 0x3fcaad75
3fca7840: 0x00000fca 0x3fcaad7e 0x000000a0 0x00000020 0x3fca9344 0x0000000c 0x00000001 0x3fca94f0
3fca7860: 0x0000008c 0x3fca93b0 0x3fca78a0 0x420e07be 0x3fca78b4 0x3fca9688 0x3fca9624 0x3fc98f30
3fca7880: 0x3fca78b4 0x3fca9340 0x3fca962c 0x00000068 0x0000000c 0x00000000 0x3fca8fd0 0x420e6498
3fca78a0: 0x3c24cbbc 0x3fca78d8 0x3fca9040 0x00000000 0x00000000 0x3fca9e60 0x3fca8fd0 0x3fca8fd0
3fca78c0: 0x00000011 0x00000000 0x3fca78b4 0x420e7d00 0x00000000 0x3fca9688 0x3fca9560 0x3fca968c
3fca78e0: 0x3c24cbbc 0x3fca9610 0x3fcaad7f 0x4038413c 0x3fca94a0 0x3c245504 0x3fca94a0 0x00000000
3fca7900: 0x3fca94a0 0x3fca9680 0x00000020 0x4210d1c4 0x3fca79f0 0x00000001 0x40383082 0x3fca79c0
3fca7920: 0x3fca9610 0x3c24cbbc 0x3fca968c 0x00000000 0x3fca9688 0x00000000 0x3fc98f30 0x3c245000
3fca7940: 0x00000000 0x00000000 0x00000000 0x3fca78f0 0x3fcaac00 0x00000000 0x00000010 0x00000000
3fca7960: 0x00000000 0x00000000 0x0000007c 0x0000001a 0x00000000 0x00000000 0x00000000 0x3fc98f8c
3fca7980: 0x00000068 0x3fca9610 0x3fca79c0 0x420e07be 0x3fcaac00 0x3fca9470 0x00000080 0x00000004
3fca79a0: 0x00000003 0x00000041 0x000000e9 0x00000001 0x00000001 0x00000000 0x3fca9480 0x4211c9a6
3fca79c0: 0x3fc9e000 0x3fc9e000 0x00000000 0x600c2194 0x3fca7a5c 0x4211c8c4 0x4211caac 0x00000000
3fca79e0: 0x00000005 0x00000041 0x3fcaab2c 0x00000000 0x00000000 0x00000000 0x4211c962 0x3fca7a64
3fca7a00: 0x00000041 0x00000001 0x00000001 0x00000005 0x00000041 0x00000003 0x00000004 0x00000080
3fca7a20: 0x00000000 0x00000000 0x00000000 0x3fca79c0 0x00000001 0x3fca7a58 0x00000005 0x3c239000
3fca7a40: 0x00000001 0x00000002 0x3c225000 0x4211cbb4 0x00000000 0x00000000 0x42010080 0x00800001
3fca7a60: 0x0000007a 0x3fca7a5c 0x4211c8c4 0x4211caac 0x00000020 0x00000002 0x3fca9450 0x3fca7a00
3fca7a80: 0x3fcb89bc 0x3fc98f30 0x00000001 0x420fc000 0x3c235000 0x3c235000 0x3fc9e000 0x420fc33c
3fca7aa0: 0x3fcb89bc 0x3fc98f30 0x3fca89bc 0x420fc4e2 0x00000000 0x00000000 0x00000000 0x00000000
3fca7ac0: 0x00000000 0x00000000 0x00000000 0x3fca7ab0 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 64dfa59b3

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380d88
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5820,len:0xfd8
load:0x403cc710,len:0x948
load:0x403ce710,len:0x2c20
entry 0x403cc710
```

sometimes it throws error about not enough memory, but it reports missing 1GB of memory...:
```Traceback (most recent call last):
  File "<stdin>", line 75, in <module>
  File "display_driver_framework.py", line 187, in __init__
MemoryError: memory allocation failed, allocating 1070239896 bytes```

both problems occur in ```display = gc9a01.GC9A01([...]``` line

kdschlosser, 2025-01-28:
There are other that are running using the C3 and they do not have any issues. I am inclined to think that the problem might be a defective board of the amount of memory being detected is incorrect. The build system has no control over the detection of the amount of memory. The only thing that is controllable is the size of the partitions made on the flash storage and the speed of the memory/flash and the read/write mode that is being used... Perhaps the correct read/write mode is not being used??

I would need a link to the website where you purchased the board from so I can do some further investigation.

kdschlosser, 2025-01-28:
also if you could decode the stack trace it would be helpful for me to debug it.

maro48m, 2025-01-28:
Hi Kevin,

I'm using one of these round devices: https://aliexpress.com/item/1005006165850712.html

I've never had any issues with it on esp-idf or vanilla micropython.

I'll post decode results later when i'll be able to do it.

Thank you very much for quick response and your support!

maro48m, 2025-02-02:
Kevin,
i've tested some more - and this panick error happens every 2nd display initialization. Sometimes right when i try to create display object, sometimes after esp reboot. So i don't know if there is point in further investigating: since after such crash device reboots and is able to launch display this errors is not visible to end user.

More troublesome are errors with huge memory allocation during display init.
Right now when memory error occurd during display init ic can persist  until complete poweroff of device.

i've tried decoding addresses from stack dump, but most of them gives the same response:
```
0x3fca98f0: ?? ??:0
```
i've left addreses that point to anything :
```
0x42098674: node_set_prev at lv_ll.c:?
0x42098090: lv_ll_ins_head at ??:?
0x3fc97200: __global_pointer$ at ??:?
0x40380001: _vector_table at ??:?
0x3c265c04: ?? at /home/marek/lvgl_micropython/lib/micropython/py/objtuple.c:227
0x40384014: mp_execute_bytecode at /home/marek/lvgl_micropython/lib/micropython/py/vm.c:957 (discriminator 1)
0x3c26fe1c: ?? at /home/marek/lvgl_micropython/lib/micropython/ports/esp32/machine_pin.c:413
0x42041222: lv_display_create at ??:?
0x420e9b22: mp_obj_instance_store_attr at /home/marek/lvgl_micropython/lib/micropython/py/objtype.c:808 (discriminator 1)
 (inlined by) mp_obj_instance_attr at /home/marek/lvgl_micropython/lib/micropython/py/objtype.c:817 (discriminator 1)
0x3c2756ac: ?? at /home/marek/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/frozen_content.c:27937
0x4202bcce: mp_lv_display_create at lv_mp.c:?
0x42041202: lv_display_create at ??:?
0x3c265804: ?? at /home/marek/lvgl_micropython/lib/micropython/py/objmodule.c:114
0x42007c52: lv_fun_builtin_var_call at lv_mp.c:?
0x420ea336: mp_obj_instance_make_new at /home/marek/lvgl_micropython/lib/micropython/py/objtype.c:356
0x420ebf00: mp_load_method at /home/marek/lvgl_micropython/lib/micropython/py/runtime.c:1229
0x40383f60: mp_execute_bytecode at /home/marek/lvgl_micropython/lib/micropython/py/vm.c:1042 (discriminator 1)
0x42117778: mp_compile_to_raw_code at /home/marek/lvgl_micropython/lib/micropython/py/compile.c:3658 (discriminator 1)
 (inlined by) mp_compile at /home/marek/lvgl_micropython/lib/micropython/py/compile.c:3673 (discriminator 1)
0x4038390a: mp_execute_bytecode at /home/marek/lvgl_micropython/lib/micropython/py/vm.c:285
0x3c26cb94: ?? at /home/marek/lvgl_micropython/lib/micropython/py/vmentrytable.h:38
0x3c2654dc: ?? at /home/marek/lvgl_micropython/lib/micropython/py/objgenerator.c:39
0x3c265000: ?? at /home/marek/lvgl_micropython/lib/micropython/py/objexcept.c:351
0x420e2ff8: fun_bc_call at /home/marek/lvgl_micropython/lib/micropython/py/objfun.c:270
0x4212baa0: parse_compile_execute at /home/marek/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:120
0x4212ba1a: parse_compile_execute at /home/marek/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:78 (discriminator 1)
0x4212bcd0: pyexec_raw_repl at /home/marek/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:510
0x4212b96a: mp_reader_stdin_readbyte at /home/marek/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:203
0x4212bba4: mp_reader_stdin_close at /home/marek/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:234
0x42104000: st_word at /home/marek/lvgl_micropython/lib/micropython/lib/oofatfs/ff.c:621
 (inlined by) f_mkfs at /home/marek/lvgl_micropython/lib/micropython/lib/oofatfs/ff.c:5723
0x3fc9ad30: ?? at /home/marek/lvgl_micropython/lib/micropython/py/mpstate.c:33
0x4210477a: vprintf_null at /home/marek/lvgl_micropython/lib/micropython/ports/esp32/main.c:101
0x42104932: mp_task at /home/marek/lvgl_micropython/lib/micropython/ports/esp32/main.c:171 (discriminator 1)
```
My recent stack dump that i've used to decode:
```
A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

LVGL MicroPython 
IDF version : c9763f62
Machine     : ESP32C3 module with ESP32C3

Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42098674  RA      : 0x42098090  SP      : 0x3fca9540  GP      : 0x3fc97200  
TP      : 0x3fc3b538  T0      : 0x4005890e  T1      : 0x0000000f  T2      : 0x000088b2  
S0/FP   : 0x3fca9570  S1      : 0x3fcab284  A0      : 0x3fcacd04  A1      : 0x000083c2  
A2      : 0x3fcacdc0  A3      : 0x3fcacfb0  A4      : 0x3fcacdc0  A5      : 0x000085a3  
A6      : 0x00000002  A7      : 0x00000002  S2      : 0x3c26cb94  S3      : 0x00000000  
S4      : 0x3fcab204  S5      : 0x3c2756ac  S6      : 0x00000068  S7      : 0x3c2654dc  
[... 28 lines cut ...]
3fca9800: 0x00000041 0x00000005 0x3c245000 0x00000004 0x00000001 0x3fcab2b0 0x00000041 0x4212baa0
3fca9820: 0x3fca989c 0x4212b96a 0x4212bba4 0x00000001 0x00000000 0x00000005 0x3fcaca14 0x00000000
3fca9840: 0x00000000 0x00000000 0x4212ba1a 0x00000041 0x3fca98a4 0x00000001 0x00000004 0x3c245000
3fca9860: 0x00000005 0x00000041 0x3c259000 0x00000080 0x3c255000 0x3c255000 0x3c255000 0x3fca9820
3fca9880: 0x00000003 0x00000002 0x00000001 0x4212bcd0 0x00000000 0x00000000 0x3f010080 0x0080fa01
3fca98a0: 0x0000006f 0x3fca989c 0x4212b96a 0x4212bba4 0x00000020 0x00000002 0x3fcaaea0 0x3fc9ad00
[... cut ...]
3fca98e0: 0x3fca0000 0x4210477a 0x3fca0000 0x42104932 0x00000000 0x00000000 0x00000000 0x3fca98f0
3fca9900: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9920: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 06bef061b

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380f28
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5820,len:0xfd8
load:0x403cc710,len:0x948
load:0x403ce710,len:0x2c20
entry 0x403cc710
```

kdschlosser, 2025-02-02:
there is a single review for your display and hat person is saying that the thing doesn't work right. You may have gotten bit by a counterfeit MCU. Code is going to execute the same way each and every single time you run it. If runs fine one time and then you reboot it and it crashes it's not going to be the code. There is something that is changing as the result of the reboot that is causing the crash.

what are you using to connect to the ESP32?

maro48m, 2025-02-02:
Link i've provided was't exact shop i bought my esp from, rather to show shat kind of device i have.
I've never encountered similar errors in vanilla micropython or esp-idf (also in project using lvgl 8).

I'm connecting over usb.

kdschlosser, 2025-02-02:
You are the only person encountering this crash running on the C3 which leads me to believe it is the board and not the code. If others had the same problem with it crashing in the same manner then I could see where it might be a code issue. Get what I am saying? I know that you can end up with all kinds of issues including the memory error with the oddball number if the setup for the memory is not correct. I don't have a datasheet for the display to be able to determine if that is the issue or not and a lot of the time these Chinese manufacturers do not include the information that is needed in order to check. The only way to check is by having to look through some demo code if there is any available or by trial and error.

kdschlosser, 2025-02-02:
what are you using to connect to the ESP32?, as in software not the type of connection.

maro48m, 2025-02-03:
I totally understand your point of view.
i've been using jetbrains intellij with micropython plugin, but not it's not working with newest IDE version, so i've switched to Thonny.

kdschlosser, 2025-02-03:
Just to let you know. You MUST perform a hard reset when rebooting the ESP32. Thonny does soft resets when it restarts the ESP32 and I am willing to bet that is where your problem is stemming from. So when you see the REPL restart press the reset button on your ESP32 and reconnect Thonny.

This is a known issue and the reason why I had asked what you were using for software to connect to the ESP32.

This is the reason why it sometimes works and sometimes doesn't. I am willing to bet it only crashes immediately after Thonny soft resets the ESP32

maro48m, 2025-02-03:
I never had such issue - ive used ESP32 for some time now and don't remember crash after soft boot.
I usually reboot device with Ctrl+D from repl. 
Also my ESP32C3 family (because problem is on all my C3 devices- i have two with gc9a01 and 3 dev boards), when i hammer Ctrl+D in thonny it simply reboots.

I've stumbled today on mpy with gc9a01 incorporated:
```https://github.com/russhughes/gc9a01_mpy```
i've flashed available firmware, took some examples, modified pin numbers - and this one works without ANY problems...
i can reboot as much as i want - it always runs and screen is initialised.

I have ILI9341 display somewhere, i'll connect it to C3 dev board and see how this works...

kdschlosser, 2025-02-03:
It has to do with the hardware not being reset when a soft reset is performed.

MicroPython was written in a manner that only resets the hardware when their wrapper code is used. I have no way to know when a soft reset is occurring to be able to gracefully shutdown the display drivers. Unfortunately you are not able to do it in your code either because there is no way for your code to know when a software reset is going to occur. I tried to correct the issue more than once and it has not worked. There are no problems if you use the reset button after a soft reset is performed. Sorry to disappoint but the end game is I have spent a large amount of time trying to correct a problem that is actually not in my code. It is in MicroPython's and also in Thonny's where the issues are stemming from. I have tried to get it corrected in either place without any luck in getting it done.

That driver is a bitbang driver and it is going to be horribly slow. It also doesn't include LVGL at all so it only has primitive rendering ability which all has to be done in python code which is going to be really slow.

I am sorry that pressing the reset button is not a good enough work around for ya. There are other ways of connecting to the board other than Thonny. Like using a terminal emulator and mpremote to send files to the display. These would not have an issue because when the terminal emulator connects it automatically performs a hard reset of the ESP32 and you would need to disconnect the terminal emulator in order to use mpremote to transfer files.

Another option is for you to update the code in Thonny so it will disconnect and then reconnect after it exits the raw repl. That would cause the ESP32 to perform a hard reset.

maro48m, 2025-02-03:
Kevin,
I'm really sorry i have wasted your time.

Thank you very much for clarification how thonny works.
Hard reset is not big problem for me - i just have to start using it.

Best regards,
Marek

kdschlosser, 2025-02-03:
Slow down there speedy.

I just rewrote some of the connection bits in thonny for ya...

download the attached file and remove the `.txt` extension from the end of the filename and drop the file into `thonny/plugins/micropython`. That should force a hard reset instead of a soft reset.

[serial_connection.py.txt](https://github.com/user-attachments/files/18641098/serial_connection.py.txt)

kdschlosser, 2025-02-03:
I should have been able to perform a hard reset simply by wiggling the DTR line in the serial connection. The problem there is MicroPython is hijacking the hardware CDC stack and dropping in the use of tinyusb to handle it. And or coarse they didn't add in handling of the DTR and CTS lines in order to handle the hardware reset or to change the boot mode. This is forcing the need to disconnect and reconnect and even that might not work properly because of how they have done it. It's really infuriating with some of the decisions that have been made with respects to MicroPython. I am hoping that the terminal disconnect will restart the thing. I have my fingers crossed on that one. I do not have a C3 to test with.

maro48m, 2025-02-03:
> Slow down there speedy.
> 
> I just rewrote some of the connection bits in thonny for ya...
> 
> download the attached file and remove the `.txt` extension from the end of the filename and drop the file into `thonny/plugins/micropython`. That should force a hard reset instead of a soft reset.
> 
> [serial_connection.py.txt](https://github.com/user-attachments/files/18641098/serial_connection.py.txt)
> 
> give it a go and see if it works.

I didn't have micropython dir in plugins.
I've created one and dropped your file.
Unfortunately this did not helped - after display init and soft boot  device crashes.
Let's leave it - i don't have problem with resetting in thonny. I was thinking that these crashes also will occur when esp is running it's code without ide connection.

closed 2025-03-04 by maro48m