lvgl-micropython/lvgl_micropython #273: Elecrow 5 inch with ESP

spienz, 2025-01-28:
Hy Kevin,

with the newest Build from today my code will crash.

Only some snippets

TOUCH_DEVICE = I2C.Device(
    I2C_BUS,
    dev_id=gt911.I2C_ADDR,
    reg_bits=gt911.BITS
)
indev = gt911.GT911(TOUCH_DEVICE, startup_rotation=lv.DISPLAY_ROTATION._0)

without this line the display is running but logically no touch

MPY: soft reboot
Traceback (most recent call last):
  File "<stdin>", line 138, in <module>
  File "gt911.py", line 83, in __init__
  File "gt911.py", line 103, in hw_reset
  File "gt911.py", line 50, in _write_reg
  File "i2c.py", line 123, in write_mem
  File "i2c.py", line 82, in writeto_mem
OSError: [Errno 116] ETIMEDOUT

any ideas

Regards

Sebastian

>>>

kdschlosser, 2025-01-28:
need to see the bus initialization code,

spienz, 2025-01-28:
I2C_BUS = I2C.Bus(
    host=1,
    scl=_CTP_SCL,
    sda=_CTP_SDA,
    freq=400000,
    use_locks=False
)

kdschlosser, 2025-01-28:
set host to 0

spienz, 2025-01-29:
Hy,

changeing it into 0 won't fix it.

Reagards

Sebastian

kdschlosser, 2025-01-29:
won't or didn't?

kdschlosser, 2025-01-29:
This is part of the problem... `MPY: soft reboot`. If you are using Thonny that is the cause of the issue. MicroPython only resets the software and not the hardware when it performs a soft reboot. If the hardware doesn't get reset you can end up with all kinds of problems. You need to perform a hard reset when rebooting.

spienz, 2025-01-29:
Hey Kevin,

both :-) , i know the problem with the soft reboot, for testing it's simpler, but also with an hard reset (reset switch) it isn't working.

an i2c bus scan is also not working. This code was also working with an older build.

```python
from i2c import I2C
import gt911
import lcd_bus
import task_handler
import lvgl as lv  # NOQA
import rgb_display
#import lv_utils
import time

I2C_BUS = I2C.Bus(
    host=0,
    scl=20,
    sda=19,
    freq=400000,
    use_locks=False
)

TOUCH_DEVICE = I2C.Device(
    I2C_BUS,
    dev_id=gt911.I2C_ADDR,
    reg_bits=gt911.BITS
)

print('Scan i2c bus...')
devices = I2C_BUS.scan()

if len(devices) == 0:
  print("No i2c device !")
else:
  print('i2c devices found:',len(devices))
  

  for device in devices:  
    print("Decimal address: ",device," | Hexa address: ",hex(device))
    #print("\n")  

    

```

kdschlosser, 2025-01-29:
I have not changes the I2c at all. it should be working without an issue..

is your clone up to date?

spienz, 2025-01-29:
yes it is, im always do a new clone bevor compiling

kdschlosser, 2025-01-30:
I just tested the gt911 driver and I2C and it is working using the current repo code. I am not sure what is causing it to not work for you. maybe a connection problem to the ESP32 if you are wiring the display and ESP32 together? Maybe your pin definitions are not correct? I don't know. I am only able to tell you that the code does work. I am not sure where to go from here...

spienz, 2025-01-30:
Hey,

i did no changes at all. The Display is fix mounted.

here are some screen shots, with older an the actuel binary. Maybe that help's.

Only the file size of the bin is completely different. new 2472 KB old 3618 KB.

With old I2c scan is working as you can see in the screenshot.

old bin

[image]

[image]

new bin

[image]
[image]

kdschlosser, 2025-01-30:
Never mind. I know what your problem is. They made a change upstream in MicroPython that is causing the problem. They are hijacking pins 19 and 20 for USB CDC even if you don't have the pins allocated for that purpose.

add `--enable-cdc-repl=n --enable-jtag-repl=n` to your build command and that should fix your issue.

spienz, 2025-01-30:
Hy,

you are perfect man....

Works.....

But touch is hanging if i swap the silder?

Any idea?

Regards

Sebastian

kdschlosser, 2025-01-31:
explain "swap the slider" I do not understand what you are referring to.

If you want to get better sensitivity to touch input especially when scrolling or using a slider I recommend turning on touch priority. You do that by calling the `enable_input_priority()` that is in the touch driver.... 

```python
indev.enable_input_priority()
```

You can make that call after you have done calibration (if it is being done).

spienz, 2025-01-31:
Hy,

swap = move in my understanding, but my Englisch shut not be the "standard".

I will give it a try on Sunday. 

THX

kdschlosser, 2025-01-31:
swap = take part in an exchange of. "I am going to swap out a part on my car..."

getznerh, 2025-02-02:
Hi Kevin and all MP freinds! I have the same ESP 5 inch board and i had the same problems with I2C. Everything ist OK now and i have to appreciate the tremedous work of Kevin! Thank you very much! I woll try now with the SD Card driver and will convert the standard MP samples from Elecrow.. Thanx again and regards Hans

kdschlosser, 2025-02-02:
Hey Thanks for the plug, glad to see that this information was able to help you as well. It stinks that MicroPython makes assumptions based on the MCU being used. Not all boards use the USB pins for an actual USB connection. MicroPython is written so if the board supports using USB directly then it will shanghai those pins and nothing else is able to use them unless you go in and manually edit MicroPython or create a custom board. Personally I feel that it should be defaulted so they are not used for USB and those that have USB can turn it on if they want but that is just me I guess.. I added a command line parameter to make it easier for a user to turn it on and off which does make things a lot easier.

spienz, 2025-02-03:
Hy,

yes it works, SD Card is working too.

THX

closed 2025-02-03 by spienz