lvgl-micropython/lvgl_micropython #276: Suggestion - add 3 more dependencies to linux builds

gitcnd, 2025-02-03:
I cannot **_believe_** how easy this was to install and build - your "make.py" is amazing!

For "completeness", here's the only 3 things you overlooked:

When "make" isn't present, `build-essential` needs to be installed.
When "venv" fails, `python3.10-venv` needs to be installed.
When "cmake" isn't present `cmake` needs to be installed.

That, in a nutshell, is all that's needed on a fresh Linux(or WSL) machine (I used Ubuntu 22.04.5 LTS on WSL1 running on Windows 10N)

```
sudo apt install build-essential
sudo apt install python3.10-venv
sudo apt install cmake
git clone https://github.com/lvgl-micropython/lvgl_micropython
cd lvgl_micropython
python3 make.py esp32 BOARD=ESP32_GENERIC BOARD_VARIANT=SPIRAM DISPLAY=st7796 INDEV=xpt2046
dir build/lvgl_micropy_ESP32_GENERIC-SPIRAM-4.bin
-rwxrwxrwx  2464304 2025-02-03 12:15 build/lvgl_micropy_ESP32_GENERIC-SPIRAM-4.bin
```

kdschlosser, 2025-02-03:
TY for the comments about the script. I did my best to make it as easy as possible for a user to compile and get up and running. The dependencies are what people get all tangled up with usually and this is because of differences in environments. With my script taking care of it is will ensure that the environment gets set up in the manner that it needs to in order to compile successfully. 

If a user is not able to compile it easily and get a working firmware in the end they will walk away. The big thing with Python is it is supposed to be easy to use and that got lost somewhere when MicroPython was written. If compiling needs to be done by the user then that is also part of the use and it should be easy to do. Unfortunately that is not the case with MicroPython and I did my best to correct it. 

As far as the issues you had...

* venv: should come with the esp-idf I am not sure why you had an issue there. 
* build-essential: already listed in the README.md file as needing to be installed by the user prior to running the script. 
                          https://github.com/lvgl-micropython/lvgl_micropython/blob/2a6c2e0d798b5f839aeb9f3de0d6d6919aebde4b/README.md?plain=1#L178
* cmake: already listed in the README.md file as needing to be installed by the user prior to running the script. 
              https://github.com/lvgl-micropython/lvgl_micropython/blob/2a6c2e0d798b5f839aeb9f3de0d6d6919aebde4b/README.md?plain=1#L179

kdschlosser, 2025-02-03:
OH, and the reason why those requirements needs to be installed by the user is more often than not the package manager needs to be run using sudo in order to install the packages. I am not able to spawn a subprocess as sudo unless the user runs the build script using sudo which would not be a smart thing for a user to do. So I list them as requirements in the readme instead.

mark-hoy, 2025-02-11:
The Python venv is needed IMHO for Linux development, not everyone will be doing esp32 or other cross compiler embedded work.

kdschlosser, 2025-02-11:
venv should be getting installed by the ESP-IDF. That is the ONLY thing that uses it. Compiling for unix or any other port venv is not used at all so there is no need to install it. IDK why it didn't get installed by the ESP-IDF when you compiled.

The comment I made was in reference to not being able to install using a package manager because the package managers need to be run using sudo most of the time and that needs a user to enter a password.