lvgl-micropython/lvgl_micropython #285: i2c seems broken on BOARD=ESP32_GENERIC_S3 DISPLAY=rgb_display INDEV=GT911

gitcnd, 2025-02-04:
**Describe the bug**
My gt911 touch chip works and can be seen in arduino and also plain micropython.
When build under RedHat 9.3 as follows however:-
`python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=rgb_display INDEV=GT911 ; echo esp32_8048S043`
it is no longer possible to see my device.  Something must be messing with either the i2c bus or pins 19 or 20 I guess?

**Expected behavior**
i2c can be found

**Images**

This is what it looks like in normal micropython:
[image]
(The gt911 i2c address 0x5D is 93 in decimal)
```
esp32-lcd-10:/$ scani2c --scl=20 --sda=19
Scanning I2C devices on bus0 scl=gpio20 sda=gpio19
I2C devices found: [93]
```

This is what I see with lvgl_micropython
[image]

```shell
esp32-lcd-10:/$ scani2c --scl=20 --sda=19
Scanning I2C devices on bus0 scl=gpio20 sda=gpio19
No I2C devices found.
```

**Exact make and model number of the MCU that you are compiling for or the firmware is running on. **

* Make: esp32s3
* Model: wroom-1

This board here: https://de.aliexpress.com/item/1005005100043791.html

**For ESP32 MCU's The PSRAM and FLASH SPI type, quad SPI or octal SPI.**

* PSRAM: quad 8mb
* FLASH:  16mb

I need to know the OS and OS version of the machine that compiled the binary. If using a VM then I need to know the OS and OS version the VM is running. (WSL == VM)
 - OS: Red Hat Enterprise Linux release 9.3 (Plow)

**Build Command**

```shell
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=rgb_display INDEV=GT911 ; echo esp32_8048S043
```
The display works fine - just the touch-chip cannot be seen...

gitcnd, 2025-02-04:
Happy to dig and and try to fix this if you want to give me some clues about where to start!

kdschlosser, 2025-02-04:
You need to add `--enable-cdc-repl=n --enable-jtag-repl=n` to your build command.

gitcnd, 2025-02-05:
Those --switches didn't fix the problem, but manually disabling what they do did fix it:

https://github.com/lvgl-micropython/lvgl_micropython/discussions/287

closed 2025-02-05 by gitcnd

kdschlosser, 2025-02-05:
They do fix the problem. If you are making modifications to the MicroPython core files that is going to mess up the build script so things will not work properly. I strongly advise against making any changes to any files in MicroPython. You can also create a custom board and use the `--custom-board` build command if you want to modify any of the internal MicoPython macros or set specific ESP-IDF config settings.