lvgl-micropython/lvgl_micropython #286: ESP32S3 N8R8 Unable to run st7789 normally

HenTaiCJN, 2025-02-04:
**Describe the bug**
st7789 has no response and no error

I have encountered the same problem in other st7789 drivers. I can work normally by specifying polarity and phase parameters as 1. However, machine.SPI.Bus cannot use these two parameters.

I tried to create SPI using the following code and  received an error. 
```py
machine.SPI(1, baudrate=40000000, polarity=1,phase=1, sck=machine.Pin(14), mosi=machine.Pin(13), miso=None)
```

```shell
TypeError: can't create 'SPI' instances
```

**For ESP32 MCU's The PSRAM and FLASH SPI type, quad SPI or octal SPI.**

* PSRAM: 8MB octal
* FLASH:  8MB quad

kdschlosser, 2025-02-04:
```
import lcd_utils

mode = lcd_utils.spi_polarity_phase_to_mode(1, 1)

spi_bus = machine.SPI.Bus(host=1, sck=14, mosi=13, miso=-1)

lcd_bus.SPIBus(
    spi_bus=spi_bus,
    cs=-1,
    freq=40000000,    
    mode=mode
)
```

kdschlosser, 2025-02-04:
polarity and phase are bit shifted values of mode.

polarity 0, phase 0 = mode 0 = 00 (binary)
polarity 0, phase 1 = mode 1 = 01 (binary)
polarity 1, phase 0 = mode 2 = 10 (binary)
polarity 1, phase 1 = mode 3 = 11 (binary)

mode from polarity and phase
```
mode = (polarity << 1) | phase
```

and the reverse is
```
polarity = mode >> 1
phase = mode & 0x1
```

HenTaiCJN, 2025-02-05:
> ```
> import lcd_utils
> 
> mode = lcd_utils.spi_polarity_phase_to_mode(1, 1)
> 
> spi_bus = machine.SPI.Bus(host=1, sck=14, mosi=13, miso=-1)
> 
> lcd_bus.SPIBus(
>     spi_bus=spi_bus,
>     cs=-1,
>     freq=40000000,    
>     mode=mode
> )
> ```

Thank you for your reply. I tried to add the `mode` parameter, but it prompted that there were too many parameters
```py
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
    mode=3
)
```

```shelll
TypeError: extra keyword arguments given
```

I looked up the definitions under `lvgl_cicropython/lcd_bus-pyi` and found that they use `polarity` and `phase`. Then I tried using these two parameters in the code, but the result was still `TypeError: extra keyword arguments given`. This is very confusing

```py
class SPIBus:

    def __init__(
        self,
        *,
        spi_bus: _spi.SPI.Bus,
        freq: int,
        dc: int,
        cs: Optional[int] = -1,
        polarity: int = 0,
        phase: int = 0,
        firstbit: int = _spi.SPI.MSB,
        cs_high_active: bool = False,
        dc_low_on_data: bool = False,
    ):
        ...
```

kdschlosser, 2025-02-05:
My bad on that it's `spi_mode` not `mode`

HenTaiCJN, 2025-02-05:
> My bad on that it's `spi_mode` not `mode`

Well, the parameter name seems to be correct now, but I'm not sure if it works. Because my screen is still unresponsive.
Here is my code, hoping to provide reference and assistance.  

I remember that width and height are also important parameters, maybe there is no matching data in the st7789 driver?  

By the way, how can I make lvgl a sub project of Micropython? I would like to add lvgl to my existing Micorpython project, but lv_finding_macropython does not support projects larger than mpy v1.20.0

```py
import lcd_bus
from micropython import const
import machine

# display settings
_WIDTH = const(172)
_HEIGHT = const(320)
_BL = const(21)
_RST = const(46)
_DC = const(9)

_MOSI = const(13)
_MISO = const(-1)
_SCK = const(14)
_HOST = const(1)  # SPI2

_LCD_CS = const(10)
_LCD_FREQ = const(40000000)

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK,    
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
    spi_mode=3
)

import st7789  # NOQA
import lvgl as lv  # NOQA

display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_RGB,
    rgb565_byte_swap=True,
)

import task_handler  # NOQA

display.set_power(True)
display.init()
display.set_backlight(100)

th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)

```

kdschlosser, 2025-02-05:
OooOOooO You are getting bit because you are not passing the reset to the display driver..

```
display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    reset_pin=_RST,
    reset_state=st7789.STATE_HIGH ,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_RGB,
    rgb565_byte_swap=True,
)
```

You might need to change the reset state to `STATE_LOW` if it is still not working.

By not having the pin defined it is left in a floating state and the LCD could be interpreting that as a reset signal which is latching the display into a never ending reset cycle. When supplying the reset pin to the driver the driver attaches to the GPIO and sets the output on the pin to be the opposite of what you supply for reset state. This removes the "float"

HenTaiCJN, 2025-02-05:
> OooOOooO You are getting bit because you are not passing the reset to the display driver..
>
> ```
> display = st7789.ST7789(
>     data_bus=display_bus,
>     display_width=_WIDTH, [...]

Thank you so much, it was successfully run after being specified as `STATE_LOW`

kdschlosser, 2025-02-05:
The other issue is you are using SPI2 (host=1) which is a hardware SPI designation. That means you MUST use hardware assigned pin numbers..

CS=10
MOSI=11
SCK=12
MISO=13

if you are not connected to those pins you need to use SPI3 (host=2)

Your current pin values are

CS = 10
MOSI = 13
SCK = 14
MISO = -1

MISO you can set to -1 without any issue
CS can also be set to -1 if you meet the following conditions:
* CS line from the display is connected to ground
* the display is the only device on the SPI Bus.

kdschlosser, 2025-02-05:
OK so it is working now...
Good...

If you have the ability to connect to the hardware pins you will get a faster transfer rate. When not using the hardware SPI pins the transfer speed is limited to something like 26mhz where as the hardware pins are 80 mhz. It's a very large difference.

closed 2025-02-05 by kdschlosser