lvgl-micropython/lvgl_micropython #291: New .toml build feature produces non-working binaries.

gitcnd, 2025-02-05:
I hope you don't mind I made this into an "issue", so progress is all in one place.  It's a continuation from https://github.com/lvgl-micropython/lvgl_micropython/discussions/287#discussioncomment-12069416 and https://github.com/lvgl-micropython/lvgl_micropython/discussions/282#discussioncomment-12067827 etc.

**Describe the bug**
This makes a .bin which does not work:
`python3 make.py --toml=/dl/l3/custom_board_and_toml_examples/esp32_8048S070C.toml `
The .bin from this does work:
`python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=rgb_display INDEV=GT911 --enable-cdc-repl=n --enable-jtag-repl=n`

**Expected behavior**
should work

**If your problem is running the binding please past the code you are using below.***

cat /dl/l3/custom_board_and_toml_examples/esp32_8048S070C.toml

```toml
[MCU.esp32]
BOARD = "ESP32_GENERIC_S3"
BOARD_VARIANT = "SPIRAM_OCT"
octal_flash = true
flash_size = 8
enable_jtag_repl = "n"
enable_cdc_repl = "n"
enable_uart_repl = "y"
uart_repl_bitrate = 115200

[I80Bus.display_bus]
data0 = 15
data1 = 7
data2 = 6
data3 = 5
data4 = 4
data5 = 9
data6 = 46
data7 = 3
data8 = 8
data9 = 16
data10 = 1
data11 = 14
data12 = 21
data13 = 47
data14 = 48
data15 = 45
freq = 12000000
hsync = 39
vsync = 40
de = 41 
pclk = 42
hsync_front_porch = 210
hsync_back_porch = 16
hsync_pulse_width = 30
hsync_idle_low = true
vsync_front_porch = 22
vsync_back_porch = 10
vsync_pulse_width = 13
vsync_idle_low = true
de_idle_high = false
pclk_idle_high = false
pclk_active_low = true

[I2C.Bus.i2c_bus]
host = 0
scl = 19
sda = 20
freq = 400000

[I2C.Device.indev_device]
bus = "i2c_bus"
dev_id = "gt911.I2C_ADDR"
reg_bits = "gt911.BITS"

[RGBDisplay.display]
data_bus = "display_bus"
display_width = 800
display_height = 480
backlight_pin = 2
color_space = "lv.COLOR_FORMAT.RGB565"

[display.init]
params = []

[GT911.indev]
device = "indev_device"

[display.set_color_inversion]
params = [true]

[display.set_rotation]
params = ["lv.DISPLAY_ROTATION._90"]

[display.set_backlight]
params = [100]

[task_handler.TaskHandler]
params=[]
```

**Exact make and model number of the MCU that you are compiling for or the firmware is running on. **

* Make: Sunton esp32_8048S070C
* Model: esp32-s3

**For ESP32 MCU's The PSRAM and FLASH SPI type, quad SPI or octal SPI.**

* PSRAM: quad SPI
* FLASH:  ?

I need to know the OS and OS version of the machine that compiled the binary. If using a VM then I need to know the OS and OS version the VM is running. (WSL == VM)
 - OS: Ubuntu 22.04.5 LTS

**Build Command**

```shell
[cnd@RoG: Err:1~/Downloads]$ esptool.py --port $PORT write_flash 0x0 ~/Downloads/lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-8-rgb_display-gt911-esp32_8048S070rebuild.bin  # Works    ^C^C
[cnd@RoG: Err:1~/Downloads]$ esptool.py --port $PORT write_flash 0x0 ~/Downloads/lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-8_OCTFLASH-esp32_8048S070C-tomlnew.bin # Does not boot 
esptool.py v4.7.0
Serial port /dev/ttyS6
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 34:85:18:6c:27:98
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00265fff...
Compressed 2513872 bytes to 1481515...
Wrote 2513872 bytes (1481515 compressed) at 0x00000000 in 130.1 seconds (effective 154.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
[cnd@RoG:~/Downloads]$ perl ~/Downloads/mcu_serial.pl -port $PORT --reset

resetting...
Use Control-] ( ^] ) to quit terminal emulator. (dtr= rts=)
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fce3810,len:0xfec
load:0x403c9700,len:0x4
load:0x403c9704,len:0xab0
load:0x403cc700,len:0x2cc0
entry 0x403c9880
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fce3810,len:0xfec
load:0x403c9700,len:0x4
load:0x403c9704,len:0xab0
load:0x403cc700,len:0x2cc0
entry 0x403c9880

[cnd@RoG:~/Downloads]$ dir | grep 8048S070
-rwxrwxrwx 1 cnd cnd        1249 2025-02-05 20:21  esp32_8048S070C.toml*
-rwxrwxrwx 1 cnd cnd     2513872 2025-02-05 20:48  lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-8_OCTFLASH-esp32_8048S070C-toml.bin*
-rwxrwxrwx 1 cnd cnd     2516544 2025-02-05 21:36  lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-8-rgb_display-gt911-esp32_8048S070rebuild.bin*
-rwxrwxrwx 1 cnd cnd     2513872 2025-02-06 09:04  lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-8_OCTFLASH-esp32_8048S070C-tomlnew.bin*
-rwxrwxrwx 1 cnd cnd     2516544 2025-02-06 09:37  lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-8-rgb_display-gt911-esp32_8048S070rebuild2.bin*

[cnd@RoG:~/Downloads]$ esptool.py --port $PORT write_flash 0x0 lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-8-rgb_display-gt911-esp32_8048S070rebuild2.bin # Works
esptool.py v4.7.0
Serial port /dev/ttyS6
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 34:85:18:6c:27:98
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00266fff...
Compressed 2516544 bytes to 1483622...
Wrote 2516544 bytes (1483622 compressed) at 0x00000000 in 130.3 seconds (effective 154.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
[cnd@RoG:~/Downloads]$ perl ~/Downloads/mcu_serial.pl -port $PORT --reset

resetting...
Use Control-] ( ^] ) to quit terminal emulator. (dtr= rts=)
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1108
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbdc
load:0x403cc700,len:0x2f4c
entry 0x403c989c
stack: 656 out of 15360
GC: total: 64000, used: 3760, free: 60240, max new split: 8257536
 No. of 1-blocks: 17, 2-blocks: 3, max blk sz: 64, max free sz: 3430
Touch Product id: 911
Touch Firmware version: 0x1060
Touch Vendor id: 0x0
Touch resolution: width=800, height=480
LVGL (9.2.2) MicroPython (1.24.1) Binding compiled on 2025-02-06; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
>>> 

```

** Complete logs from the Build process **
[build_log_failed.txt](https://github.com/user-attachments/files/18680691/build_log_failed.txt)
[build_log_worked.txt](https://github.com/user-attachments/files/18680693/build_log_worked.txt)

I'm happy to help - let me know anything you need me to do.

I also have some questions, if this is the right place to ask them:

1. When we get this working, I will produce a pile of working .toml files for all the boards I've done so far - e.g. esp32_8048S070C.toml - do you want me to contribute these to lvgl_micropython?  If yes, what folder(s) do you want them in, how do you want them named, and what other info is suitable for putting into them (e.g. comments? datasheet urls? 3rd party pin references? (non-affiliated) store links? Additional sections for non-lvgl pins, like LEDs, IMU, ambient-light sensors, speakers, SD cards, qwiik headers, etc etc?
2. Is it worth trying to make these .toml files usable by python itself?  We are a mere "whisker" away from having a solution where code written for one device, can run _unmodified_ on some other device - if it was able to read from these .toml files in order to know pin arrangements, screen sizes, and initialisation processes.
3. If we want to ship a sample .py demo file to go with the .toml files, what's the best idea for what to put into that demo?

kdschlosser, 2025-02-06:
OH and another thing you are not doing is you are not using the flash command from the build. Don't do that. Use the one that is output from the build. There are additional parameters that are in the command that you are not using.

kdschlosser, 2025-02-06:
also PLEASE do not edit the output at all. I am seeing difference in the output that can only happen from you altering it.

as an example..
```
[cnd@RoG: Err:1~/Downloads]$ esptool.py --port $PORT write_flash 0x0 ~/Downloads/lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-8-rgb_display-gt911-esp32_8048S070rebuild.bin  # Works    ^C^C

[cnd@RoG: Err:1~/Downloads]$ esptool.py --port $PORT write_flash 0x0 ~/Downloads/lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-8_OCTFLASH-esp32_8048S070C-tomlnew.bin # Does not boot
```

but later on down this is what is seen as being used for the flash command, Not the one above. 
```
esptool.py --port $PORT write_flash 0x0 lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-8-rgb_display-gt911-esp32_8048S070rebuild2.bin # Works
```

while I know that this specifically is moot it is showing that you are going in and editing things. Please do not do that.

kdschlosser, 2025-02-06:
You are also editing the output binary file names. Please don't do that either as the file name contains information that helps to identify things. If there is something off in the build it can show up in the binary name and would help identify the problem.

kdschlosser, 2025-02-06:
the only way I am going to be able to diagnose the issue is if everything is 100% untouched. Just build and flash using the command that is provided at the end of the build. Don't do anything else..

gitcnd, 2025-02-06:
OK - I got the .toml builder working.  The issue seemed to be that I copied the ESP32_GENERIC_S3 example .toml file and built on that, without deleting the things at the start which I didn't understand.  I removed them all, and the following now builds OK:

cat /lvgl/06/custom_board_and_toml_examples/esp32_8048S070Cb.toml 
```
[MCU.esp32]
BOARD = "ESP32_GENERIC_S3"
BOARD_VARIANT = "SPIRAM_OCT"
INDEV = "gt911"
DISPLAY = "rgb_display"
enable_jtag_repl = "n"
enable_cdc_repl = "n"

[I80Bus.display_bus]
data0 = 15
data1 = 7
data2 = 6
data3 = 5
data4 = 4
data5 = 9
data6 = 46
data7 = 3
data8 = 8
data9 = 16
data10 = 1
data11 = 14
data12 = 21
data13 = 47
data14 = 48
data15 = 45
freq = 12000000
hsync = 39
vsync = 40
de = 41 
pclk = 42
hsync_front_porch = 210
hsync_back_porch = 16
hsync_pulse_width = 30
hsync_idle_low = true
vsync_front_porch = 22
vsync_back_porch = 10
vsync_pulse_width = 13
vsync_idle_low = true
de_idle_high = false
pclk_idle_high = false
pclk_active_low = true

[I2C.Bus.i2c_bus]
host = 0
scl = 19
sda = 20
freq = 400000

[I2C.Device.indev_device]
bus = "i2c_bus"
dev_id = "gt911.I2C_ADDR"
reg_bits = "gt911.BITS"

[RGBDisplay.display]
data_bus = "display_bus"
display_width = 800
display_height = 480
backlight_pin = 2
color_space = "lv.COLOR_FORMAT.RGB565"

[display.init]
params = []

[GT911.indev]
device = "indev_device"

[display.set_color_inversion]
params = [true]

[display.set_rotation]
params = ["lv.DISPLAY_ROTATION._90"]

[display.set_backlight]
params = [100]

[task_handler.TaskHandler]
params=[]
```

closed 2025-02-06 by gitcnd