lvgl-micropython/lvgl_micropython #292: Unable to use .toml build function - Ubuntu on WSL

0ut4t1m3, 2025-02-06:
Hey Kevin,

I'm not sure what's going on but I'm having trouble using the TOML function. I'm getting the below error and I've tried starting with a fresh VM and still hit the same issue. I know it's not your script as you and others are able to run it fine:
```
mike@DESKTOP-K5B8ST9:~/lvgl_micropython$ python3 make.py --toml=/home/mike/toml/t-display_amoled.toml
Traceback (most recent call last):
  File "/home/mike/lvgl_micropython/builder/toml_reader.py", line 326, in run
    toml_data = toml.load(f)
                ^^^^^^^^^^^^
  File "/usr/lib/python3.12/tomllib/_parser.py", line 63, in load
    raise TypeError(
TypeError: File must be opened in binary mode, e.g. use `open('foo.toml', 'rb')`

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/mike/lvgl_micropython/make.py", line 62, in <module>
    build_command = toml_reader.run(toml_path, driver_out_path)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mike/lvgl_micropython/builder/toml_reader.py", line 362, in run
    raise SyntaxError(f'Unable to parse .toml file ({toml_path})') from err
SyntaxError: Unable to parse .toml file (/home/mike/toml/t-display_amoled.toml)
```

This is on Ubuntu 24.04 on WSL, any ideas what I'm doing wrong?

kdschlosser, 2025-02-06:
Give me a minute and I will fix it.

In the meantime you will need to run `pip3 install toml`. I am going to only use that one and not the one that is built into Python. It appears as tho it has a different API even tho it is not supposed to. So I am not going to even mess around with it.

kdschlosser, 2025-02-06:
OK I made the change. clone the repo again and give it a run after installing toml

0ut4t1m3, 2025-02-06:
I get the venv warning trying to run pip3 install toml:
```
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```
I installed with ```sudo apt install python3-toml``` - I hope this is the same package.
It did change the error I'm seeing:
```
mike@DESKTOP-K5B8ST9:~/lvgl_micropython$ python3 make.py --toml=/home/mike/toml/t-display_amoled.toml
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/toml/decoder.py", line 511, in loads
    ret = decoder.load_line(line, currentlevel, multikey,
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/toml/decoder.py", line 778, in load_line
    value, vtype = self.load_value(pair[1], strictly_valid)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/toml/decoder.py", line 910, in load_value
    raise ValueError("This float doesn't have a leading "
ValueError: This float doesn't have a leading digit

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/mike/lvgl_micropython/builder/toml_reader.py", line 319, in run
    toml_data = toml.load(f)
                ^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/toml/decoder.py", line 156, in load
    return loads(f.read(), _dict, decoder)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/toml/decoder.py", line 514, in loads
    raise TomlDecodeError(str(err), original, pos)
toml.decoder.TomlDecodeError: This float doesn't have a leading digit (line 40 column 1 char 607)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/mike/lvgl_micropython/make.py", line 62, in <module>
    build_command = toml_reader.run(toml_path, driver_out_path)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mike/lvgl_micropython/builder/toml_reader.py", line 355, in run
    raise SyntaxError(f'Unable to parse .toml file ({toml_path})') from err
SyntaxError: Unable to parse .toml file (/home/mike/toml/t-display_amoled.toml)
```
This is the contents of the file:
```toml
[MCU.esp32]
BOARD = "ESP32_GENERIC_S3"
BOARD_VARIANT = "SPIRAM_OCT"
flash_size = 16
enable_jtag_repl = 'y'
enable_cdc_repl = 'n'
enable_uart_repl = 'n'
debug = false
uart_repl_bitrate = 115200
DISPLAY = "~/drivers/rm67162/"

[SPI.Bus.spi_bus]
host = 1
mosi = 18
miso = 8
sck = 47

[SPIBus.display_bus]
spi_bus = "spi_bus"
freq = 80000000
dc = 7
cs = 6

[I2C.Bus.i2c_bus]
host = 0
scl = 2
sda = 3
freq = 400000

[I2C.Device.indev_device]
bus = "i2c_bus"
dev_id = "cst816s.I2C_ADDR"
reg_bits = "cst816s.BITS"

[RM67162.display]
data_bus = "display_bus"
display_width = 536
display_height = 240
reset_pin = 17
reset_state = rm67162.STATE_LOW
power_pin = 38
color_space = lv.COLOR_FORMAT.RGB565
rgb565_byte_swap = True

[display.init]
params = []

[cst816s.indev]
device = "indev_device"

[display.set_brightness]
params = [200]

[task_handler.TaskHandler]
params=[]
```

0ut4t1m3, 2025-02-06:
lol the TOML syntax highlighting in github gave me a few things to fix it seems, might just be my config now hold on

0ut4t1m3, 2025-02-06:
Here's the corrected TOML, now working thanks Kevin
```toml
[MCU.esp32]
BOARD = "ESP32_GENERIC_S3"
BOARD_VARIANT = "SPIRAM_OCT"
flash_size = 16
enable_jtag_repl = 'y'
enable_cdc_repl = 'n'
enable_uart_repl = 'n'
uart_repl_bitrate = 115200
DISPLAY = "~/drivers/rm67162/"

[SPI.Bus.spi_bus]
host = 1
mosi = 18
miso = 8
sck = 47

[SPIBus.display_bus]
spi_bus = "spi_bus"
freq = 80000000
dc = 7
cs = 6

[I2C.Bus.i2c_bus]
host = 0
scl = 2
sda = 3
freq = 400000

[I2C.Device.indev_device]
bus = "i2c_bus"
dev_id = "cst816s.I2C_ADDR"
reg_bits = "cst816s.BITS"

[RM67162.display]
data_bus = "display_bus"
display_width = 536
display_height = 240
reset_pin = 17
reset_state = "rm67162.STATE_LOW"
power_pin = 38
color_space = "lv.COLOR_FORMAT.RGB565"
rgb565_byte_swap = true

[display.init]
params = []

[cst816s.indev]
device = "indev_device"

[display.set_brightness]
params = [200]

[task_handler.TaskHandler]
params=[]
```

closed 2025-02-06 by 0ut4t1m3

kdschlosser, 2025-02-11:
Hey I just noticed you are using a custom display driver. mind posting the code for that??

0ut4t1m3, 2025-02-11:
Hey mate, sure thing. I've only tested it with one display so hopefully works as is on others. I'm just forcing the values for the orientation table in the class, I'm not sure if that's what you intended a driver do but it does do the job.

I added brightness methods because OLED and backlight just seems strange :P

_rm67162_init.py
```python
from micropython import const
import time

_SLPOUT = const(0x11)
_MADCTL = const(0x36)
_DISPON = const(0x29)
_SETPGE = const(0xFE)
_PIXFMT = const(0x3A)
_BRIGHT = const(0x51)

def init(self):
    
    self._br_val = 0
    
    time.sleep_ms(120)
    self.set_params(_SETPGE, b'\x00')
    self.set_params(_SLPOUT)
    time.sleep_ms(120)
    
    self.set_params(_SETPGE, b'\x05')
    self.set_params(0x05, b'\x05')
    
    self.set_params(_SETPGE, b'\x01')
    self.set_params(0x73, b'\x25')
    
    self.set_params(_SETPGE, b'\x00')
    self.set_params(_MADCTL, b'\x60')
    self.set_params(_PIXFMT, b'\x55')
    self.set_params(_BRIGHT, b'\x00')
    self.set_params(_DISPON)
```
rm67162_init.py
```python
import display_driver_framework
import lvgl as lv

STATE_HIGH = display_driver_framework.STATE_HIGH
STATE_LOW = display_driver_framework.STATE_LOW
STATE_PWM = display_driver_framework.STATE_PWM

BYTE_ORDER_RGB = display_driver_framework.BYTE_ORDER_RGB
BYTE_ORDER_BGR = display_driver_framework.BYTE_ORDER_BGR

class RM67162(display_driver_framework.DisplayDriver):
    _ORIENTATION_TABLE = (0x60, 0x43, 0xe2, 0x83)
    
    def __init__(
        self,
        data_bus,
        display_width,
        display_height,
        reset_pin=None,
        reset_state=STATE_LOW,
        power_pin=None,
        color_space=lv.COLOR_FORMAT.RGB565,
        rgb565_byte_swap=True
    ):
        super().__init__(
            data_bus=data_bus,
            display_width=display_width,
            display_height=display_height,
            reset_pin=reset_pin,
            reset_state=reset_state,
            power_pin=power_pin,
            color_space=color_space,
            rgb565_byte_swap=rgb565_byte_swap
        )
    
    def set_brightness(self, val=0):
        self._br_val = val
        self.set_params(0x51, bytes([val]))
        
    def get_brightness(self):
        return self._br_val
```

kdschlosser, 2025-02-11:
I changed a few things in the driver. A lot of it was done perfect. All I really did was expand the functionality of it.
It is added so you can take a look if you want. 

Awesome job with this.

I tried to write the display drivers in a manner that made it very flexible so when displays are added that have some kind of goofy things that need to be done it can be done without having to write the entire driver over again.

0ut4t1m3, 2025-02-11:
Oh wow you've really taken it up a level thank you.

I am getting problems with the orientation values you have though. The image is split like each odd line is offset from the even ones. If I force my values of 0x60, 0x43, 0xe2, 0x83 it goes sane again.

If it helps it appears that the default orientation for the panel is landscape but when using your values and set_rotation(0) it seems to be using a buffer in portrait but the touch is still in landscape

kdschlosser, 2025-02-11:
Give it a try now. I messed up the values for the orientation table. They have a couple of extra bits and one bit is flip flopped from what it should be. Hopefully it will work now.

kdschlosser, 2025-02-11:
> Oh wow you've really taken it up a level thank you.

I kind of had to in order to make sure it wasn't only keyed to a single type of display. Gotta make sire that if the IC is pair with a different display it will still work. I also needed to make it so it would work with all the the different data bus types the IC supports and the MCU supports. There are some initialization values that are different based on those things. And I figured what the hell I saw that it had some extras available so I decided tho throw those into the mix. It has a nice feature of adjusting the colors based on the ambient light. You would need to connect a sensor to the MCU and remap the sensor values to a 0.0-100.0 scale which isn't that hard to do.

I also changed how the brightness works, it now works on a 0.0 to 100.0 scale. This is easier for people to wrap their heads around because it is a percentage value. 

OH on that automatic color adjustment. You can manually set that to whatever you want I placed it right in the middle at power on so if the colors are a little bit funky you can adjust them accordingly.

There is also a sunlight readable mode. it is defaulted to off and you can change that to a value of 0, 1, 2 and 3.. -1 turns it off.

kdschlosser, 2025-02-11:
so now it is baked in and you will not need to worry about supplying your own. Make sure you update your toml file and remove the `DISPLAY=` from the MCU section.

0ut4t1m3, 2025-02-11:
Cheers mate, much appreciated.

kdschlosser, 2025-02-11:
no worries m8