lvgl-micropython/lvgl_micropython #293: An error occurred while initializing the bus

HenTaiCJN, 2025-02-06:
I'm trying to bundle a project into my own project and I'm almost there but I'm getting the following error.
Maybe I didn't port the api driver elegantly enough? I selectively copied the files under api_drivers directly into the modules file.

```shell
Traceback (most recent call last):
File "main.py", line 47, in
File "st7789.py", line 60, in __init__
File "display_driver_framework.py", line 231, in __init__
File "display_driver_framework.py", line 281, in _init_bus
SyntaxError: Cannot convert 'bound_method' to pointer!
MicroPython v1.23.0-dirty on 2025-02-06; Generic ESP32S3 module with ESP32S3
```

```py
import lcd_bus
from micropython import const
import machine

# display settings
_WIDTH = const(170)
_HEIGHT = const(320)
_BL = const(21)
_RST = const(46)
_DC = const(9)

_MOSI = const(13)
_MISO = const(-1)
_SCK = const(14)
_HOST = const(1)  # SPI2

_LCD_CS = const(10)
_LCD_FREQ = const(80000000)

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK,    
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
    spi_mode=3
)

import st7789  # NOQA
import lvgl as lv  # NOQA

display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    reset_pin=_RST,
	reset_state=st7789.STATE_LOW,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_RGB,
    rgb565_byte_swap=True,
	offset_x=37
)

import task_handler  # NOQA

display.set_power(True)
display.init()
display.set_backlight(100)

th = task_handler.TaskHandler()

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, 0)

```

HenTaiCJN, 2025-02-07:
> are you using the latest code from the repo?
> 
> What are you using for a build command?
> 
> What MCU are you compiling for?
> 
> There is a template for the issues for a reason. It's so I don't have to ask all of these kinds of questions, they should already be answered when opening up an issue

This is a bit complicated. In short, I used the Micropython1.23.0 repository as the basis and added the files in this repository. In the Makefile, I added a user module pointing to the cmake in ext_mod. Finally, I modified the file according to make.py and added cmake parameters.

kdschlosser, 2025-02-07:
You do not have to do any of that..

All you need to do is follow the instructions in the README.md file. The only build requirements you need to install are the ones that get installed using your OS's package manager. All other requirements are collected for you by the build script. It also does any setup of those requirements as well.

It is literally a single line build command that take care of everything.

as an example...
```
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=st7789 INDEV=gt911 --flash-size=16 --octal-flash --enable-cdc-repl=n
```

The script adjusts the partition file to optimize the amount of space the user is able to use. It also adjusts it if the application partition is too small to fit the firmware. It literally does everything for you. It's not complicated to use. The README.md file gives you all of the information you need to know to get up and running.

HenTaiCJN, 2025-02-07:
> You do not have to do any of that..
>
> All you need to do is follow the instructions in the README.md file. The only build requirements you need to install are the ones that get installed using your [...]

I understand, but the project is a bit too coupled and not suitable for my development needs. Anyway, thank you very much for your project, I am only one step away from success

kdschlosser, 2025-02-07:
I am not going to be able to help you at all if you do not use it as it was written. Making alterations and modifications is going to cause issues I can gurantee you that.

closed 2025-02-07 by kdschlosser