lvgl-micropython/lvgl_micropython #294: mbedtls compilation issues on arm64 macOS

thatbudakguy, 2025-02-08:
**Describe the bug**
Compilation on macOS fails during the building of mbedtls.

**If you are having a problem compiling paste the build output below. (NO SCREENSHOTS)**

```shell
python make.py macOS
Compiling mpy-cross....
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.

Generating build files....
/Users/nick/Developer/lvgl_micropython/api_drivers/py_api_drivers/frozen/display/display_driver_framework.py
/Users/nick/Developer/lvgl_micropython/api_drivers/py_api_drivers/frozen/display/rgb_display_framework.py
/Users/nick/Developer/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/touch_calibration/touch_cal_data.py
/Users/nick/Developer/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/touch_calibration/touch_calibrate.py
/Users/nick/Developer/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/_indev_base.py
/Users/nick/Developer/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/button_framework.py
/Users/nick/Developer/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/encoder_framework.py
/Users/nick/Developer/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/keypad_framework.py
/Users/nick/Developer/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/pointer_framework.py
/Users/nick/Developer/lvgl_micropython/api_drivers/py_api_drivers/fs_driver.py
/Users/nick/Developer/lvgl_micropython/api_drivers/py_api_drivers/frozen/io_expander/io_expander_framework.py
/Users/nick/Developer/lvgl_micropython/api_drivers/common_api_drivers/frozen/other/i2c.py
/Users/nick/Developer/lvgl_micropython/api_drivers/common_api_drivers/frozen/other/task_handler.py
Compiling....
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from /Users/nick/Developer/lvgl_micropython/ext_mod/lcd_bus
Including User C Module from /Users/nick/Developer/lvgl_micropython/ext_mod/lcd_utils
Including User C Module from /Users/nick/Developer/lvgl_micropython/ext_mod/lvgl
mkdir -p build-standard/genhdr
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/core
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/display
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/draw
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/draw/nxp/pxp
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/draw/nxp/vglite
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/draw/renesas/dave2d
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/draw/sdl
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/draw/sw
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/draw/sw/blend
LVGL-GEN build-standard/lv_mpy.c
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/draw/vg_lite
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/drivers/display/drm
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/drivers/display/fb
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/drivers/display/ili9341
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/drivers/display/lcd
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/drivers/display/renesas_glcdc
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/drivers/display/st7735
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/drivers/display/st7789
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/drivers/display/st7796
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/drivers/evdev
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/drivers/glfw
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/drivers/libinput
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/drivers/nuttx
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/drivers/qnx
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/drivers/sdl
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/drivers/wayland
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/drivers/windows
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/drivers/x11
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/font
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/indev
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/layouts
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/layouts/flex
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/layouts/grid
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/libs/barcode
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/libs/bin_decoder
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/libs/bmp
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/libs/ffmpeg
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/libs/freetype
mkdir -p build-standard//Users/nick/Developer/lvgl_micropython/lib/lvgl/src/libs/fsdrv
[... 763 lines cut ...]
41 |     MBEDTLS_INTERNAL_VALIDATE_RET(cond, MBEDTLS_ERR_MPI_BAD_INPUT_DATA)
|     ^
../../lib/mbedtls/library/bignum.c:966:5: error: call to undeclared function 'MBEDTLS_INTERNAL_VALIDATE_RET'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
966 |     MPI_VALIDATE_RET(X != NULL);
|     ^
../../lib/mbedtls/library/bignum.c:41:5: note: expanded from macro 'MPI_VALIDATE_RET'
41 |     MBEDTLS_INTERNAL_VALIDATE_RET(cond, MBEDTLS_ERR_MPI_BAD_INPUT_DATA)
|     ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make: *** [build-standard/lib/mbedtls/library/bignum.o] Error 1
make: *** Waiting for unfinished jobs....
```

**Exact make and model number of the MCU that you are compiling for or the firmware is running on.**

* Make: Apple
* Model: M3 Pro 
 * OS: 14.7.2

**Build Command**

```shell
python make.py macOS
```

kdschlosser, 2025-02-08:
I am not 100% sure what is causing this issue yet. It was reported [here](https://github.com/lvgl-micropython/lvgl_micropython/discussions/290). I have to go poking about upstream with mbedtls to see what is going on with it.

kdschlosser, 2025-02-08:
OK I just updated the repo. Get a fresh clone and give it a try. I removed the parallel compiling so I can get the actual order in which things are being compiled. Then I will be able to step through the files to see how the include tree is. I think there might be something out of sorts that I need to change to get the files to be included in the correct order.

thatbudakguy, 2025-02-08:
Here's the result (same error):

```
python make.py macOS DISPLAY=sdl_display INDEV=sdl_pointer

collecting MicroPython 1.24.1
Submodule 'lib/micropython' (https://github.com/micropython/micropython) registered for path 'lib/micropython'
Cloning into '/Users/nick/Developer/lvgl_micropython/lib/micropython'...
From https://github.com/micropython/micropython
* branch            ecfdd5d6f9be971852003c2049600dc7b3e2a838 -> FETCH_HEAD
Submodule path 'lib/micropython': checked out 'ecfdd5d6f9be971852003c2049600dc7b3e2a838'
Submodule 'lib/micropython' (https://github.com/micropython/micropython) registered for path 'lib/micropython'
Cloning into '/Users/nick/Developer/lvgl_micropython/lib/micropython'...
From https://github.com/micropython/micropython
* branch            ecfdd5d6f9be971852003c2049600dc7b3e2a838 -> FETCH_HEAD
Submodule path 'lib/micropython': checked out 'ecfdd5d6f9be971852003c2049600dc7b3e2a838'

collecting LVGL
this might take a while...
Submodule 'lib/lvgl' (https://github.com/lvgl/lvgl) registered for path 'lib/lvgl'
Cloning into '/Users/nick/Developer/lvgl_micropython/lib/lvgl'...
From https://github.com/lvgl/lvgl
* branch            7f07a129e8d77f4984fff8e623fd5be18ff42e74 -> FETCH_HEAD
Submodule path 'lib/lvgl': checked out '7f07a129e8d77f4984fff8e623fd5be18ff42e74'
Submodule 'lib/lvgl' (https://github.com/lvgl/lvgl) registered for path 'lib/lvgl'
Cloning into '/Users/nick/Developer/lvgl_micropython/lib/lvgl'...
From https://github.com/lvgl/lvgl
* branch            7f07a129e8d77f4984fff8e623fd5be18ff42e74 -> FETCH_HEAD
Submodule path 'lib/lvgl': checked out '7f07a129e8d77f4984fff8e623fd5be18ff42e74'

collecting pycparser
Submodule 'lib/pycparser' (https://github.com/eliben/pycparser) registered for path 'lib/pycparser'
Cloning into '/Users/nick/Developer/lvgl_micropython/lib/pycparser'...
From https://github.com/eliben/pycparser
* branch            3cf6bf5eb16f5eadd4a058e41596145c407a79ad -> FETCH_HEAD
Submodule path 'lib/pycparser': checked out '3cf6bf5eb16f5eadd4a058e41596145c407a79ad'
Compiling mpy-cross....
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
GEN build/genhdr/moduledefs.split
GEN build/genhdr/moduledefs.collected
Module registrations updated
GEN build/genhdr/moduledefs.h
GEN build/genhdr/root_pointers.split
GEN build/genhdr/root_pointers.collected
Root pointer registrations updated
GEN build/genhdr/root_pointers.h
mkdir -p build/py
mkdir -p build/shared/runtime
CC ../shared/runtime/gchelper_generic.c
LINK build/mpy-cross
__TEXT	__DATA	__OBJC	others	dec	hex
393216	16384	0	4295016448	4295426048	100070000

Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from /Users/nick/Developer/lvgl_micropython/ext_mod/lcd_bus
Including User C Module from /Users/nick/Developer/lvgl_micropython/ext_mod/lcd_utils
Including User C Module from /Users/nick/Developer/lvgl_micropython/ext_mod/lvgl
Updating submodules: lib/mbedtls lib/berkeley-db-1.xx lib/micropython-lib
Submodule 'lib/berkeley-db-1.xx' (https://github.com/micropython/berkeley-db-1.xx) registered for path 'lib/berkeley-db-1.xx'
Submodule 'lib/mbedtls' (https://github.com/ARMmbed/mbedtls.git) registered for path 'lib/mbedtls'
Submodule 'lib/micropython-lib' (https://github.com/micropython/micropython-lib.git) registered for path 'lib/micropython-lib'
Cloning into '/Users/nick/Developer/lvgl_micropython/lib/micropython/lib/berkeley-db-1.xx'...
Cloning into '/Users/nick/Developer/lvgl_micropython/lib/micropython/lib/mbedtls'...
[... 817 lines cut ...]
../../lib/mbedtls/library/bignum.c:41:5: note: expanded from macro 'MPI_VALIDATE_RET'
41 |     MBEDTLS_INTERNAL_VALIDATE_RET(cond, MBEDTLS_ERR_MPI_BAD_INPUT_DATA)
|     ^
../../lib/mbedtls/library/bignum.c:910:5: error: call to undeclared function 'MBEDTLS_INTERNAL_VALIDATE_RET'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
910 |     MPI_VALIDATE_RET(X != NULL);
|     ^
../../lib/mbedtls/library/bignum.c:41:5: note: expanded from macro 'MPI_VALIDATE_RET'
41 |     MBEDTLS_INTERNAL_VALIDATE_RET(cond, MBEDTLS_ERR_MPI_BAD_INPUT_DATA)
|     ^
../../lib/mbedtls/library/bignum.c:923:5: error: call to undeclared function 'MBEDTLS_INTERNAL_VALIDATE_RET'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
923 |     MPI_VALIDATE_RET(X != NULL);
|     ^
../../lib/mbedtls/library/bignum.c:41:5: note: expanded from macro 'MPI_VALIDATE_RET'
41 |     MBEDTLS_INTERNAL_VALIDATE_RET(cond, MBEDTLS_ERR_MPI_BAD_INPUT_DATA)
|     ^
../../lib/mbedtls/library/bignum.c:966:5: error: call to undeclared function 'MBEDTLS_INTERNAL_VALIDATE_RET'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
966 |     MPI_VALIDATE_RET(X != NULL);
|     ^
../../lib/mbedtls/library/bignum.c:41:5: note: expanded from macro 'MPI_VALIDATE_RET'
41 |     MBEDTLS_INTERNAL_VALIDATE_RET(cond, MBEDTLS_ERR_MPI_BAD_INPUT_DATA)
|     ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make: *** [build-standard/lib/mbedtls/library/bignum.o] Error 1
```

kdschlosser, 2025-02-08:
I think I know what is going on. I think you have mbedtls installed into your system and it is using the header files from that instead of from micropython.

kdschlosser, 2025-02-08:
mbedtls is not using relative imports. It is using absolute imports with the base directory being "mbedtls". What happens here is because the path to the header files is not in the same folder as the source files the first thing clang checks is the system include paths. I do not know if there is a way to override this at all.

kdschlosser, 2025-02-08:
I just pushed a commit that might fix it. Give it a try and lemme know

thatbudakguy, 2025-02-08:
I think you're on to something. I did indeed have mbedtls installed via brew (it is a dependency of ffmpeg). After uninstalling ir and re-cloning this repo, the compilation doesn't error, but just hangs after compiling the SDL drivers. I've tried a few times and it seems to run for over 30min with no end in sight.

I somehow just noticed in the README that I needed to `brew install armmbed/formulae/arm-none-eabi-gcc`, which I didn't have – I wonder if this is related. When I do, brew tells me:

```
Error: The `brew link` step did not complete successfully
The formula built, but is not symlinked into /opt/homebrew
Could not symlink bin/arm-none-eabi-addr2line
Target /opt/homebrew/bin/arm-none-eabi-addr2line
is a symlink belonging to arm-none-eabi-binutils. You can unlink it:
  brew unlink arm-none-eabi-binutils

To force the link and overwrite all conflicting files:
  brew link --overwrite arm-none-eabi-gcc

To list all files that would be deleted:
  brew link --overwrite arm-none-eabi-gcc --dry-run

Possible conflicting files are:
/opt/homebrew/bin/arm-none-eabi-addr2line -> /opt/homebrew/Cellar/arm-none-eabi-binutils/2.41/bin/arm-none-eabi-addr2line
/opt/homebrew/bin/arm-none-eabi-ar -> /opt/homebrew/Cellar/arm-none-eabi-binutils/2.41/bin/arm-none-eabi-ar
/opt/homebrew/bin/arm-none-eabi-as -> /opt/homebrew/Cellar/arm-none-eabi-binutils/2.41/bin/arm-none-eabi-as
/opt/homebrew/bin/arm-none-eabi-c++ -> /opt/homebrew/Cellar/arm-none-eabi-gcc@8/8.5.0_2/bin/arm-none-eabi-c++
/opt/homebrew/bin/arm-none-eabi-c++filt -> /opt/homebrew/Cellar/arm-none-eabi-binutils/2.41/bin/arm-none-eabi-c++filt
/opt/homebrew/bin/arm-none-eabi-cpp -> /opt/homebrew/Cellar/arm-none-eabi-gcc@8/8.5.0_2/bin/arm-none-eabi-cpp
/opt/homebrew/bin/arm-none-eabi-elfedit -> /opt/homebrew/Cellar/arm-none-eabi-binutils/2.41/bin/arm-none-eabi-elfedit
/opt/homebrew/bin/arm-none-eabi-g++ -> /opt/homebrew/Cellar/arm-none-eabi-gcc@8/8.5.0_2/bin/arm-none-eabi-g++
/opt/homebrew/bin/arm-none-eabi-gcc -> /opt/homebrew/Cellar/arm-none-eabi-gcc@8/8.5.0_2/bin/arm-none-eabi-gcc
/opt/homebrew/bin/arm-none-eabi-gcc-ar -> /opt/homebrew/Cellar/arm-none-eabi-gcc@8/8.5.0_2/bin/arm-none-eabi-gcc-ar
/opt/homebrew/bin/arm-none-eabi-gcc-nm -> /opt/homebrew/Cellar/arm-none-eabi-gcc@8/8.5.0_2/bin/arm-none-eabi-gcc-nm
/opt/homebrew/bin/arm-none-eabi-gcc-ranlib -> /opt/homebrew/Cellar/arm-none-eabi-gcc@8/8.5.0_2/bin/arm-none-eabi-gcc-ranlib
/opt/homebrew/bin/arm-none-eabi-gcov -> /opt/homebrew/Cellar/arm-none-eabi-gcc@8/8.5.0_2/bin/arm-none-eabi-gcov
/opt/homebrew/bin/arm-none-eabi-gcov-dump -> /opt/homebrew/Cellar/arm-none-eabi-gcc@8/8.5.0_2/bin/arm-none-eabi-gcov-dump
/opt/homebrew/bin/arm-none-eabi-gcov-tool -> /opt/homebrew/Cellar/arm-none-eabi-gcc@8/8.5.0_2/bin/arm-none-eabi-gcov-tool
/opt/homebrew/bin/arm-none-eabi-gprof -> /opt/homebrew/Cellar/arm-none-eabi-binutils/2.41/bin/arm-none-eabi-gprof
/opt/homebrew/bin/arm-none-eabi-ld -> /opt/homebrew/Cellar/arm-none-eabi-binutils/2.41/bin/arm-none-eabi-ld
/opt/homebrew/bin/arm-none-eabi-ld.bfd -> /opt/homebrew/Cellar/arm-none-eabi-binutils/2.41/bin/arm-none-eabi-ld.bfd
/opt/homebrew/bin/arm-none-eabi-nm -> /opt/homebrew/Cellar/arm-none-eabi-binutils/2.41/bin/arm-none-eabi-nm
/opt/homebrew/bin/arm-none-eabi-objcopy -> /opt/homebrew/Cellar/arm-none-eabi-binutils/2.41/bin/arm-none-eabi-objcopy
/opt/homebrew/bin/arm-none-eabi-objdump -> /opt/homebrew/Cellar/arm-none-eabi-binutils/2.41/bin/arm-none-eabi-objdump
/opt/homebrew/bin/arm-none-eabi-ranlib -> /opt/homebrew/Cellar/arm-none-eabi-binutils/2.41/bin/arm-none-eabi-ranlib
/opt/homebrew/bin/arm-none-eabi-readelf -> /opt/homebrew/Cellar/arm-none-eabi-binutils/2.41/bin/arm-none-eabi-readelf
/opt/homebrew/bin/arm-none-eabi-size -> /opt/homebrew/Cellar/arm-none-eabi-binutils/2.41/bin/arm-none-eabi-size
/opt/homebrew/bin/arm-none-eabi-strings -> /opt/homebrew/Cellar/arm-none-eabi-binutils/2.41/bin/arm-none-eabi-strings
/opt/homebrew/bin/arm-none-eabi-strip -> /opt/homebrew/Cellar/arm-none-eabi-binutils/2.41/bin/arm-none-eabi-strip
```

Should I go ahead and overwrite? Not sure what the implications of this are.

kdschlosser, 2025-02-09:
the arm drivers are for compiling for an MCU not for compiling for your OS...

I made a change that is probably messing up the build. I will revert the change and we can see what happens form there.

kdschlosser, 2025-02-09:
ok all done. give it another try.

thatbudakguy, 2025-02-09:
> the arm drivers are for compiling for an MCU not for compiling for your OS...

I figured this was the case, but just wanted to make sure 😅 I do eventually want to compile for ESP32 and ESP32C3 targets, but a working macOS build is the first priority.

Anyway, after a fresh clone, the macOS build succeeded!! Many thanks for your persistence and quick replies.

closed 2025-02-09 by thatbudakguy

kdschlosser, 2025-02-09:
no worries m8. glad to have been of help. Environment thing can really muck things up sometimes and it takes some digging to find out what the actual cause is. This one wasn't so bad but it would have been easier if clang provided a path chain for the error instead of the single line.