lvgl-micropython/lvgl_micropython #295: unix/macOS SDL example doesn't open a window

thatbudakguy, 2025-02-09:
I'm probably missing something extremely obvious here, but after successfully compiling for macOS with:
```sh
python make.py macOS DISPLAY=sdl_display INDEV=sdl_pointer
chmod +x ./build/lvgl_micropy_macOS
```

The binary works fine and I can invoke scripts in the usual way, but using the [example given in the README](https://github.com/lvgl-micropython/lvgl_micropython?tab=readme-ov-file#unixmacos) (as a file `test.py`), nothing happens when I invoke it:

```sh
./build/lvgl_micropy_macOS ./test.py
```

...the script just exits immediately without error. Should a window be opened to display the UI? `lv_binding_micropython` seems to have a method `lv.sdl_window_create`, but that's not defined in this build.

kdschlosser, 2025-02-09:
man O man I get this all the time... I cannot even begin to tell you how many times I have been asked about this problem...

OK so look at the script you are running. Now think about what would happen if you ran the script using CPython. Once CPython gets to the end of the script there is nothing else to run is there? so it exits right? Same thing is happening here. You do not have any looping code to keep the script running.

Best way to test is to just open up the repl and import the test file. the repl has looping code that holds it open.

so remove the ` ./test.py` from the command and once in the repl key in `import test.py`

kdschlosser, 2025-02-09:
Once you get a working screen I will tell you how to go about looping the code so it stays running if you want to execute a script directly from the command line.

thatbudakguy, 2025-02-09:
I figured it was something like this! I had tried putting a `time.sleep` in `test.py` to prevent early exit but still nothing rendered.

When importing nothing happens, but when I send ctrl-c, the window suddenly appears:
```py
LVGL (9.2.2) MicroPython (1.24.1) Binding compiled on 2025-02-09; darwin [GCC 4.2.1] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import test.py
^CTraceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "test.py", line 42, in <module>
KeyboardInterrupt: 
```

[image]

kdschlosser, 2025-02-09:
this is going to be a bugger to diagnose because I don't have a mac..

paste the script you are running. I want to see the code.

thatbudakguy, 2025-02-09:
🤦🏻  apparently the `time.sleep` was still in there, which was causing the hanging (duh). After removing that, when I do `import test.py` from the REPL, all works normally.

How would I go about inserting a loop that it can be run from the command line?

mark-hoy, 2025-02-09:
Since I've also got a couple of Macs, I tried the sample code from the README on my M1 Mac Mini running Sequoia 15.1.1.

The window opens and I can move the slider around.

```
$  cd lvgl_micropython
$  python3 make.py macOS DISPLAY=sdl_display INDEV=sdl_pointer
$ chmod +x build/lvgl_micropy_macOS
$  cd ../simple/
$ ../lvgl_micropython/build/lvgl_micropy_macOS 
LVGL (9.2.2) MicroPython (1.24.1) Binding compiled on 2025-02-09; darwin [GCC 4.2.1] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import sample
import sample
>>> 2025-02-09 14:16:01.548 lvgl_micropy_macOS[12264:111506] +[IMKClient subclass]: chose IMKClient_Modern
2025-02-09 14:16:01.548 lvgl_micropy_macOS[12264:111506] +[IMKInputSession subclass]: chose IMKInputSession_Modern
```

I have the following brew items
```
bash-5.2$ brew list
==> Formulae
arm-none-eabi-binutils  curl                    hiredis                 libyaml                 pkgconf                 rtmpdump
arm-none-eabi-gcc       dfu-util                icu4c@76                llvm                    platformio              sdl2
autoconf                dotnet                  isl                     lsusb                   potrace                 six
bash                    dotnet@8                jansson                 lz4                     pycparser               sqlite
blake3                  emacs                   libevent                m4                      pyenv                   tcl-tk@8
brotli                  esptool                 libffi                  mpdecimal               pyinstaller             tree-sitter
c-ares                  flake8                  libidn2                 mpfr                    pylint                  unbound
ca-certificates         fmt                     libmpc                  nettle                  python-certifi          virtualenv
ccache                  gettext                 libnghttp2              ninja                   python-setuptools       wget
certifi                 gh                      libssh2                 node                    python-tk@3.11          xxhash
cffi                    git                     libtasn1                openssl@3               python@3.11             xz
click                   gitversion              libunistring            p11-kit                 python@3.12             z3
cmake                   gmp                     libusb                  pcre2                   python@3.13             zstd
cryptography            gnutls                  libuv                   picocom                 readline

==> Casks
devtoys macfuse thonny
```

I have also recently installed VirtualBox onto my M1 MacMini, loaded up Ubuntu 24, added in the various packages to build and successfully created lvgl_micropython_unix. I launched it, imported the sample code and it also brought up the slider.

kdschlosser, 2025-02-09:
I suspect that you didn't have the task_handler added. That is what pumps the window for events so things like the close button will work and mouse movement will work. It's one thing to have it render but without any input it is pretty useless. LOL..

kdschlosser, 2025-02-09:
I partially wrote the keyboard driver for it. I have to write the LVGL "keypad" portion of the code for it to work. At some point I will get around to it.

thatbudakguy, 2025-02-10:
Thanks for your help!

closed 2025-02-10 by thatbudakguy