lvgl-micropython/lvgl_micropython #298: Stub generation is missing some typedefs

thatbudakguy, 2025-02-16:
**Describe the bug**
The stub generation step that creates `lvgl.pyi` does not include many typedefs at the top that are referenced later in the file, causing it to not be interpreted by the IDE.

**Details**
I had been using the stubs provided in [this thread on the LVGL forums](https://forum.lvgl.io/t/how-does-one-create-micropython-stubs-of-lvgl-bindings-for-unix-port/15233/5), but noticed that a few typedefs appeared to not match the actual compiled binary – understandable given that the thread is almost a year old at this point.

In looking to acquire new stubs, I noticed that the build command generates them for me (nice!). However, when I ported the generated `lvgl.pyi` file into the `typings/` folder of my local unix micropython project, most of the typings couldn't be loaded. When I inspected the file, I saw a lot (hundreds) of warnings like `Undefined name style_prop_t`. It seemed that classes defined later in the file were referencing typedefs that didn't exist.

Comparing with the `lvgl.pyi` provided in the thread, I noticed that it included a section of typedefs at the top that began:
```py
flag_t = int
rb_color_t = int
thread_prio_t = int
[...]
```

This appears to be missing from the `lvgl.pyi` that was generated during my build.

**Build Command**

OS: macOS 15.3.1
CPU: Apple M3 Pro

```shell
python make.py macOS DISPLAY=sdl_display INDEV=sdl_pointer
```

kdschlosser, 2025-02-16:
I am aware that there is quite a bit that is not right in it. Feel free to fix things and submit PR's for it.

kdschlosser, 2025-03-01:
I am working on a new script to generate the stub file. I just wanted to let you know that I was doing that.

kdschlosser, 2025-03-01:
If you want to have a look over it to see if there are any errors in it. 

[lvgl.pyi.txt](https://github.com/user-attachments/files/19039143/lvgl.pyi.txt)