lvgl-micropython/lvgl_micropython #301: Use of external fonts

F4HSL-Yves, 2025-02-19:
Hello Kevin, Hope you are doing well, i try to use external fonts (ttf files) with your binding. I have used the lvgl font converter to get a .c file, but the lvgl.load_font does not work (not supported ii think). Could you tell me how to simple use an external font in order to WriteText (specially with UTF8 accentued characters) on the screen of my esp32 lcd touch device ? 
Regards
Yves

F4HSL-Yves, 2025-02-19:
Problem solved, i use this and it works 

label = lv.label(scrn)
import fs_driver
fs_drv = lv.fs_drv_t()
fs_driver.fs_register(fs_drv, 'S')
my_font = lv.binfont_create("S:/ARIALBD111.bin")

ARIALBD111.bin generated with lvgl converter, have to put the correct symbol range :-)

issue can be closed

closed 2025-02-19 by kdschlosser

asherevan, 2025-02-20:
How would you use the font once you have created it?

F4HSL-Yves, 2025-02-21:
Assumed my_font is the font created you can use it like this (with a label) :
label.set_style_text_font(my_font, 0)

asherevan, 2025-02-21:
Thank you. I realized I was not doing the fs_driver thing and so the created font was just a NoneType object.