lvgl-micropython/lvgl_micropython #302: GT911 touchscreen problems.

asherevan, 2025-02-23:
I have a LilyGo T-Deck (a st7789 display with a gt911 touchscreen) and the screen works great but when the rotation on the display is set to 1 the touchscreen seems to be reversed.
Do you have any ideas on how to fix this?

kdschlosser, 2025-02-23:
That's happening because the orientation table for the display driver needs to be changed to match your display....

give this a shot and lemme know if it works. Put this code into the same python file that you are uploading to the ESP32 that has the drivers in it. If you are using the TOML file or a custom board lemme know and I will tell you what to do in that case.

```python

_MADCTL_MV = const(0x20)
_MADCTL_MX = const(0x40)
_MADCTL_MY = const(0x80)

display._ORIENTATION_TABLE = (
    0x0,
    _MADCTL_MV | _MADCTL_MY,  # swap xy, mirror y   
    _MADCTL_MY | _MADCTL_MX,  # mirror y, mirror x
    _MADCTL_MV | _MADCTL_MX,  # swap xy, mirror x
)

```

asherevan, 2025-02-23:
It worked! Thank you!

asherevan, 2025-02-23:
But if I was using a custom board (I would like to use the one for the LilyGo T-Deck but it doesn’t work yet) how would I do it?

closed 2025-02-23 by asherevan

kdschlosser, 2025-02-23:
Once you hammer out each of the drivers and get things working properly if you send me the code I can write a custom board for it that will work properly. Hammering out the drivers because the TDeck has so many things attached is something that is going to take time to do.