lvgl-micropython/lvgl_micropython #303: Compile error with esp32-C3

yeyf2023, 2025-02-24:
I used the following command to compile the firmware for ESP32 - C3, but after an error was prompted, the compilation could no longer proceed.
python3 make.py --optimize-size --flash-size=4 esp32 BOARD=ESP32_GENERIC_C3 DISPLAY=st7789 

Error:
[1240/1935] Building C object esp-idf/main_esp32c3/CMakeFiles/__idf_main_esp32c3.dir/__/usb.c.obj
[1242/1935] Building C object esp-idf/main_esp32c3/CMakeFiles/__idf_main_esp32c3.dir/__/usb_serial_jtag.c.obj
FAILED: esp-idf/main_esp32c3/CMakeFiles/__idf_main_esp32c3.dir/__/usb_serial_jtag.c.obj
/home/yeyufeng/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -DBERKELEY_DB_CONFIG_FILE=\"/home/yeyufeng/lvgl_micropython/lib/micropython/extmod/berkeley-db/berkeley_db_config_port.h\" -DESP_PLATFORM -DFFCONF_H=\"/home/yeyufeng/lvgl_micropython/lib/micropython/lib/oofatfs/ffconf.h\" -DIDF_VER=\"c9763f62\" -DLFS1_NO_ASSERT -DLFS1_NO_DEBUG -DLFS1_NO_ERROR -DLFS1_NO_MALLOC -DLFS1_NO_WARN -DLFS2_NO_ASSERT -DLFS2_NO_DEBUG -DLFS2_NO_ERROR -DLFS2_NO_MALLOC -DLFS2_NO_WARN -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMICROPY_ESP_IDF_4=1 -DMICROPY_MODULE_FROZEN_MPY="(1)" -DMICROPY_PY_BTREE=1 -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_VFS_FAT=1 -DMICROPY_VFS_LFS2=1 -DNDEBUG -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/yeyufeng/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/config -I/home/yeyufeng/lvgl_micropython/lib/micropython -I/home/yeyufeng/lvgl_micropython/lib/micropython/lib/berkeley-db-1.xx/include -I/home/yeyufeng/lvgl_micropython/ext_mod/spi3wire -I/home/yeyufeng/lvgl_micropython/ext_mod/spi3wire/include -I/home/yeyufeng/lvgl_micropython/ext_mod/lcd_bus -I/home/yeyufeng/lvgl_micropython/ext_mod/lcd_bus/esp32_include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/esp_lcd/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/esp_lcd/interface -I/home/yeyufeng/lvgl_micropython/lib -I/home/yeyufeng/lvgl_micropython/lib/lvgl -I/home/yeyufeng/lvgl_micropython/lib/micropython/ports/esp32 -I/home/yeyufeng/lvgl_micropython/lib/micropython/ports/esp32/boards/ESP32_GENERIC_C3 -I/home/yeyufeng/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3 -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/newlib/platform_include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/freertos/config/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/freertos/config/include/freertos -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/freertos/config/riscv/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/freertos/esp_additions/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include/soc -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/esp_hw_support/include/soc/esp32c3 -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/heap/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/log/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/soc/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/soc/esp32c3 -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/soc/esp32c3/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/hal/platform_port/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/hal/esp32c3/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/hal/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/esp_rom/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/esp_rom/include/esp32c3 -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32c3 -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/esp_common/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/esp_system/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/esp_system/port/soc -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/esp_system/port/include/riscv -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/esp_system/port/include/private -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/riscv/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/lwip/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/lwip/include/apps -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/lwip/include/apps/sntp -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/lwip/lwip/src/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/lwip/port/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/lwip/port/freertos/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/lwip/port/esp32xx/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/lwip/port/esp32xx/include/sys -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/app_update/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bootloader_support/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bootloader_support/bootloader_flash/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/esp_app_format/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/esp_bootloader_format/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/esp_partition/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/include/esp32c3/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/common/osi/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/common/api/include/api -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/common/btc/profile/esp/blufi/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/common/btc/profile/esp/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/common/hci_log/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ans/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/bas/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/dis/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gap/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gatt/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/hr/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/htp/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ias/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ipss/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/lls/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/prox/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/cts/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/tps/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/hid/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/sps/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/components/bt/host/nimble/nimble/nimble/host/util/include -I/home/yeyufeng/lvgl_micropython/lib/esp-idf/co [... cut ...]
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/yeyufeng/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/log/idf_py_stderr_output_30356 and /home/yeyufeng/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/log/idf_py_stdout_output_30356
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: Leaving directory '/home/yeyufeng/lvgl_micropython/lib/micropython/ports/esp32'
make: *** [Makefile:71: all] Error 1

kdschlosser, 2025-02-24:
I just pushed a commit that should fix the problem.

kdschlosser, 2025-02-24:
also, your build command is incorrect

it should be

```
python3 make.py esp32 BOARD=ESP32_GENERIC_C3  --optimize-size --flash-size=4 DISPLAY=st7789 
```

with 4mb of flash space I really suggest not using `--optimize-size` as this is going to really slow things down when it's running. The firmware should only be 2.5mb in size so you have more than enough space. The C3 only has a single core so you want to keep the optimization for performance.

yeyf2023, 2025-02-24:
Thanks for updates and tips.
I have compiled it successfully, and I am now preparing for the LVGL test.

In addition
I encountered the following problem when testing LVGL after ESP32-S3 was successfully compiled, and it cannot be displayed

A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

LVGL MicroPython 
IDF version : c9763f62
Machine     : Generic ESP32S3 module with Octal-SPIRAM with ESP32S3

Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060430  A0      : 0x8200304a  A1      : 0x3fcbe090  
A2      : 0x3c216270  A3      : 0x00000002  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x821b8d1c  A9      : 0x3fcbe080  
A10     : 0x3c216270  A11     : 0x00000002  A12     : 0x000001f3  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000002  SAR     : 0x00000006  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x40056f08  LEND    : 0x40056f12  LCOUNT  : 0x00000000  

Backtrace: 0xfffffffd:0x3fcbe090 0x42003047:0x3fcbe0b0 0x42002d9d:0x3fcbe0f0 0x420d1427:0x3fcbe130

ELF file SHA256: 8192802af

Rebooting...
���ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40383266
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1108
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbdc
load:0x403cc700,len:0x2f4c
entry 0x403c989c
LVGL (9.2.2) MicroPython (1.24.1) Binding compiled on 2025-02-23; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.

My code is as follows
`import lcd_bus
from micropython import const
import machine

# display settings
_WIDTH = const(240)
_HEIGHT = const(240)
_BL = const(4)
_RST = const(21)
_DC = const(16)

_MOSI = const(17)
_MISO = const(-1)
_SCK = const(18)
_HOST = const(1)  # SPI2

_LCD_CS = const(-1)
_LCD_FREQ = const(40000000)

# _TOUCH_CS = const(18)
# _TOUCH_FREQ = const(10000000)

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)

# we are going to let the display driver sort out the best freame buffer size and where to allocate it to.
# fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
# fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import st7789  # NOQA
import lvgl as lv  # NOQA

display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    reset_pin=21,
    reset_state=st7789.STATE_LOW,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)

display.set_power(True)
display.init()
display.set_backlight(100)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)

import task_handler  # NOQA
task_handler.TaskHandler()

`

kdschlosser, 2025-02-24:
you don't have a _BUFFER_SIZE constant

kdschlosser, 2025-02-24:
it would also be helpful if you would decode the backtrace from the error. You can look up how to do it by doing a good search for "ESP32 decode backtrace"

The firmware you are running is also for an ESP32 S3 and not an ESP32 C3.

kdschlosser, 2025-02-24:
OK sorry about the S3 comment. I didn't realize that was a different MCU.

I am going to need to know more information, if you can provide me a link to the data sheet for the board that would be very helpful.

yeyf2023, 2025-02-24:
This is my data sheet using the MCU ESP32 S3 N16R8

[esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf](https://github.com/user-attachments/files/18936671/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf)

My build command 
python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16 DISPLAY=st7789

kdschlosser, 2025-02-24:
give this a try for your script..

```python
import lcd_bus
from micropython import const
import machine

_WIDTH = const(240)
_HEIGHT = const(240)
_BL = const(4)
_RST = const(21)
_DC = const(16)

_MOSI = const(17)
_MISO = const(-1)
_SCK = const(18)
_HOST = const(1) # SPI2

_LCD_CS = const(-1)
_LCD_FREQ = const(40000000)

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)

import st7789 # NOQA
import lvgl as lv # NOQA

display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    reset_pin=_RST,
    reset_state=st7789.STATE_LOW,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7789.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)

display.set_power(True)
display.init()
display.set_backlight(100)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)

import task_handler # NOQA
task_handler.TaskHandler()
```

yeyf2023, 2025-02-24:
I used the code above, but it still doesn't display with a hard reset.

Here is a picture of the ESP32S3
[image]

kdschlosser, 2025-02-24:
I think you have gotten a counterfeit ESP32. Espressif has their name etched into the chip with their logo. I am not seeing that on yours. This has happened to me with ones I bought from Amazon.

yeyf2023, 2025-02-25:
Thanks for tips. I will investigate.
Yesterday, I tested the 135 * 240 st7789 LCD with this S3 MCU, and it was able to display, but there were still some issues.

[image]

tetopik, 2025-03-02:
> I used the code above, but it still doesn't display with a hard reset.
> 
> Here is a picture of the ESP32S3
> 
> [image]

I'm actually own those lcd, it requires the spi polarity to be reversed.
```py
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    spi_mode=2, # phase | (polarity<<1)
    freq=_LCD_FREQ,
    dc=_DC,
    cs=_LCD_CS,
)
```

kdschlosser, 2025-03-02:
and there is the answer. use `mode=2`

yeyf2023, 2025-03-03:
Thanks for the tip.
I noticed that the LCD does not have a CS pin, and the setting when not using LVGL, which has been resolved.
but i used spi_mode=3. phase=1 polarity=1