lvgl-micropython/lvgl_micropython #304: external SPI font

wxzed, 2025-02-25:
I want to extend an external SPI font library. Is there a way to implement this at the Python layer? I found that I can't reassign get_glyph_bitmap at the Python level.

kdschlosser, 2025-02-25:
OoOoo Hmmm...

You would need to write a driver to access the data on the font chip That is not the hard part. what the hard part is is figuring out how the font is stored on the chip. What is the format of it? I would need to know the make and model of the chip to pull up the datasheet and see what is says....

wxzed, 2025-02-25:
Not that detailed, I just don’t know how to call my own get_glyph_bitmap. I just want to understand the general structure. With this operation, I can only use get_glyph_dsc.
```
self.desc.font = lv.font_t()
self.desc.font.get_glyph_dsc = test_get_glyph_dsc
```

wxzed, 2025-02-27:
Thank you for your response. I tried the method you provided, but it didn't work.
Now, I want to try another approach using a different module (mod). Here is a rough outline of my code:
```
static lv_font_t my_custom_font = {
    .get_glyph_dsc = my_get_glyph_dsc,
    .get_glyph_bitmap = my_get_glyph_bitmap,
    .line_height = 16,  
    .base_line = 0
};

typedef struct {
    mp_obj_base_t base;
    const lv_font_t *font;  
} lv_font_obj_t;

static mp_obj_t lv_font_obj_get_font(mp_obj_t self_in) {
    lv_font_obj_t *self = MP_OBJ_TO_PTR(self_in);
    //return MP_OBJ_FROM_PTR(&self->font); 
    return MP_OBJ_NEW_SMALL_INT((intptr_t)(self->font));
    //return mp_const_none;
}
static MP_DEFINE_CONST_FUN_OBJ_1(lv_font_obj_get_font_obj, lv_font_obj_get_font);

static mp_obj_t extern_font_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_kw, const mp_obj_t *args) {
    lv_font_obj_t *self = m_new_obj(lv_font_obj_t);
    self->base.type = &machine_exfont_type; 
    self->font = &lv_font_montserrat_16;
    return MP_OBJ_FROM_PTR(self); 
}

static void extern_font_print(const mp_print_t *print, mp_obj_t self_in, mp_print_kind_t kind) {
    mp_printf(print, "<Exfont object>");
}

static const mp_rom_map_elem_t lv_font_obj_locals_dict_table[] = {
    { MP_ROM_QSTR(MP_QSTR_font), MP_ROM_PTR(&lv_font_obj_get_font_obj) },  
};

static MP_DEFINE_CONST_DICT(lv_font_obj_locals_dict, lv_font_obj_locals_dict_table);

MP_DEFINE_CONST_OBJ_TYPE(
    machine_exfont_type,
    MP_QSTR_Exfont,
    MP_TYPE_FLAG_NONE,
    print, extern_font_print,
    make_new, extern_font_make_new,
    locals_dict, &lv_font_obj_locals_dict
    );
```
My intention is that the type of myfont.font() should be 'lv_font_t', so that I can use it in the same way as lv.font_montserrat_16. However, I have tried several methods but haven't achieved this yet.
Do you have any suggestions?
```
>>> from machine import Exfont
>>> myfont = Exfont()
>>> dir(myfont)
['__class__', 'font']
>>> type(myfont.font())
<class 'int'>

>>> import lvgl as lv
>>> type(lv.font_montserrat_16)
<class 'lv_font_t'>
>>> 
```

kdschlosser, 2025-02-27:
it's not going to work. It is an example of what you need to do. I don't know exactly what you have to do because I don't know what the font chip is or how it stores the data.

kdschlosser, 2025-02-27:
if you get me a link to the data sheet for the font chip I will be able to help more with this.

wxzed, 2025-02-27:
[test.zip](https://github.com/user-attachments/files/19001391/test.zip)
I don’t have the datasheet here, but I do have the previous driver code from LVGL-C.
Here is the part where the font is defined.
```
// Callback function to get the glyph descriptor
bool myGetGlyphDscCb_16(const lv_font_t * font, lv_font_glyph_dsc_t * dsc_out, uint32_t unicode_letter, uint32_t unicode_letter_next) {
    // Set the glyph descriptor, including information such as glyph width, height, offset, etc.
    //printf("ff\n");
    dsc_out->box_h = 12;   /* Height of the glyph bitmap (in pixels) */
    dsc_out->box_w = 16;   /* Width of the glyph bitmap (in pixels) */
    if(unicode_letter < 128){
      dsc_out->adv_w = ASCII_GetInterval(unicode_letter,ASCII_12_A);   /* Letter spacing */
    }else{
      dsc_out->adv_w = 12;   
    }
    dsc_out->ofs_x = 0;    /* X offset of the glyph bitmap (in pixels) */
    dsc_out->ofs_y = 0;    /* Y offset of the glyph bitmap (in pixels), relative to the baseline */
    dsc_out->bpp = 1;      /* Bits per pixel: 1/2/4/8 */
    dsc_out->is_placeholder = false;
    return true;           /* true: glyph found; false: glyph not found */

}

// Callback function to get the glyph bitmap
const uint8_t * myGetGlyphBitmapCb_16(const lv_font_t * font, uint32_t unicode_letter) {
    // Return pointer to glyph bitmap data
    // Replace the example return value below with actual bitmap data for character "你"
    static uint16_t u2g;
    if (unicode_letter < 128) {
        ASCII_GetData(unicode_letter, ASCII_12_A, _pBits);
    } else {
        u2g = U2G(unicode_letter);
        gt_12_GetData((u2g >> 8) & 0xff, u2g & 0xff, _pBits);
    }
    return (const uint8_t*)_pBits; // Return pointer to array containing bitmap data for character "你"
}

// Callback function to get the glyph descriptor
bool myGetGlyphDscCb_24(const lv_font_t * font, lv_font_glyph_dsc_t * dsc_out, uint32_t unicode_letter, uint32_t unicode_letter_next) {
    // Set glyph descriptor, including glyph width, height, offset, etc.
    dsc_out->box_h = 24;   /* Height of the glyph bitmap (in pixels) */
    dsc_out->box_w = 24;   /* Width of the glyph bitmap (in pixels) */
    if (unicode_letter < 128) {
        dsc_out->adv_w = ASCII_GetInterval(unicode_letter, ASCII_24_B);   /* Letter spacing */
    } else {
        dsc_out->adv_w = 24;   /* Letter spacing */
    }
    dsc_out->ofs_x = 0;    /* X offset of the glyph bitmap (in pixels) */
    dsc_out->ofs_y = 0;    /* Y offset of the glyph bitmap (in pixels), relative to the baseline */
    dsc_out->bpp = 1;      /* Bits per pixel: 1/2/4/8 */
    dsc_out->is_placeholder = false;
    return true;           /* true: glyph found; false: glyph not found */
}

// Callback function to get the glyph bitmap
const uint8_t * myGetGlyphBitmapCb_24(const lv_font_t * font, uint32_t unicode_letter) {
    // Return pointer to glyph bitmap data
    // Replace the example return value below with actual bitmap data for character "你"
    static uint16_t u2g;
    if (unicode_letter < 128) {
        ASCII_GetData(unicode_letter, ASCII_24_B, _pBits);
    } else {
        u2g = U2G(unicode_letter);
        GBK_24_GetData((u2g >> 8) & 0xff, u2g & 0xff, _pBits);
    }
    return (const uint8_t*)_pBits; // Return pointer to array containing bitmap data for character "你"
}

const lv_font_t my_custom_font_16 = {
  .get_glyph_dsc = myGetGlyphDscCb_16,
  .get_glyph_bitmap = myGetGlyphBitmapCb_16,
  .line_height = 15,  // Line height
  .base_line = 3,     // Baseline
  .subpx = LV_FONT_SUBPX_NONE,
};  // Font parsing structure for the font chip

const lv_font_t my_custom_font_24 = {
  .get_glyph_dsc = myGetGlyphDscCb_24,
  .get_glyph_bitmap = myGetGlyphBitmapCb_24,
  .line_height = 27,
  .base_line = 3,
  .subpx = LV_FONT_SUBPX_NONE,
};  // Font parsing structure for the font chip
```

kdschlosser, 2025-02-27:
Fantastic. That is what is needed to handle the LVGL side of things. What is not included is the code that actually fetches the data from the font chip...

These functions are missing..
* ASCII_GetData
* ASCII_GetInterval
* U2G
* GBK_24_GetData
* gt_12_GetData

kdschlosser, 2025-02-27:
OK so here you go. Make sure to read the comments near the top. 

```python
import lvgl as lv
from micropython import const  # NOQA

import machine

# replace the "..." in the line below with the size of the `_pbits` array
_PBITS_SIZE = const(...)

# set up your SPI bus. If the bus is shared with something else then you only
# need to create the device passing the shared bus to all of the things that 
# use the bus.

spi_bus = machine.SPI.Bus(
    host=1,
    mosi=...,
    miso=...,
    sck=...,
)

spi_device = machine.SPI.Device(
    spi_bus=spi_bus,
    cs=...,
    freq=20000000
)

_ASCII_5X7 = const(1)
_ASCII_7X8 = const(2)
_ASCII_6X12 = const(3)
_ASCII_12_A = const(4)  # 12 * 16
_ASCII_8X16 = const(5)
_ASCII_12X24_A = const(6)
_ASCII_12X24_P = const(7)
_ASCII_16X32 = const(8)
_ASCII_16_A = const(9)
_ASCII_24_B = const(10)
_ASCII_32_B = const(11)

def gt_read_data(sendbuf, _, receivebuf, receivelen):
    spi_device.write(sendbuf)

    buf = bytearray(0x00)

    for i in range(receivelen):
        buf[0] = 0x00
        spi_device.write_readinto(buf, buf)
        receivebuf[i] = buf[0]

    return 1

def r_dat_bat(address, DataLen, pBuff):
    buf = bytearray([0x03])
    spi_device.write(buf)  # <发送读取命令

    buf[0] = (address >> 16) & 0xFF
    spi_device.write(buf)  # <发送24bit地址

    buf[0] = (address >> 8) & 0xFF
    spi_device.write(buf)

    buf[0] = address & 0xFF
    spi_device.write(buf)

    for i in range(DataLen):
        buf[0] = 0XFF
        spi_device.write_readinto(buf, buf)  # <循环读数
        pBuff[i] = buf[0]
  
[... 253 lines cut ...]
    if unicode_letter < 128:
        ASCII_GetData(unicode_letter, _ASCII_24_B, _pBits)
    else:
        u2g = U2G(unicode_letter)
        GBK_24_GetData((u2g >> 8) & 0xff, u2g & 0xff, _pBits)

    return _pBits  # Return pointer to array containing bitmap data for character "你"

my_custom_font_16 = lv.font_t(dict(
  get_glyph_dsc=myGetGlyphDscCb_16,
  get_glyph_bitmap=myGetGlyphBitmapCb_16,
  line_height=15,  # Line height
  base_line=3,  # Baseline
  subpx=lv.FONT_SUBPX.NONE  # NOQA
))  # Font parsing structure for the font chip

my_custom_font_24 = lv.font_t(dict(
  get_glyph_dsc=myGetGlyphDscCb_24,
  get_glyph_bitmap=myGetGlyphBitmapCb_24,
  line_height=27,
  base_line=3,
  subpx=lv.FONT_SUBPX.NONE,  # NOQA
))  # Font parsing structure for the font chip

```

kdschlosser, 2025-02-27:
That is only the first version of the code and I am expecting it to be pretty slow. We can always tweak and optimize the code once we know it is working properly.

wxzed, 2025-02-28:
Thank you very much for your reply, it looks like exactly what I was looking for. I will verify it on my end. Thanks again.

kdschlosser, 2025-02-28:
no worries m8. It may or may not work properly out of the box but at least we have something to work off of.

tangjie133, 2025-03-04:
Hello, now I also encountered the same problem, using the above method to run the error, the error content:

> Traceback (most recent call last):
  File "<stdin>", line 348, in <module>
SyntaxError: Cannot convert 'function' to pointer!

kdschlosser, 2025-03-04:
what? 

Why are you trying to hijack someone else's issue?

tangjie133, 2025-03-04:
It was provided by a friend of mine who can't handle it right now.