lvgl-micropython/lvgl_micropython #305: Python version 3.13 unsupported?

kamil-n, 2025-02-28:
Hello kdschlosser,

**Describe the bug**
`compile()` fails with "RuntimeError: unable to locate python version used in the ESP-IDF"

**Expected behavior**
Python 3.13 is supported

It all comes down to line [1472](https://github.com/lvgl-micropython/lvgl_micropython/blob/main/builder/esp32.py#L1472), where Python versions >= 3.8 are listed. Manjaro now ships with Python 3.13. It is a trivial fix and I wouldn't bug you (heh) about it, but I'm curious as to why those versions are listed there in such way. I compiled my binaries successfully and everything works so far.

I appreciate your hard work. Kind regards, 
kamil-n

kdschlosser, 2025-03-01:
The reason why those versions are keyed in like that is the script is locating the virtual environment that was used when compiling. Those environments have `esptool` in them and that tool is what is used to flash the ESP32 with. So in order to provide a complete command the user needs to run in order to flash their ESP I need to locate where a working python is that has `esptool` installed.

The build script when it runs actually creates 3 separate binaries that need to be written to the ESP32.  What I do in that script is I package those 3 binaries into a single binary to make it easier to deal with and also to make it so it's only a single file.

closed 2025-03-01 by kdschlosser

kamil-n, 2025-03-01:
Thank you for the explanation.

I understand that it needs to point (full path) to the virtual env. binary (for me it is `/home/kamil-n/.espressif/python_env/idf5.2_py3.13_env/bin/python`).
But if the esptool binary was built with system Python, couldn't we just take the version from [sys.version_info](https://docs.python.org/3/library/sys.html#sys.version_info)? Maybe check if `sys.version_info[1] > 7` to mirror current logic.

Again, appreciating you hard work, kind regards,
kamil-n

kdschlosser, 2025-03-01:
The reason why it is not like that is because esp-idf used to download it's own already set up venv from their servers. This has changed but I saw no reason to update the code as it works without any issues. Python 1.13 is pretty new and has just now started to ship with some OS's as the packaged Python. It only took me a minute to adjust the code to support Python 3.13. It is actually better like this so I know when a new version of Python comes out that things are working properly.

kdschlosser, 2025-03-01:
OH!! and the other reason why I am not using the version of the Python that is running the build script is because of how the IDF build system locates Python. It could be running on a completely different version of Python that what was used to run the build script. That's if the user has more than one installed.

commit e99e3f62 (2025-03-01): add python 3.13 to supported versions..

Fixes #305