lvgl-micropython/lvgl_micropython #307: Yep it was designed for the Pi

morrisis, 2025-03-01:
> isn't RPI  Raspberry Pi?
> 
>  

 _Originally posted by @kdschlosser in [#306](https://github.com/lvgl-micropython/lvgl_micropython/issues/306#issuecomment-2692467421)_

Sure the display is designed for the Raspberry Pi. It works well on the ESP32. And it reportedly works in lvgl_micropython, or at least there is a driver for it that is supported. I just haven't got it working yet.

This is one of many pages I read to help get it to work on the ESP32
https://www.tech-sparks.com/using-esp32-with-arduino-ide-to-control-an-lcd-display/

Any ideas?

kdschlosser, 2025-03-02:
Oh OK.

Have you tried flipping the reset state from LOW to HIGH to see if that works?

kdschlosser, 2025-03-02:
some of the ST displays are LOW and I believe most of the ILI displays are HIGH.