lvgl-micropython/lvgl_micropython #309: ESP32-P4

dcmcshan, 2025-03-04:
Here's a fun one for you.  For a new project, I just purchased an ESP32 P4 from Waveshare.  It has a MIPI/DSI...

dcmcshan, 2025-03-04:
I didn't say it would be easy - I said it would be fun :)

The dev board I have ordered (https://www.waveshare.com/esp32-p4-nano.htm) uses a ESP32-C6-MINI for WiFi and BLE, so an option might be to run micropython on that and let the P4 live in C space or whatever it prefers.  But I've not looked into it much as yet.

kdschlosser, 2025-03-04:
what I hate with these dev boards is they add all kinds of crap to them that most times you never use. Take that waveshare board.. Out of 55 pins there are only 23 pins left that the user is able to use and only 6 of them are ADC pins. All the rest are consumed by things like the Ethernet and the connection to the C6. Why would they have WiFi AND Ethernet? That seems rather silly.

Make a mare bones board with no extras on it and then design the extras so they can stack on top of each other. Don't use header pins for the stacking. Design it so that they bridge across using a connector that is not a pin header and can be physically attached using standoffs. Have the pins broken out to the edges to holes a user can add a header to if they want to use pin headers. This gives them the option of using pin headers for development or designing a board to stack on top to do what they want. Keeps the whole package nice and neat.

favnec5, 2025-05-13:
Hi Kevin, it's been a long time.
What do you think about the MIPI interface of the P4 ?
it leaves me totally uncertain about all the work you've already done.
I'll be receiving this little newt week : https://docs.m5stack.com/en/core/Tab5
Can't wait to test this thing ! :-)

kdschlosser, 2025-05-13:
the P4 is still not yet supported in MicroPython.

favnec5, 2025-05-13:
Damn !  i saw your discussion with mpy team... this is not clear. had you tried it with your P4 that you were due to receive ?

kdschlosser, 2025-05-13:
There is a decent amount of work that needs to be done to get it to work because every single ESP32 up to the P4 has had WiFi built into it. There are also additional pin definitions that need to be handled as well as the processor being different than any of the other ESP32's made before it. There is also a 2D graphics accelerator built in that it would be foolish to not take advantage of as well.

favnec5, 2025-05-25:
Hi Kevin, i worked a little.
please, look at this : https://github.com/orgs/micropython/discussions/17359
This is here : https://github.com/favnec5/micropython/tree/favnec5-esp32p4-1
I push it on MPY master.

kdschlosser, 2025-05-25:
I have upgraded MicroPython to 1.25 and I have also upgraded the ESP-IDF to 5.4. There are some other kinks I am working out in the code to fix some other issues then I will release it. from there was can look at the P4 board development you have been working on. I have a p4 sitting right here on my desk that I have been wanting to use. 

I already have written  DSI driver for the p4.. it will be exciting to see how that all works out.

favnec5, 2025-05-25:
> I have upgraded MicroPython to 1.25 and I have also upgraded the ESP-IDF to 5.4. There are some other kinks I am working out in the code to fix some other issues then I will release it. from there [...]

oh yeah ! :-)

favnec5, 2025-05-31:
I’m trying to write p4 dsi  usermode driver , from m5 tab5 demo code. it crashs my p4 at this time :-/ … lol !!
I tested lot other thing on my m5 tab5, like wifi, Bluetooth, uart, pwm, i2c, brightness, power gestion, it’s ok.