lvgl-micropython/lvgl_micropython #310: Initializing ILI9342C display for M5Stack Tough

thatbudakguy, 2025-03-05:
I'm trying to get a minimal LVGL example working on an [M5Stack Tough](https://docs.m5stack.com/en/core/tough) device (ESP32).

This device has an ESP32 with an ILI9342C display, AXP192 power management IC, and I believe an FT6X36 for touch. I think its SPI bus may be shared between the display and an SD card reader.

So far, I've compiled with:
```
python make.py esp32 DISPLAY=ILI9341 BOARD=ESP32_GENERIC
```

And flashed and got into the REPL, which works fine.

I'm currently running this script:

```py
import ili9341  # NOQA
import lcd_bus  # NOQA
import lvgl as lv
from machine import SPI
from micropython import const

# Display config
_WIDTH = const(320)
_HEIGHT = const(240)
_FREQ = const(40000000)  # 40MHz

# Pin mappings
_HOST = const(2)
_MOSI = const(23)
_MISO = const(19)
_SCK = const(18)
_CS = const(5)
_DC = const(15)
_RST = const(33)
_BL = const(3)
_PWR = const(2)

spi_bus = SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK,
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=_DC,
    cs=_CS,
    freq=_FREQ,
)

display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    reset_pin=_RST,
    reset_state=ili9341.STATE_HIGH,
    color_space=lv.COLOR_FORMAT.RGB565, # ILI9341 only supports RGB565
    power_pin=_PWR,
)

display.set_power(True)
print("Display power on")

display.init(1)
print("Display init")

display.set_backlight(100)
print("Display backlight on")

scrn = lv.screen_active()

width = scrn.get_width()
height = scrn.get_height()

center_x = int(width / 2)
center_y = int(height / 2)

obj_width = int(width / 5)
obj_height = int(height / 5)

obj1 = lv.obj(scrn)
obj1.set_pos(center_x - 10 - obj_width, center_y - 10 - obj_height)
obj1.set_size(obj_width, obj_height)
obj1.set_style_bg_color(lv.color_hex(0xFF0000), 0)

obj2 = lv.obj(scrn)
obj2.set_pos(center_x  + 10, center_y - 10 - obj_height)
obj2.set_size(obj_width, obj_height)
obj2.set_style_bg_color(lv.color_hex(0x00FF00), 0)

obj3 = lv.obj(scrn)
obj3.set_pos(center_x  - int(obj_width / 2), center_y + 10)
obj3.set_size(obj_width, obj_height)
obj3.set_style_bg_color(lv.color_hex(0x0000FF), 0)

print("Objects created")

import task_handler

task_handler.TaskHandler()
```

The script completes successfully and all messages are printed to the console, at which point it exits. Nothing appears to be drawn to the screen; it stays black.

After executing, the device makes a fairly audible whining noise that persists until I reset it. If I run the script twice in a row without resetting, I get a fatal error that reboots the device.

## Potentially useful links
- https://github.com/m5stack/M5Stack/blob/master/src/utility/In_eSPI_Setup.h (SPI config from vendor's library?)
- https://github.com/lemariva/micropython-core2 (the CORE2 is almost identical to the Tough)
- https://github.com/sci-bots/m5-lvgl/issues/1
- https://github.com/russhughes/ili9342c_mpy (alternative driver for ILI9342C)

thatbudakguy, 2025-03-05:
I've tried flipping the ILI9341 type (calling `init` with 1 and 2) and also flipping the `reset_state`; neither seemed to have an effect.

kdschlosser, 2025-03-05:
That thing has a crap load of hardware in it.

The things that are going to require drivers to be written are the following.

10 channel ADC
8 channel ADC
2 channel DAC
2 channel DAC
4 channel PMU

The reason why is because those are basically channel expanders. They use I2C or SPI to add more input/output pins. Some of the display connections are made to these things as well as the touch panel connections.

kdschlosser, 2025-03-05:
There is also a microphone, speakers, real time clock and sd card reader.. I am sure I am missing a few things as well.

thatbudakguy, 2025-03-05:
> That thing has a crap load of hardware in it.

Most of that other stuff I don't need (at least immediately), but maybe that means I should just dive in and start making a custom board definition...

I'm guessing the main roadblock is the PMU, because a bunch of stuff connects to it and from what I can tell the vendor's own libraries have code related to it integrated into the display drivers (and other places too).

kdschlosser, 2025-03-05:
You still need to write drivers for some of them because they control the reset lines and the power lines for the display and the touch. that is the reason why they are not turning on.

kdschlosser, 2025-03-05:
I am working on writing the drivers needed to get the display and touch powered up. It's gonna take me a few days tho as I have a lot of things going on right now personally.

thatbudakguy, 2025-03-05:
Grateful for any assistance you can give! No rush.

thatbudakguy, 2025-03-06:
At this point you've probably seen already, but just in case it's helpful, there are some drivers already out there that might be useful to reference (presuming you're working on the AXP192):

- A [pure python one](https://gist.github.com/ropg/7216ba90a9d7697114d4ba8aea7bee3c)
- A [platform-agnostic C one](https://github.com/tuupola/axp192) 
- A [fork of the above one](https://github.com/usedbytes/axp192)

Thanks in advance!

kdschlosser, 2025-03-06:
There is a specific way that I need to have the driver written so it can be used as a base class for board specific versions of it.

kdschlosser, 2025-03-08:
OK so this is where I am with it...

The code is to large to paste so I attached the file of what I have so far.

[axp192_2.py.txt](https://github.com/user-attachments/files/19142864/axp192_2.py.txt)

I have written it so you access the GPIO's much in the same way as you would using the native MicroPython classes (Pin, ADC, PWM). They are really close to being identical in their API with the MicroPython classes to it will be easy to use. There are some modifications I made to how the display drivers and touch drivers work. I will commit those changes once I am finished with this driver because this driver will get pushed up to the repo as well. 

The driver above is not specific to your board. We will subclass the driver and put in the things needed to make it specific to your board. There are a lot of M5 boards that use this IC and I am sure that they are not going to be wired the same. That is the reason why I wrote it this way, I want to be able to use the driver no matter the board the IC is being used on. Trying to keep things as flexible as possible and also to be able to reuse code as much as possible. Less to maintain that way.

thatbudakguy, 2025-04-06:
Can you tell me how I should go about using this? Not sure how to proceed from here.