lvgl-micropython/lvgl_micropython #313: Compile error with rp2

yeyf2023, 2025-03-16:
I used the following command to compile rp2 (rp2040), and I got an error.
Command:
python3 make.py rp2 BOARD=RPI_PICO DISPLAY=st7789
Error:
make[3]: *** [CMakeFiles/firmware.dir/build.make:3699: CMakeFiles/firmware.dir/machine_spi.c.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:1737: CMakeFiles/firmware.dir/all] Error 2
make[1]: *** [Makefile:91: all] Error 2
make: *** [Makefile:59: all] Error 1

kdschlosser, 2025-03-17:
I need someone that is familiar with the RP2040 that is willing to help with wring proper drivers for that MCU. I have not had any luck locating anyone that is willing to lend a hand with it. I don't know the hardware or the SDK. I don't have time available to learn it at the moment. If you know someone that has the knowledge maybe if you ask them they might be willing to lend a hand with it. Or perhaps you have the knowledge to be able to help out.

Sorry for the bad news on that. In the coming weeks I will work on it to get the SPI driver working again but that is about all I am going to be able to do.

hemonu, 2025-04-12:
I have a PICO_W board available and have a look into that. What has to be done is to write a machine_spi.c file for the rp2 in the micropy_updates/rp2 folder. It has to have the same interface and functionality as the files in the esp32 and stm32 subfolders. Is that correct?
I will spend some time on that, but no firm promise, that I will be able to make it ...